platformio/platform-espressif32 #95: ULP Support ?

timpur, 2018-06-14:
Looking into support for ULP for platfrom io, any chance this will happen soon ?

https://github.com/espressif/arduino-esp32/issues/1491
https://github.com/duff2013/arduino_ulp

Cchavarriaalcala, 2019-02-05:
👍🏼👍🏼👍🏼🙌🏼

sch0bert, 2019-02-05:
Hello @timpur,

Really need this support and I will appreciate to prioritize as it limits the capabilities of ESP32 in PlatformIO.

Hope to hear from you

Regards,
Pascal.

brezuicabogdan, 2019-04-12:
Any news on this feature? Thanks.

sch0bert, 2019-04-14:
@brezuicabogdan I'm also waiting this implementation, is sad that Platformio guys don't prioritize neither provide an estimated delivery date...

norges, 2019-05-08:
I just found out that ulp for the esp32 is not supported. As I like the platformio approach to use one system for all projects I miss this support very much. Please add it to platformio.

sch0bert, 2019-05-08:
@norges I completely agree with you, I had to move all my projecto to traditional ESP-IDF tools because of this. The worst part is that they don't provide any ETA of when is it going to be supported, this is terrible! The issue was reported last year!

hectorMtz017, 2019-05-08:
@timpur we would really appreciate the ETA of when it will be supported. Currently working all of my projects outside of platformIO because of this.

timpur, 2019-05-10:
Dont think i have time to do this atm ... feel welcome to make a PR :P

brentforman, 2019-06-03:
i need to use the ULP. I have been relegated to make files and notepad++ for editing. please bring an end to this masochism and support ULP stuff in platformIO

extesy, 2019-06-03:
Tagging @ivankravets @valeros as primary contributors to this repo.

sch0bert, 2019-06-03:
@brentforman My recommendation is to move to the regular ESP-IDF + Eclipse, PlatformIO team seems to never get this job done. I paid for the Professional plan and I completely regret about it.

ivankravets, 2019-06-04:
@sch0bert Thank you so much for supporting us with PIO Plus! It helps us so much to keep PlatformIO alive and neutral to any architecture/dev/platform.

As I remember, we wanted to resolve this issue a few months ago. However, our team said that it requires a lot of efforts. This Espressif 32 dev/platform is community driven and we are opened for the PR. We will try to contact @Espressif and ask them to help with this issue.

Our main vision that semiconductor companies will personally maintain integration components (such as this dev/platform) for PlatformIO. From the other side, PlatformIO Team will work on the PlatformIO Core. This is actually what we have been doing last months. See upcoming release of [PlatformIO Core 4.0](https://github.com/platformio/platformio-core/blob/develop/HISTORY.rst).

If we do not have updates on this issue after PlatformIO Core 4.0 release, we will back here and try to resolve it personally.

> My recommendation is to move to the regular ESP-IDF + Eclipse

Yes, I think this is a normal temporary solution. If we resolve this issue, your code will be compatible with PlatformIO.

Sorry for the issue.

valeros, 2019-06-20:
Hi, guys @timpur @Cchavarriaalcala @sch0bert @brezuicabogdan @hectorMtz017 @brentforman @extesy ! We finally found some time and added experimental support for the ULP coprocessor. Would be great if some of you could install the espressif32 platform from the [dev branch](http://docs.platformio.org/en/latest/platforms/espressif32.html#stable-and-upstream-versions) and test it with your projects. Here is an early [doc page](http://docs.platformio.org/en/latest/platforms/espressif32.html#ulp-coprocessor-programming) and some basic examples:
https://github.com/platformio/platform-espressif32/tree/develop/examples/espidf-ulp-adc
https://github.com/platformio/platform-espressif32/tree/develop/examples/espidf-ulp-pulse

brezuicabogdan, 2019-06-20:
This is great news @valero. I will try to do this week although I cannot promise anything.

ivankravets, 2019-06-28:
Please update to https://github.com/platformio/platform-espressif32/releases/tag/v1.9.0

closed 2019-06-28 by ivankravets