platformio/platform-espressif32 #102: using custom or lib in my folder structure

wegunterjr, 2018-07-02:
PDetected non-PlatformIO lib_dir option in [env:esp32dev] section

Is it possible to use this feature? I am getting this warning when compiling.

ivankravets, 2018-07-02:
This option should be declared in `[platformio]` section. See http://docs.platformio.org/en/latest/projectconf/section_platformio.html#lib-dir

closed 2018-07-02 by ivankravets

wegunterjr, 2018-07-02:
right.. i had it in the platformio, but still got the warning...
```
Detected non-PlatformIO 'lib_dir' option in '[env:esp32dev]' section
```
```
[env:esp32dev]
platform = espressif32
lib_dir = lib/BLE Light
lib_extra_dirs = ~/Documents/Arduino/libraries
board = esp32dev
framework = arduino

lib_deps =
      PubSubClient
      ArduinoJson
      AutoPID
      WiFiClientSecure
;      ESP32 BLE Arduino
```
do i need it if I have libraries in the lib folder or will it automatically search for the lib folder?
Also, do i need to use 
``` 
#include <specialLib.h>
or
#include "specialLib.h"