platformio/platform-espressif32 #103: esp-mdf

kemaldede, 2018-07-02:
Can you add esp-mdf support please? Than k you.
https://github.com/espressif/esp-mdf

PreferHardware, 2018-08-15:
+1 please.

Elijahg, 2018-12-10:
+1 for support for this. It's a fairly small SDK so shouldn't be too much work, and it's pretty stable now. Unfortunately simply adding the header files in the MDF to the platformio.ini doesn't work.

Elijahg, 2018-12-10:
+1 for support for this. It's a fairly small SDK so shouldn't be too much work, and it's pretty stable now. Unfortunately simply adding the header files in the MDF to the platformio.ini doesn't work. If there's any chance you could look at adding support @ivankravets, it'd be very much appreciated. I've just switched to Platformio and really like it a lot, thank you for all your efforts so far! Unfortunately without ESP-MDF support I'm forced to use Platformio purely as a text editor!

rikkertkoppes, 2019-01-03:
I have started getting esp-mdf to work here: https://github.com/nlcamarillo/platform-espressif32/commits/espmdf

This is all quite new to me, so half of the time I don't know what I am doing. I managed to get some projects compiling. This is what I did:

I copied the `esp-idf` builder to `esp-mdf` and added and adjusted paths to make it work, also I added `esp-mdf` as a framework and as a framework option to the `esp32dev` board. See the commits for the changes I made and the shortcuts I took (like ignoring some libraries)

Next, I cloned the [`esp-mdf`](https://github.com/espressif/esp-mdf) repo into `~/.platformio/packages/framework-espmdf`. Make sure you clone recursively. I guess the repo could also be hosted somewhere else and the package repository added to the platform.json. But I don't know how that works. A PR for this would be appreciated, as it would also solve the next problems:

I merged the sdkconfig of one of my existing esp-mdf (non-platformio) projects with the default sdkconfig of a (platformio) esp-idf project. A lot of items where missing. This can probably be solved by adding a working default sdkconfig and adjusting the [copy script](https://github.com/nlcamarillo/platform-espressif32/blob/espmdf/builder/frameworks/espmdf.py#L109-L117) in the builder

I copied `esp-idf/components/esp32/ld/esp32.common.ld` from the `esp-idf` framework in platformio to the `esp-mdf` framework. I added bits about `_iram_end`, `soc_reserved_memory_region_start` and `soc_reserved_memory_region_end`. I really have no clue to what is happening in these files, so some info here is appreciated.

So this got my little get_started mdf project compiling and uploaded from platformio. But not working yet. It keeps rebooting on the esp dev board.

I'd like to head your experiences. Maybe this helps taking some steps in the right direction.

Phando, 2019-02-08:
Have you had any luck? 
Your effort looks non trivial. There is another option for the Arduino IDE, PainlessMesh, but it too causes the chips to reboot all the time.  If you get the MDF working please update this thread and let us know!

ussjoin, 2019-09-20:
+1 to this request.

extesy, 2019-09-20:
@ussjoin Do you see the 👍 symbol right under the original issue description? Click on it, that is how you do +1 and not by spamming everyone's notifications inbox.

ramsesgarciad, 2021-06-19:
+1 to this request.

sguarin, 2021-11-03:
+1 to this request