platformio/platform-espressif32 #104: why does program compile larger in platformiO than in Arduino?

wegunterjr, 2018-07-03:
When I compile with PlatformIO i get a program of almost 1.9MB, but when I compile with arduino I get only 1.37....

ivankravets, 2018-07-03:
Do you use the latest Arduino Core with Arduino IDE?

wegunterjr, 2018-07-03:
l think so. I pulled in the latest from the github and ran the get.exe How can i verify? 
WHat should I expect to see in regards to difference in size.

ivankravets, 2018-07-03:
Please provide screenshots with memory usage from Arduino IDE and PIO IDE.

wegunterjr, 2018-07-03:
Arduino 1.8.4
[image]

Same program compiled in Platformio
DATA:    [==        ]  24.2% (used 71444 bytes from 294912 bytes)
PROGRAM: [========= ]  90.5% (used 1778762 bytes from 1966080 bytes)

ivankravets, 2018-07-03:
Do you use the latest version of dev/platform v1.1.0? PIO IDE > PIO Home > Platforms

wegunterjr, 2018-07-03:
I am running PlatformIO 3.6.0a4

wegunterjr, 2018-07-03:
[image]

and this version of PlatformIO:  You're up-to-date!
PlatformIO 3.6.0a4 is currently the newest version available.

ivankravets, 2018-07-03:
Thanks! Could you provide a simple project to reproduce this issue? Or pack it to archive and send to me@ikravets.com? We will not share it in public.

wegunterjr, 2018-07-03:
@ivankravets  will do.  Sent.

wegunterjr, 2018-07-03:
@ivankravets i saw in the Espressif/arduino-esp32 gitter some conversations this morning about a recent update also causing someone else a problem...no resolution in the channel.

ivankravets, 2018-07-03:
See the latest messages in https://gitter.im/espressif/arduino-esp32

closed 2018-07-03 by ivankravets