platformio/platform-espressif32 #105: Backtrace decoder for ESP32

me21, 2018-07-03:
There is a backtrace decoder for ESP32 which is installed in Arduino IDE: https://github.com/me-no-dev/EspExceptionDecoder

And another CLI backtrace decoder written in Python: https://github.com/janLo/EspArduinoExceptionDecoder

It would be nice to use it from Platformio IDE somehow. Serial port monitor is already there, it should be possible to copy&paste the backtrace.

KrrishnaShah, 2018-08-10:
is there any way to back-trace in ESP-IDF

yaqwsx, 2018-11-16:
I briefly looked at the implementation of monitor in PlatformIO. It uses miniterm from PySerial, therefore, it is not easy to extend it by post-processing of the output unless we change PySerial or PlatformIO ships its own monitor.

vallettea, 2019-03-15:
would also be interested

gaelj, 2019-10-28:
2 years later and this issue is still open... Oh well I guess I wiill have to keep on using the horrible Arduino IDE to debug my ESP32 Guru Meditations.

pierreverbakel, 2019-11-05:
It keeps amazing me how the Arduino IDE keeps ahead on PlatformIO on some vital points, like this one. I do edit en Visual Code editor with PlatformIO Plugin, try to compile and run in PlatformIO, but then revert to Arduino IDE to understand those crashes. So one could call this IIDDEE-Platform development. Almost like Turbo, but really it feels like using the bird in the Flintstones putting its beak on the rock-record since I know it will do the job... It feels backward indeed ...

louislewis2, 2019-11-10:
Guys, again this is ridiculous that such an important feature is simply left untouched.

yaqwsx, 2019-11-10:
It is, but it would require a significant change to the PIO Serial (e.g. hack the MiniTerm or implement a custom one), so I understand it can take a while to implement.

However, I don't understand why there is no reaction from the PIO maintainers for over a year...

louislewis2, 2019-11-10:
I would honestly settle for a copy and paste solution, even that initially would be useful.

pfeerick, 2019-11-14:
Try [janLo/EspArduinoExceptionDecoder](https://github.com/janLo/EspArduinoExceptionDecoder)... if you copy the stacktrace to a file as suggested in the readme, and then run `decoder.py` from the PIO command line, it should 'just work'. Plus, since it is python based, it should work anywhere PlatformIO does.

yaqwsx, 2019-11-15:
Since PlatformIO currently uses Python3, it should be possible to use contextlib.redirect_stdout around the [call of miniterm](https://github.com/platformio/platformio-core/blob/703b29a05e2481a744e3c5913db814ebf7bdb499/platformio/commands/device.py#L201) in the PlatformIO core. Then a custom file-like object could do the translation. However, this requires a change to the platform API - it has to be able to provide a custom post-processor. Who should I contact with such a request/consultation?

rohansingh, 2020-01-10:
It's super-annoying but here's what I use as a workaround:

```
~/.platformio/packages/toolchain-xtensa32/bin/xtensa-esp32-elf-addr2line -pfiaC -e .pio/build/$ENV/$PROJECT.elf
```

Replace `$ENV` and `$PROJECT` with appropriate values.

g01d10x, 2020-02-26:
I personally, and suspect many others would also be very interested in having this added to PlatformIO :D thanks for the great work so far !!

closed 2020-03-12 by ivankravets

ivankravets, 2020-03-13:
Thanks to @Tasssadar 's  great contribution this issue has been finally resolved! We need some feedback/testing before making the final release of PlatformIO Core 4.3 and this dev-platform. Please re-test:

1. Open PlatformIO IDE Terminal and run `pio upgrade --dev`.
2. Use the upstream version of this dev-platform. See docs https://docs.platformio.org/en/latest/platforms/espressif32.html#upstream
3. Run `pio update`.

See documentation https://docs.platformio.org/en/latest/core/userguide/device/cmd_monitor.html#filters

LeedanJohnson, 2020-03-17:
This works a treat when the upstream dev-platform doesn't break things. On a small demo script, everything went off without a hitch, but when compiling and flashing a larger project (that works fine on the stable platform) something goes wrong. I get a bootloop with garbage coming over the serial line.

valeros, 2020-03-18:
Hi @LeedanJohnson ! The upstream should work as well, could you please provide more detail or even open a new issue?

LeedanJohnson, 2020-03-18:
I'd love to supply more details here, but I really don't know what's going wrong.
While I have "platform = espressif32" in my platformio.ini file, my full project compiles and runs without a problem (Except for an exception occasionally getting thrown that I'm trying to track down).

If I change that to 'platform = https://github.com/platformio/platform-espressif32.git', the project still compiles and will upload, but will bootloop and spit garbage over the serial line.

Interestingly enough, if I purposefully introduce an exception early on in the program, the same project will start up and run as intended (hitting the exception, and dumping the registers before restarting).

LeedanJohnson, 2020-03-18:
The issue is back again, with the same behaviour, although I'm not getting the garbled text over serial now. The following is what comes over serial. The second type of boot loops forever.
`rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:1044
load:0x40078000,len:8896
load:0x40080400,len:5828
entry 0x400806ac
ets Jun  8 2016 00:22:57

rst:0x10 (RTCWDT_RTC_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:1044
load:0x40078000,len:8896
load:0x40080400,len:5828
entry 0x400806ac
`

valeros, 2020-03-18:
@LeedanJohnson What framework/board do you use?

LeedanJohnson, 2020-03-18:
I think this will answer those questions, if not let me know.

[env:esp32dev]
platform = https://github.com/platformio/platform-espressif32.git
board = esp32dev
framework = arduino
monitor_speed = 115200
upload_speed = 921600
build_flags = -DCORE_DEBUG_LEVEL=5

bakwc, 2021-06-07:
I have the same issue.

Version:
```
$ platformio --version
PlatformIO Core, version 5.1.1
```

Config:
```
[env:esp32dev]
platform = espressif32
board = esp32cam
framework = arduino
board_build.f_cpu = 240000000L
upload_protocol = esptool

lib_extra_dirs = lib

build_flags = -DCORE_DEBUG_LEVEL=5

monitor_port = /dev/cu.usbserial-1450
monitor_speed = 115200
```

```
platformio device monitor

--- Available filters and text transformations: colorize, debug, default, direct, esp32_exception_decoder, hexlify, log2file, nocontrol, printable, send_on_enter, time
--- More details at http://bit.ly/pio-monitor-filters
--- Miniterm on /dev/cu.usbserial-1450  115200,8,N,1 ---
--- Quit: Ctrl+C | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H ---

...

Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.
Core 1 register dump:
PC      : 0x400d1437  PS      : 0x00060130  A0      : 0x800d153a  A1      : 0x3ffe9de0
A2      : 0xfecd2c98  A3      : 0x3ffc28f0  A4      : 0x3ffc2cf4  A5      : 0x3ffda948
A6      : 0x3ffe9d9c  A7      : 0x3ffe9d9c  A8      : 0x800d1437  A9      : 0x3ffe9dc0
A10     : 0x00000064  A11     : 0xd4063a52  A12     : 0xd4063a52  A13     : 0x3ffda948
A14     : 0x3ffe9d7c  A15     : 0x3ffe9d7c  SAR     : 0x00000004  EXCCAUSE: 0x0000001c
EXCVADDR: 0xfecd2c9c  LBEG    : 0x4008cb31  LEND    : 0x4008cb41  LCOUNT  : 0xffffffff

ELF file SHA256: 0000000000000000
```

Any ideas?

ivankravets, 2021-06-09:
`pio device monitor -f esp32_exception_decoder`. See https://docs.platformio.org/en/latest/core/userguide/device/cmd_monitor.html#filters

tablatronix, 2021-11-09:
What would cause
Error: (25, 'Inappropriate ioctl for device')
on osx

ivankravets, 2021-11-09:
@tablatronix what is your macOS version? Try to remove `~/.platformio/python3` and `~/.platformio/penv` folders, restart VSCode.

tablatronix, 2021-11-09:
I am not using vscode, cli, but I will poke around a bit in python

ivankravets, 2021-11-09:
You don't need VSCode, just remove those folders and use installer script https://docs.platformio.org/en/latest/core/installation.html#super-quick-mac-linux.

It seems you installed PlatformIO Core globally where there is an outdated `pyserial` package.

egnor, 2022-06-30:
As a breadcrumb for others, if you're adding the filter and it's not doing anything, make sure you have `build_type = debug` in your platformio.ini (https://github.com/platformio/platform-espressif32/issues/393)

PR #286 (2020-03-12): Add esp_exception_decoder filter for device monitor
**Requires https://github.com/platformio/platformio-core/pull/3383.**

Adds a monitor filter that can decode esp32 exception traces, it looks like this with a debug build (the indented part is new): [image]

It was tested on Linux with Python3 and Windows with Python 2 with `esp32dev` board. 

Add `--filter=esp32_exception_decoder` to your monitor_flags to activate it.

Fixes #105