platformio/platform-espressif32 #109: Task watchdog got triggered issue after upgrading to 1.1.0. Works fine in 1.0.2

tsathishkumar, 2018-07-05:
I have a code that looks like below,

```

void setup_wifi();

void WiFiEvent(WiFiEvent_t event)
{
  switch (event)
  {
  case SYSTEM_EVENT_AP_START:
    //can set ap hostname here
    WiFi.softAPsetHostname(_node_name.c_str());
    //enable ap ipv6 here
    WiFi.softAPenableIpV6();
    break;

  case SYSTEM_EVENT_STA_START:
    //set sta hostname here
    WiFi.setHostname(_node_name.c_str());
    break;
  case SYSTEM_EVENT_STA_CONNECTED:
    //enable sta ipv6 here
    WiFi.enableIpV6();
    break;
  case SYSTEM_EVENT_AP_STA_GOT_IP6:
    //both interfaces get the same event
    Serial.print("STA IPv6: ");
    Serial.println(WiFi.localIPv6());
    Serial.print("AP IPv6: ");
    Serial.println(WiFi.softAPIPv6());
    break;
  case SYSTEM_EVENT_STA_GOT_IP:
    Serial.println("WiFi connected");
    Serial.println("IP address: ");
    Serial.println(WiFi.localIP());
    node_ip = WiFi.localIP().toString();
    wifi_connected = true;
    break;
  case SYSTEM_EVENT_STA_DISCONNECTED:
    wifi_connected = false;
    Serial.println("WiFi Disconnected");
    setup_wifi();

    break;
  }
}

void setup_wifi()
{
  if (!wifi_reconnect_in_progress)
  {
    int local_count = 0;
    wifi_reconnect_in_progress = true;

    WiFi.onEvent(WiFiEvent);
    const char *ssid = _ssid.c_str();
    const char *password = _password.c_str();

    Serial.print("Connecting to WiFi SSID: ");
    Serial.println(ssid);
    WiFi.begin(ssid, password);
    while (WiFi.status() != WL_CONNECTED)
    {
      if (++local_count == 20)
      {
        ESP.restart();
      }
      Serial.print(".");
      delay(500);
    }
    wifi_reconnect_in_progress = false;
  }
}

void setup() {
    setup_wifi();
}

void loop() {
    ble_scan_for_devices();
}
```

This code was working perfectly fine in `espressif32@1.0.2`, but after upgrading to `espressif32@1.1.0` it started throwing the below error message after executing the loop few times( the number of execution is not consistent).

```
Task watchdog got triggered. The following tasks did not reset the watchdog in time:
 - IDLE (CPU 0)
Tasks currently running:
CPU 0: wifi
CPU 1: IDLE
```

It appears that the core that handles wifi goes Idle or something and doesn't send a reset to watchdog. But I'm failing to understand what has changed in 1.1.0 and where should I send the reset message from.

ivankravets, 2018-08-10:
Could you reproduce this issue with the latest version of dev/platform? Please do VSCode > Tasks > Run ... > PaltformIO: Update...

ivankravets, 2018-08-13:
Please reopen if you still need help

closed 2018-08-13 by ivankravets

tsathishkumar, 2018-08-14:
I confirm that the upgrade works fine. Thanks.

ivankravets, 2018-08-14:
Thanks! Happy coding with PlatformIO!

tsathishkumar, 2018-08-14:
Thanks for the great project. Much appreciated. We wanted to build something similar when we were initially using Arduino IDE. After we discovered PlatformIO, the development just got easier :) Thanks