platformio/platform-espressif32 #114: [linking] multiple definition of `__cxa_guard_acquire'

TD-er, 2018-08-14:
I just tried to see if our project (ESPeasy) would build with the espressif32@1.3.0 version, which has just recently been released.

Ik get the following linker errors:

```
Linking .pioenvs\esp32dev\firmware.elf
C:\users\gijs\.platformio\packages\framework-arduinoespressif32\tools\sdk\lib\libcxx.a(cxx_guards.o): In function `__cxa_guard_acquire':
/Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/cxx/cxx_guards.cpp:121: multiple definition of `__cxa_guard_acquire'
c:/users/gijs/.platformio/packages/toolchain-xtensa32/bin/../lib/gcc/xtensa-esp32-elf/5.2.0/../../../../xtensa-esp32-elf/lib\libstdc++.a(guard.o):/builds/idf/crosstool-NG/.build/src/gcc-5.2.0/libstdc++-v3/libsupc++/guard.cc:232: first defined here

C:\users\gijs\.platformio\packages\framework-arduinoespressif32\tools\sdk\lib\libcxx.a(cxx_guards.o): In function `__cxa_guard_release':
cxx_guards.cpp:(.text.__cxa_guard_release+0x0): multiple definition of `__cxa_guard_release'
c:/users/gijs/.platformio/packages/toolchain-xtensa32/bin/../lib/gcc/xtensa-esp32-elf/5.2.0/../../../../xtensa-esp32-elf/lib\libstdc++.a(guard.o):guard.cc:(.text.__cxa_guard_release+0x0): first defined here
C:\users\gijs\.platformio\packages\framework-arduinoespressif32\tools\sdk\lib\libcxx.a(cxx_guards.o): In function `__cxa_guard_abort':
cxx_guards.cpp:(.text.__cxa_guard_abort+0x0): multiple definition of `__cxa_guard_abort'
c:/users/gijs/.platformio/packages/toolchain-xtensa32/bin/../lib/gcc/xtensa-esp32-elf/5.2.0/../../../../xtensa-esp32-elf/lib\libstdc++.a(guard.o):guard.cc:(.text.__cxa_guard_abort+0x0): first defined here
collect2.exe: error: ld returned 1 exit status
*** [.pioenvs\esp32dev\firmware.elf] Error 1
```

I don't really have a clue where to start looking, since 1.2.0 builds just fine.
It looks like something that uses locking and maybe some .h file which isn't using either #define or #pragma once kind of structures to make sure it is only built once.

So I was hoping someone could point me in the right direction where to look?

eric-devi, 2018-08-16:
Got the same problem, all of my projects won't work anymore since the last update!

trycoon, 2018-08-26:
Same here, and several of my dependencies fails:
.piolibdeps/ESP Async WebServer_ID306/src/ESPAsyncWebServer.h:38:2: error: #error Platform not supported
I also get c++11 warnings:
.platformio/packages/toolchain-xtensa32/xtensa-esp32-elf/include/c++/5.2.0/bits/c++0x_warning.h:32:2: error: #error This file requires compiler and library support for the ISO C++ 2011 standard. This support is currently experimental, and must be enabled with the -std=c++11 or -std=gnu++11 compiler options.

I think its related to this commit: https://github.com/platformio/platform-espressif32/commit/05ebccecf28f79bf1831d1df55b508936392ecad

TD-er, 2018-09-23:
I have been able to build it.
See this explanation: https://github.com/letscontrolit/ESPEasy/pull/1789#issuecomment-423834356
For our build these build flags had to be removed:
`-lstdc++ -lsupc++`

ivankravets, 2018-10-01:
Could someone provide a simple project to reproduce this issue?

closed 2018-11-24 by ivankravets