platformio/platform-espressif32 #116: unknown opcode or format name 'rsr.PRID'

huseyinkozan, 2018-08-20:
Created `espidf-hello-world` and tried to build, but got this error:

What can I do ?

IDE : Atom
PlatformIO Home : 1.0.2
PlatformIO Core : 3.6.0
Espressif 32 : 1.3.0

```
# hk @ e560 in ~/Documents/PlatformIO/Projects/180820-174701-espidf-hello-world [18:13:41]
$ pio run
Processing esp32dev (platform: espressif32; board: esp32dev; framework: espidf)
-------------------------------------------------------------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
PLATFORM: Espressif 32 > Espressif ESP32 Dev Module
SYSTEM: ESP32 240MHz 320KB RAM (4MB Flash)
DEBUG: CURRENT(esp-prog) EXTERNAL(esp-prog, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny)
Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF MODES: FINDER(chain) COMPATIBILITY(soft)
Collected 0 compatible libraries

Scanning dependencies...
No dependencies
Compiling .pioenvs/esp32dev/freertos/port.o
Compiling .pioenvs/esp32dev/freertos/tasks.o
Compiling .pioenvs/esp32dev/freertos/xtensa_intr_asm.o
/home/hk/.platformio/packages/framework-espidf/components/freertos/portmux_impl.inc.h: Assembler messages:
/home/hk/.platformio/packages/framework-espidf/components/freertos/portmux_impl.inc.h:72: Error: unknown opcode or format name 'rsr.PRID'
/home/hk/.platformio/packages/framework-espidf/components/freertos/portmux_impl.inc.h:72: Error: unknown opcode or format name 'rsr.PRID'
/home/hk/.platformio/packages/framework-espidf/components/freertos/portmux_impl.inc.h:147: Error: unknown opcode or format name 'rsr.PRID'
*** [.pioenvs/esp32dev/freertos/port.o] Error 1
Compiling .pioenvs/esp32dev/freertos/xtensa_vector_defaults.o
/home/hk/.platformio/packages/framework-espidf/components/freertos/xtensa_intr_asm.S: Assembler messages:
/home/hk/.platformio/packages/framework-espidf/components/freertos/xtensa_intr_asm.S:160: Error: unknown opcode or format name 'xsr.INTENABLE'
/home/hk/.platformio/packages/framework-espidf/components/freertos/xtensa_intr_asm.S:163: Error: unknown opcode or format name 'wsr.INTENABLE'
/home/hk/.platformio/packages/framework-espidf/components/freertos/xtensa_intr_asm.S:210: Error: unknown opcode or format name 'xsr.INTENABLE'
/home/hk/.platformio/packages/framework-espidf/components/freertos/xtensa_intr_asm.S:214: Error: unknown opcode or format name 'wsr.INTENABLE'
*** [.pioenvs/esp32dev/freertos/xtensa_intr_asm.o] Error 1
/home/hk/.platformio/packages/framework-espidf/components/freertos/portmux_impl.inc.h: Assembler messages:
/home/hk/.platformio/packages/framework-espidf/components/freertos/portmux_impl.inc.h:72: Error: unknown opcode or format name 'rsr.PRID'
/home/hk/.platformio/packages/framework-espidf/components/freertos/portmux_impl.inc.h:147: Error: unknown opcode or format name 'rsr.PRID'
*** [.pioenvs/esp32dev/freertos/tasks.o] Error 1
============================================================== [ERROR] Took 4.54 seconds ==============================================================

====================================================================== [SUMMARY] ======================================================================
Environment esp32dev            [ERROR]
Environment esp wrover kit      [SKIP]
Environment lolin 32            [SKIP]
```

huseyinkozan, 2018-08-25:
That was a locale problem which I found here : 
https://github.com/espressif/esp-idf/issues/474#issuecomment-324649133

Run Atom like this, and error gone: 
```
LC_ALL=en_US.UTF-8 atom
```

Tried setting locale at `Shell Override`, `Shell Arguments`, and `Shell Environment Variables` but no luck. Will be happy if anyone shows how to set `pio` locale at settings.

closed 2018-08-25 by huseyinkozan

kursadaltan, 2021-12-08:
[en] I have already same problem in VS Code. 
I tried **LC_ALL=en_US.UTF-8 code** command and I solved problem.

[tr] Hocam eline koluna sağlık. Yıllar sonra benim işime yaradı. Kaç saattir uğraşıyordum.

MuhammetOzturk, 2022-06-27:
Hocam halen daha ise yariyor.

kozmotronik, 2023-08-18:
Environment:
OS: Linux x86_64
IDE: Eclipse
SDK: ESP IDF v5.1

Same problem here and this is really an odd and annoying problem. ESP IDF developers should provide a solution for this locale problem! It is a nice workaround solution provided by @huseyinkozan thanks to him. It compiles all without errors when I launch Eclipse as following (same as above solutions only app name is different):
```sh
$> LC_ALL=en_US.UTF-8 eclipse
```

ahmetyusufprojects, 2025-09-13:
en:
can I do this in terminal? I have this problem in terminal and code.
tr:
bunu terminalde de yapabilir miyim? çünkü ben hem terminalde hem code da aynı sorunu yaşıyorum.