platformio/platform-espressif32 #118: Issues with folder name - cannot have spaces?

wegunterjr, 2018-08-30:
```
<address> <filename> [<address> <filename> ...]
esptool write_flash: error: argument <address> <filename>: [Errno 2] No such file or directory: 'C:\\Users\\newman\\Documents\\Projects\\CC_IoT_Grill\\ESP32'
```

The actual folder name is:  CC_IoT_Grill/ESP32 Wifi Controller

martin-ger, 2018-12-24:
Still having this problem with the latest install:
```
Looking for upload port...
Auto-detected: COM4
Uploading .pioenvs\lolin 32\firmware.bin
usage: esptool write_flash [-h] [--flash_freq {keep,40m,26m,20m,80m}]
[--flash_mode {keep,qio,qout,dio,dout}]
[--flash_size FLASH_SIZE]
[--spi-connection SPI_CONNECTION] [--no-progress]
[--verify] [--compress | --no-compress]
<address> <filename> [<address> <filename> ...]
esptool write_flash: error: argument <address> <filename>: [Errno 2] No such file or directory: 'c:\\Users\\Martin'
*** [upload] Error 2
```
My user dir is c:\\Users\\Martin Ger\\

Cannot change this with reasonable effort...
The mentioned line in #121 isn't there any more in "main.py"
Any help welcome...

martin-ger, 2018-12-24:
Okay, changing this line helps - no idea, why a current download of the platform package doesn't load a working config file.

GerdW, 2019-01-19:
Same Problem here. I dont understand how you did solve it. :-D

martin-ger, 2019-01-19:
Changed line 271 of ".platformio\platforms\espressif32\builder\main.py" to:
`env.Append(UPLOADERFLAGS=[image[0], '"%s"' % image[1]])`

ivankravets, 2019-01-21:
What are your OS and locale?

martin-ger, 2019-01-21:
Windows 10, german: de

pokono, 2019-01-22:
MacOS Mojave 10.14.2, English Canada

ivankravets, 2019-01-22:
Could someone provide TeamViewer session to help me with debugging of this issue? I can't reproduce it on my VM machines. Please contact me@ikravets.com. Thanks!

MrMauro, 2019-02-15:
MacOS Mojave 10.14.2, English Canada

Changing:
    env.Append(UPLOADERFLAGS=[image[0], image[1]])
To:
    env.Append(UPLOADERFLAGS=[image[0], '"%s"' % image[1]])
as @martin-ger suggested, worked for me.

ivankravets, 2019-02-18:
@MrMauro thank you so much for the provided remote session. I've just fixed this issue, it's our bug.

Please re-test with the latest upstream version https://docs.platformio.org/en/latest/platforms/espressif32.html#upstream

commit 5883c100 (2019-02-18): Fix issue with uploading for ESP-IDF when project path contains space // Resolve #118
closed 2019-02-18 by ivankravets