platformio/platform-espressif32 #121: Two project examples does not work with ESP32 WROVER KIT

Navis-Raven, 2018-09-03:
First one: 
I've imported the arduino blinky, and uploaded to ESP-32-WROVER however no led blinks at the result.

Second one:
I tried to do the same with the hello world example project with ESP-IDF however it fails into uploading, it gave me this:

````
> Executing task in folder 180903-182438-espidf-hello-world: C:\Users\charafeddine\.platformio\penv\Scripts\platformio.exe run --target upload <

Processing esp wrover kit (platform: espressif32; board: esp-wrover-kit; framework: espidf)
-----------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
PLATFORM: Espressif 32 > Espressif ESP-WROVER-KIT
SYSTEM: ESP32 240MHz 320KB RAM (4MB Flash)
DEBUG: CURRENT(ftdi) ON-BOARD(ftdi) EXTERNAL(esp-prog, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny)
Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF MODES: FINDER(chain) COMPATIBILITY(soft)
Collected 0 compatible libraries
Scanning dependencies...
No dependencies
Compiling .pioenvs\esp wrover kit\src\main.o
Generating partitions .pioenvs\esp wrover kit\partitions.bin
Generating LD script .pioenvs\esp wrover kit\esp32_out.ld
Compiling .pioenvs\esp wrover kit\bootloader\bootloader_start.o
Compiling .pioenvs\esp wrover kit\bootloader\flash_qio_mode.o
Compiling .pioenvs\esp wrover kit\bootloaderSupport\src\bootloader_clock.o
Compiling .pioenvs\esp wrover kit\bootloaderSupport\src\bootloader_flash.o
Compiling .pioenvs\esp wrover kit\bootloaderSupport\src\bootloader_random.o
Compiling .pioenvs\esp wrover kit\bootloaderSupport\src\bootloader_sha.o
Compiling .pioenvs\esp wrover kit\bootloaderSupport\src\efuse.o
Compiling .pioenvs\esp wrover kit\bootloaderSupport\src\esp_image_format.o
Compiling .pioenvs\esp wrover kit\bootloaderSupport\src\flash_encrypt.o
Compiling .pioenvs\esp wrover kit\bootloaderSupport\src\flash_partitions.o
Compiling .pioenvs\esp wrover kit\bootloaderSupport\src\secure_boot.o
Compiling .pioenvs\esp wrover kit\bootloaderSupport\src\secure_boot_signatures.o
Compiling .pioenvs\esp wrover kit\bootloaderLog\log.o
Compiling .pioenvs\esp wrover kit\bootloaderSPIFlash\spi_flash_rom_patch.o
Compiling .pioenvs\esp wrover kit\bootloaderMicroEcc\micro-ecc\uECC.o
Archiving .pioenvs\esp wrover kit\libbootloaderSupport.a
C:\Users\charafeddine\.platformio\packages\framework-espidf\components\log\log.c: In function 'esp_log_buffer_hex_internal':
C:\Users\charafeddine\.platformio\packages\framework-espidf\components\log\log.c:378:16: warning: pointer of type 'void *' used in arithmetic [-Wpointer-arith]
buffer += bytes_cur_line;
^
C:\Users\charafeddine\.platformio\packages\framework-espidf\components\log\log.c: In function 'esp_log_buffer_char_internal':
C:\Users\charafeddine\.platformio\packages\framework-espidf\components\log\log.c:410:16: warning: pointer of type 'void *' used in arithmetic [-Wpointer-arith]
buffer += bytes_cur_line;
^
C:\Users\charafeddine\.platformio\packages\framework-espidf\components\log\log.c: In function 'esp_log_buffer_hexdump_internal':
C:\Users\charafeddine\.platformio\packages\framework-espidf\components\log\log.c:464:16: warning: pointer of type 'void *' used in arithmetic [-Wpointer-arith]
buffer += bytes_cur_line;
^
Compiling .pioenvs\esp wrover kit\bootloaderSoc\esp32\cpu_util.o
Archiving .pioenvs\esp wrover kit\libbootloaderLog.a
Compiling .pioenvs\esp wrover kit\bootloaderSoc\esp32\rtc_clk.o
Archiving .pioenvs\esp wrover kit\libbootloaderSPIFlash.a
Compiling .pioenvs\esp wrover kit\bootloaderSoc\esp32\rtc_init.o
Compiling .pioenvs\esp wrover kit\bootloaderSoc\esp32\rtc_pm.o
Compiling .pioenvs\esp wrover kit\bootloaderSoc\esp32\rtc_sleep.o
Compiling .pioenvs\esp wrover kit\bootloaderSoc\esp32\rtc_time.o
Archiving .pioenvs\esp wrover kit\libbootloaderMicroEcc.a
Compiling .pioenvs\esp wrover kit\bootloaderSoc\esp32\soc_memory_layout.o
Compiling .pioenvs\esp wrover kit\app_update\esp_ota_ops.o
Archiving .pioenvs\esp wrover kit\libbootloaderSoc.a
Compiling .pioenvs\esp wrover kit\bootloader_support\src\bootloader_clock.o
Compiling .pioenvs\esp wrover kit\bootloader_support\src\bootloader_flash.o
Linking .pioenvs\esp wrover kit\bootloader.elf
Compiling .pioenvs\esp wrover kit\bootloader_support\src\bootloader_random.o
Building .pioenvs\esp wrover kit\bootloader.bin
Compiling .pioenvs\esp wrover kit\bootloader_support\src\bootloader_sha.o
C:\Users\charafeddine\.platformio\packages\framework-espidf\components\app_update\esp_ota_ops.c: In function 'esp_rewrite_ota_data':
[... 772 lines cut ...]
Building .pioenvs\esp wrover kit\firmware.bin
Retrieving maximum program size .pioenvs\esp wrover kit\firmware.elf
Checking size .pioenvs\esp wrover kit\firmware.elf
Memory Usage -> http://bit.ly/pio-memory-usage
DATA:    [          ]   3.0% (used 9980 bytes from 327680 bytes)
PROGRAM: [=         ]  13.2% (used 138572 bytes from 1048576 bytes)
Configuring upload protocol...
AVAILABLE: esp-prog, esptool, ftdi, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny
CURRENT: upload_protocol = esptool
Looking for upload port...
Auto-detected: COM7
Uploading .pioenvs\esp wrover kit\firmware.bin
usage: esptool write_flash [-h] [--flash_freq {keep,40m,26m,20m,80m}]
[--flash_mode {keep,qio,qout,dio,dout}]
[--flash_size FLASH_SIZE]
[--spi-connection SPI_CONNECTION] [--no-progress]
[--verify] [--compress | --no-compress]
<address> <filename> [<address> <filename> ...]
esptool write_flash: error: argument <address> <filename>: [Errno 2] No such file or directory: 'c:\\Users\\charafeddine\\Documents\\PlatformIO\\Projects\\180903-182438-espidf-hello-world\\.pioenvs\\esp'
*** [upload] Error 2
============================================================================================= [ERROR] Took 215.64 seconds =============================================================================================
Le processus du terminal s'est achevé avec le code de sortie 1

Le terminal sera réutilisé par les tâches, appuyez sur une touche pour le fermer.
````

Navis-Raven, 2018-09-03:
This is platform.ini content

```
[env:esp wrover kit]
platform = espressif32
framework = espidf
board = esp-wrover-kit
monitor_speed = 115200

```

Navis-Raven, 2018-09-04:
For the first one the defined LED_BUILTIN number is not correct, I think it should be 0 instead of the default 13

```
/*
 * Blink
 * Turns on an LED on for one second,
 * then off for one second, repeatedly.
 */

#include <Arduino.h>

#ifndef LED_BUILTIN
  // Set LED_BUILTIN if it is not defined by Arduino framework
  #define LED_BUILTIN 13
#endif

void setup()
{
  // initialize LED digital pin as an output.
  pinMode(LED_BUILTIN, OUTPUT);
}

void loop()
{
  // turn the LED on (HIGH is the voltage level)
  digitalWrite(LED_BUILTIN, HIGH);
  // wait for a second
  delay(1000);
  // turn the LED off by making the voltage LOW
  digitalWrite(LED_BUILTIN, LOW);
   // wait for a second
  delay(1000);
}
```

Navis-Raven, 2018-09-06:
To solve the first bug a config file can be added with all espressif 32 boards and their led builtin pin out. 

main.c
```c
#include"arduino.h"
#include"ESP_Config.h"

...
```

ESP_Config
```c
//Choose your board
#define LED_BUILTIN ESP_WROVER_KIT_LED

#define ESP_WROVER_KIT_LED 0
#define ESP32_FEATHER_LED 13

etc...
```

The second and third bird has been referenced here and the solution is to remove the spaces.  
https://community.platformio.org/t/esp32-esp-idf-examples-does-not-upload/4985/4

ivankravets, 2018-09-06:
Please run "Verbose Upload" form PlatformIO IDE for VSCode. You will see detailed upload command and arguments. Maybe, the issue with spaces in environment name.

Navis-Raven, 2018-09-06:
@ivankravets yes the second and third issue come from the spaces in environnement names. This is because the platform.ini is configurated like that in the default example codes for ESP-IDF

Look at here: https://github.com/platformio/platform-espressif32/blob/master/examples/espidf-hello-world/platformio.ini#L16

ivankravets, 2018-09-06:
Hm... All tests are passed. Could you provide an output from `pio run -t upload -v` command?

Navis-Raven, 2018-09-06:
Please look at here: https://github.com/platformio/platform-espressif32/blob/master/examples/espidf-hello-world/platformio.ini#L16

[image]

I don't have my board with me at the moment, but no need, because we already now (and I've tested it behind you in the issue #1883 today), that the bug come from the spaces in environement name

Navis-Raven, 2018-09-06:
> Could you provide an output from pio run -t upload -v command? 
I can do this only tomorrow if you need

ivankravets, 2018-09-06:
See 

* https://travis-ci.org/platformio/platform-espressif32
* https://ci.appveyor.com/project/ivankravets/platform-espressif32/branch/develop

Navis-Raven, 2018-09-06:
For the build it is normal that it is passing, it is the upload which causes trouble, when esptool is called !

```
usage: esptool write_flash [-h] [--flash_freq {keep,40m,26m,20m,80m}]
[--flash_mode {keep,qio,qout,dio,dout}]
[--flash_size FLASH_SIZE]
[--spi-connection SPI_CONNECTION] [--no-progress]
[--verify] [--compress | --no-compress]
<address> <filename> [<address> <filename> ...]
esptool write_flash: error: argument <address> <filename>: [Errno 2] No such file or directory: 'c:\\Users\\gry\\Documents\\PlatformIO\\Projects\\180903-182438-espidf-hello-world\\.pioenvs\\esp'
*** [upload] Error 2
============================================================================================= [ERROR] Took 215.64 seconds =============================================================================================
Le processus du terminal s'est achevé avec le code de sortie 1

Le terminal sera réutilisé par les tâches, appuyez sur une touche pour le fermer.
```

Navis-Raven, 2018-09-06:
@ivankravets look at this phrase in particular
```
No such file or directory: 'c:\\Users\\gry\\Documents\\PlatformIO\\Projects\\180903-182438-espidf-hello-world\\.pioenvs\\esp'
*** [upload] Error 2
```
Esptool doesn't find "esp" file or directory but the full name is not "esp" but "esp wrover kit" the spaces have made esptool crash

Navis-Raven, 2018-09-06:
For the first bug

The problem is that the led builtin pin number vary in function of each board:

And here is the list of all board with their led builtin number: 
https://github.com/espressif/arduino-esp32/search?utf8=✓&q=LED_BUILTIN

And for ESP-WROVER-KIT it's "LED_BUILTIN 0" (for GPIO 0, the red led of the RGB)

It is related with the issue here: https://github.com/espressif/arduino-esp32/issues/74

Navis-Raven, 2018-09-07:
@ivankravets look by yourself
with the ble advertise example with verbose option:

```
"c:\users\charafeddine\.platformio\penv\scripts\python.exe" "C:\Users\charafeddine\.platformio\packages\tool-esptoolpy\esptool.py" --chip esp32 --port "COM7" --baud 921600 --before default_reset --after hard_reset write_flash -z --flash_mode dio --flash_freq 40m --flash_size detect 0x1000 C:\Users\charafeddine\Documents\PlatformIO\Projects\180903-195503-espidf-ble-adv\.pioenvs\esp wrover kit\bootloader.bin 0x8000 C:\Users\charafeddine\Documents\PlatformIO\Projects\180903-195503-espidf-ble-adv\.pioenvs\esp wrover kit\partitions.bin 0x10000 ".pioenvs\esp wrover kit\firmware.bin"
usage: esptool write_flash [-h] [--flash_freq {keep,40m,26m,20m,80m}]
[--flash_mode {keep,qio,qout,dio,dout}]
[--flash_size FLASH_SIZE]
[--spi-connection SPI_CONNECTION] [--no-progress]
[--verify] [--compress | --no-compress]
<address> <filename> [<address> <filename> ...]
esptool write_flash: error: argument <address> <filename>: [Errno 2] No such file or directory: 'C:\\Users\\charafeddine\\Documents\\PlatformIO\\Projects\\180903-195503-espidf-ble-adv\\.pioenvs\\esp'
*** [upload] Error 2
================================================================================================= [ERROR] Took 257.19 seconds =================================================================================================
```

Navis-Raven, 2018-09-11:
@ivankravets can you check the updates on this topics please

ignaworn, 2018-09-26:
I'm having the same issue, here.
The problem is that esptool.py invokes the images files without escaping whitespaces.

```
"c:\users\ignaw\.platformio\penv\scripts\python.exe" "C:\Users\ignaw\.platformio\packages\tool-esptoolpy\esptool.py" --chip esp32 --port "COM5" --baud 921600 --before default_reset --after hard_reset write_flash -z --flash_mode dio
--flash_freq 40m --flash_size detect 0x1000 C:\Users\ignaw\Google Drive\Proyectos\esp32-test\.pioenvs\nodemcu-32s\bootloader.bin 0x8000 C:\Users\ignaw\Google Drive\Proyectos\esp32-test\.pioenvs\nodemcu-32s\partitions.bin 0x10000 ".pioenvs\nodemcu-32s\firmware.bin"usage: esptool write_flash [-h] [--flash_freq {keep,40m,26m,20m,80m}]
[--flash_mode {keep,qio,qout,dio,dout}]
[--flash_size FLASH_SIZE]
[--spi-connection SPI_CONNECTION] [--no-progress]
[--verify] [--compress | --no-compress]
<address> <filename> [<address> <filename> ...]
esptool write_flash: error: argument <address> <filename>: [Errno 2] No such file or directory: 'C:\\Users\\ignaw\\Google'
*** [upload] Error 2

My solution was to manually edit `.platformio\platforms\espressif32\builder\main.py` line 271:
```python
env.Append(UPLOADERFLAGS=[ image[0], "\"%s\"" % image[1]]) 
```

After re-building and uploading:
```
"c:\users\ignaw\.platformio\penv\scripts\python.exe" "C:\Users\ignaw\.platformio\packages\tool-esptoolpy\esptool.py" --chip esp32 --port "COM5" --baud 921600 --before default_reset --after hard_reset write_flash -z --flash_mode dio
--flash_freq 40m --flash_size detect 0x1000 "C:\Users\ignaw\Google Drive\Proyectos\esp32-test\.pioenvs\nodemcu-32s\bootloader.bin" 0x8000 "C:\Users\ignaw\Google Drive\Proyectos\esp32-test\.pioenvs\nodemcu-32s\partitions.bin" 0x10000 .pioenvs\nodemcu-32s\firmware.bin
```

Now whitespaces are between double quotes, and esptool works just fine.

```
Serial port COM5
Connecting........__
Chip is ESP32D0WDQ6 (revision 1)
Features: WiFi, BT, Dual Core
MAC: 30:ae:a4:1a:13:2c
Uploading stub...
Running stub...
Stub running...
Changing baud rate to 921600
Changed.
Configuring flash size...
Auto-detected Flash size: 4MB
Compressed 18112 bytes to 11090...
Wrote 18112 bytes (11090 compressed) at 0x00001000 in 0.1 seconds (effective 1073.3 kbit/s)...
Hash of data verified.
Compressed 3072 bytes to 82...
Wrote 3072 bytes (82 compressed) at 0x00008000 in 0.0 seconds (effective 2457.6 kbit/s)...
Hash of data verified.
Compressed 154336 bytes to 80535...
Wrote 154336 bytes (80535 compressed) at 0x00010000 in 1.1 seconds (effective 1085.0 kbit/s)...
Hash of data verified.
```

Hope this helps.

closed 2018-10-01 by ivankravets

ivankravets, 2018-10-01:
Duplicate of https://github.com/platformio/platform-espressif32/issues/118

Navis-Raven, 2018-10-02:
ok thanks @ivankravets but the first bug is not a duplicate of #118

commit 8a311060 (2018-10-01): Add default "LED_BUILTIN" for ESP Wrover Kit // Resolve #121
commit 9c1eeebc (2018-10-01): Merge branch 'develop' into feature/stage

* develop:
  Bump version to 1.4.0
  Add default "LED_BUILTIN" for ESP Wrover Kit // Resolve #121
  Add "Blink" example for ESP-IDF // Resolve #122
  Fix define value in sdkconfig file
  Update esp-idf package version
  Update sdkconfig file in esp-idf examples
  Update esp-idf build script // Issue #91, #125
  Escape space symbol in the image path // Issue #118, #121

# Conflicts:
#	platform.json