platformio/platform-espressif32 #126: I2C repeated start broken

Erriez, 2018-09-16:
The following example reads zero's when generating an I2C repeated start:

```c++
#include <Arduino.h>
#include <Wire.h>

#define DS3231_ADDR             (0xD0 >> 1)

void setup()
{
    Serial.begin(115200);
    Serial.println();

    Wire.begin();
    Wire.setClock(400000);
}

void loop()
{
    Wire.beginTransmission(DS3231_ADDR);
    Wire.write((uint8_t)0x0);
    Wire.endTransmission(false); // < Generate repeated start with false
    Wire.requestFrom((uint8_t)DS3231_ADDR, (uint8_t)7, true);
    while(Wire.available()) {
        Serial.printf("0x%02X ", Wire.read());
    }
    Serial.println();

    delay(1000);
}
```

Incorrect output of the example above with a repeated start ```Wire.endTransmission(false);```:
```0x00 0x00 0x00 0x00 0x00 0x00 0x00```

Valid data is set on the I2C bus with a repeated start, but the receive data is not read correctly in the ESP32 I2C driver. Screenshot valid I2C signals measured with a logic analyzer:

[image]

Valid ouput when change repeated start to a stop - start condition ```Wire.endTransmission(true);```:
```0x17 0x47 0x21 0x07 0x29 0x07 0x18```

This bug is already fixed in the official Espressif branch:
https://github.com/espressif/arduino-esp32/blob/master/cores/esp32/esp32-hal-i2c.c
https://github.com/espressif/arduino-esp32/blob/master/cores/esp32/esp32-hal-i2c.h
https://github.com/espressif/arduino-esp32/blob/master/libraries/Wire/src/Wire.cpp
https://github.com/espressif/arduino-esp32/blob/master/libraries/Wire/src/Wire.h

Replacing the sources above works. Can someone backport this?

ivankravets, 2018-10-01:
Does it work with upstream version?  http://docs.platformio.org/en/latest/platforms/espressif32.html#stable-and-upstream-versions

Erriez, 2018-10-06:
Unfortunately, it does not work with this upstream branch: It reads only zero's as well.

```
Wire.endTransmission(false); // < Generate repeated start with false reads only zero's
Wire.endTransmission(true); // < Generate new start with true works
```

alborzs, 2018-11-07:
Stumbled upon this issue while trying to find out why my **BNO055** (9DoF sensor) was suddenly returning only zeros after updating to 1.5.0. Is there a workaround for us to use in the meanwhile?

Erriez, 2018-11-08:
> 
> 
> Stumbled upon this issue while trying to find out why my **BNO055** (9DoF sensor) was suddenly returning only zeros after updating to 1.5.0. Is there a workaround for us to use in the meanwhile?

I suggest you to overwrite the existing sources from the ESP32 master branch:
> https://github.com/espressif/arduino-esp32/blob/master/cores/esp32/esp32-hal-i2c.c
> https://github.com/espressif/arduino-esp32/blob/master/cores/esp32/esp32-hal-i2c.h
> https://github.com/espressif/arduino-esp32/blob/master/libraries/Wire/src/Wire.cpp
> https://github.com/espressif/arduino-esp32/blob/master/libraries/Wire/src/Wire.h

ivankravets, 2019-01-10:
Should be fixed in Arduino ESP32 core. See docs how you can switch to development version of this framework https://docs.platformio.org/en/latest/platforms/espressif32.html#using-arduino-framework-with-staging-version

closed 2019-01-10 by ivankravets

glucee, 2019-01-30:
I think the problem is still there, could you check it @ivankravets

ivankravets, 2019-01-30:
Do you use the latest version of dev/platform? Please open PlatformIO IDE Terminal and type `pio update`.

ivankravets, 2019-01-30:
If you still experience this issue, please report to https://github.com/espressif/arduino-esp32/issues