platformio/platform-espressif32 #127: Erase Flash Sketch + Wifi Settings

j0hncc, 2018-09-21:
In arduino IDE there is option for 

[image]

What is the corresponding way to do those in platformio?  I am using VSCode IDE

Thanks!

vshymanskyy, 2018-10-03:
@ivankravets same comment. It could be implemented by having different upload targets

vshymanskyy, 2018-10-03:
@ivankravets this is also needed for ESP8266

atatistcheff, 2018-11-23:
Also having this issue.  I have to go back to the Arduino IDE to clear these settings on the Wemos D1 (ESP-8266).

closed 2018-11-24 by ivankravets

ivankravets, 2018-11-24:
See https://docs.platformio.org/en/latest/platforms/espressif32.html#erase-flash

ivankravets, 2018-11-24:
Please note that you will need to use [upstream version](https://docs.platformio.org/en/latest/platforms/espressif32.html#upstream) of dev/platformio while it has not been released yet.

richard-scott, 2019-04-02:
I've just tried this with Platformio-3.6.6 in VS Code and now I get this when trying to upload:
```
AVAILABLE: espota, esptool
CURRENT: upload_protocol = esptool
Looking for upload port...
Auto-detected: COM5
Uploading C:\Users\rscott\AppData\Local\Temp\pio-build\c70d26bfa2\build\nodemcuv2\firmware.bin
warning: espcomm_sync failed
error: espcomm_open failed
error: espcomm_upload_mem failed
*** [upload] Error 2
```

Does anyone have a link to the fix as my Google Foo isn't working for me right now ;-(

NavadeepGaneshU, 2020-04-12:
PS C:\Users\HP1> esptool.py erase_flash
esptool.py v2.8
Found 1 serial ports
Serial port COM4
Connecting........_____....._____....._____....._____....._____....._____....._____
COM4 failed to connect: Failed to connect to Espressif device: Invalid head of packet (0x48)

A fatal error occurred: Could not connect to an Espressif device on any of the 1 available serial ports.
PS C:\Users\HP1>

Unable to erase the flash at all. ESP-32 CAM board
Can anyone help me here?

richard-scott, 2020-04-15:
Do you have GPIO0 connected to GND before connecting 5V?
This puts the board into Flashing mode.

[image]

NavadeepGaneshU, 2020-04-15:
Oh, Yes. I realise i made a mistake there ! . But, even after correcting, it shows the below error, Giving default partition shows sketch too large and my partition scheme is huge app (3MB no OTA). 
----------------------------------------------------------------------------------------
Warning: Could not auto-detect Flash size (FlashID=0x0, SizeID=0x0), defaulting to 4MB
Compressed 8192 bytes to 47...

Writing at 0x0000e000... (100 %)
Wrote 8192 bytes (47 compressed) at 0x0000e000 in 0.0 seconds (effective 3449.3 kbit/s)...

A fatal error occurred: Timed out waiting for packet header
A fatal error occurred: Timed out waiting for packet header

richard-scott, 2020-04-16:
Maybe this to your "[env]" in platformio.ini

```
board_build.partitions = min_spiffs.csv
```

Look in here for other partiton csv files (assuming your on Windows):
C:\Users\\%USERNAME%\\.platformio\packages\framework-arduinoespressif32\tools\partitions

Can you post your platformio.ini so we can see what you are using to build with?

NavadeepGaneshU, 2020-04-16:
platformio config file : 
-----------------------------------------------------------------------------
[platformio]
src_dir = .
lib_dir = ../..

[env:nano32]
platform = espressif32@1.9.0
board = nano32
framework = arduino
board_build.partitions = partition-table.csv
build_flags =
    -DNO_GLOBAL_EEPROM
    -DDEBUG_EEPROM32_ROTATE_PORT=Serial

NavadeepGaneshU, 2020-04-16:
platformio config  file :+1: 
-------------------------------------------------------------------------------------------
`[platformio]
src_dir = .
lib_dir = ../..

[env:nano32]
platform = espressif32@1.9.0
board = nano32
framework = arduino
board_build.partitions = partition-table.csv
build_flags =
    -DNO_GLOBAL_EEPROM
    -DDEBUG_EEPROM32_ROTATE_PORT=Serial
`

richard-scott, 2020-04-20:
Why do you set your own partitions?
It should work fine without that?

NavadeepGaneshU, 2020-04-20:
Sorry, I didn't understand that . 
So, how can I fix this now ?

richard-scott, 2020-04-20:
comment out this line in your platformio.ini:

```
board_build.partitions = partition-table.csv
```

NavadeepGaneshU, 2020-04-20:
Tried it and I see the same error again. 

There were 3 files with name "platformio.ini" in 
D: >> Arduino >> Libraries >> EEPROM32_Rotate>> Examples>> basic , dump and info locations. 

Commented that line in all 3 files.

Can I get some of your other contact details so that I can reach to you soon to fix this issue? 
E-Mail me at navadeepganesh.ngu@gmail.com 
Please help me In fixing this issue.

richard-scott, 2020-04-21:
That folder structuer is wrong for PlatformIO.
If I were you, I'd create a new project in PlatformIO and base that on the 'MakerAsia Nano32' board.

Don't add any new code to that, but try and build and upload what the IDE gives you as a template.
Only once that works, start adding more custom code.

I'm sorry, but I can't provide help outside of this issue.

NavadeepGaneshU, 2020-04-23:
Ok, thank you. 
So ,if I try the same (to upload code) on another PC (Other Arduino IDE), it should work fine right ? 
There isn't any issue on ESP32 CAM side right ?

nilipta, 2020-10-24:
![https://user-images.githubusercontent.com/11875156/45855951-20165780-bd18-11e8-9ffb-4f7c739a8dfd.png]

Can you please teach me why there are 3 options are there to erase flash ?
What are the benefits to use them ?

commit faf9a065 (2018-11-24): Implement "erase" target // Resolve #127
commit a7b1fe6d (2018-11-24): Merge branch 'develop' into feature/stage

* develop:
  Implement "erase" target // Resolve #127
  Automatically generate known debug tools and upload protocols
  Add new boards: Pycom LoPy & LoPy4 (#134)
  Force forward slash for paths for openocd-esp32, new board D-duino-32 (#139)