platformio/platform-espressif32 #130: Support software breakpoints in ESP32

NikhilMJ, 2018-10-11:
Starting with ESP-IDF v3.1, software breakpoints are being supported by the esp-idf. 
Seems that the arduino core has also migrated to this latest version.

Seems like currently are able to set only 2 breakpoints. These are hardware breakpoints. Get the following warning when trying to set the third break point. 
> Warn : xtensa_add_breakpoint: max slot reached, slot=2
[image]

Was support for software breakpoints added to pio for vsc?

ivankravets, 2018-10-11:
What is your OS? Could you try the latest version of openOCD from https://github.com/espressif/openocd-esp32/releases

Please replace files in `~/.platformio/packages/tool-openocd-esp32` and keep `package.json` without changes.

Does it work?

NikhilMJ, 2018-10-12:
It didn't. I now get this error. 
** OpenOCD init failed **
shutdown command invoked

As an fyi, I am using the wrover kit with the below ini config: 
```[env:esp-wrover-kit]
platform = espressif32
board = esp-wrover-kit
framework = arduino
upload_port=COM5
debug_port=COM4```

ivankravets, 2018-10-12:
Please remove this `debug_port=COM4````

NikhilMJ, 2018-10-14:
Apologies for the delayed response. I was messing around with the settings when I sent that ini config. Never got back to getting this working.

ivankravets, 2018-10-29:
Please open PIO IDE and upgrade PIO Core to 3.6.1. Also, please update all packages using "VSCode > Left sidebar > PlatformIO(icon) > Update packages..."

Do software breakpoints work now?

ivankravets, 2018-11-05:
Please reopen if you still need help

NikhilMJ, 2018-11-18:
Not yet Ivan. Got this running again. Am able to set a breakpoint and then step from there, but still can't set additional software breakpoints. I updated the core and the various packages. I'm on pio version 3.6.2a2.

Some screenshots of error: 
[image]

ivankravets, 2018-11-24:
I've just tested software breakpoints and they work for me. Please try the next:

1. Launch debug session
2. Do "Step Over" to `initArduino();`
3. Set breakpoint on `void loopTask(void *pvParameters)`, we need to stop when `loopTask` will be inited
4. Now, you can set as many breakpoints as you need in arduino code. They should work.

[image]

closed 2019-01-10 by ivankravets