platformio/platform-espressif32 #131: DOIT ESP32 Devkit V1 upload timeout in less than 1 second!

ERROPiX, 2018-10-16:
Hi,

I am trying to use PlatformIO IDE in VS Code, I have already checked for updates for boards and platforms etc. all is up to date!

Uploading code to Arduino UNO works fine, but DOIT esp32 devkit doesn't work.
I was using this board in both Arduino IDE or with VS Code Arduino extension without problems.
I also tried uploading to it via the included esptool.py directly and it worked.

## PlatformIO.ini (default with no changes)
```
[env:esp32doit-devkit-v1]
platform = espressif32
board = esp32doit-devkit-v1
framework = arduino
```
## Upload Task Output
```
> Executing task in folder ESP32_Blink: C:\Users\errouaguy\.platformio\penv\Scripts\platformio.exe run --target upload <

Processing esp32doit-devkit-v1 (platform: espressif32; board: esp32doit-devkit-v1; framework: arduino)
----------------------------------------------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
PLATFORM: Espressif 32 > DOIT ESP32 DEVKIT V1
SYSTEM: ESP32 240MHz 320KB RAM (4MB Flash)
DEBUG: CURRENT(esp-prog) EXTERNAL(esp-prog, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny)
Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF MODES: FINDER(chain) COMPATIBILITY(soft)
Collected 24 compatible libraries
Scanning dependencies...
No dependencies
Retrieving maximum program size .pioenvs\esp32doit-devkit-v1\firmware.elf
Checking size .pioenvs\esp32doit-devkit-v1\firmware.elf
Memory Usage -> http://bit.ly/pio-memory-usage
DATA:    [          ]   4.2% (used 13884 bytes from 327680 bytes)
PROGRAM: [=         ]  13.6% (used 178020 bytes from 1310720 bytes)
Configuring upload protocol...
AVAILABLE: esp-prog, esptool, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny
CURRENT: upload_protocol = esptool
Looking for upload port...
Auto-detected: COM5
Uploading .pioenvs\esp32doit-devkit-v1\firmware.bin
Serial port COM5
Connecting...
Traceback (most recent call last):
File "C:\Users\errouaguy\.platformio\packages\tool-esptoolpy\esptool.py", line 2849, in <module>
_main()
File "C:\Users\errouaguy\.platformio\packages\tool-esptoolpy\esptool.py", line 2842, in _main
main()
File "C:\Users\errouaguy\.platformio\packages\tool-esptoolpy\esptool.py", line 2550, in main
esp.connect(args.before)
File "C:\Users\errouaguy\.platformio\packages\tool-esptoolpy\esptool.py", line 444, in connect
last_error = self._connect_attempt(mode=mode, esp32r0_delay=False)
File "C:\Users\errouaguy\.platformio\packages\tool-esptoolpy\esptool.py", line 424, in _connect_attempt
self.sync()
File "C:\Users\errouaguy\.platformio\packages\tool-esptoolpy\esptool.py", line 363, in sync
timeout=SYNC_TIMEOUT)
File "C:\Users\errouaguy\.platformio\packages\tool-esptoolpy\esptool.py", line 306, in command
self.write(pkt)
File "C:\Users\errouaguy\.platformio\packages\tool-esptoolpy\esptool.py", line 269, in write
self._port.write(buf)
File "c:\users\errouaguy\.platformio\penv\lib\site-packages\serial\serialwin32.py", line 323, in write
raise writeTimeoutError
serial.serialutil.SerialTimeoutException: Write timeout
*** [upload] Error 1
====================================================== [ERROR] Took 6.53 seconds ======================================================
The terminal process terminated with exit code: 1

Terminal will be reused by tasks, press any key to close it.
```
## Screenshot
[image]

ERROPiX, 2018-10-17:
Hi @ivankravets,

Just to give some more details, this is the manual command I entered in the terminal, and the upload was successful

```
python C:/Users/errouaguy/.platformio/packages/tool-esptoolpy/esptool.py
    --chip esp32 
    --port COM5 
    --baud 921600 
    --before default_reset 
    --after hard_reset 
    
    write_flash
        -z 
        --flash_mode dio 
        --flash_freq 80m 
        --flash_size detect 
        0x1000 C:/Users/errouaguy/.platformio/packages/framework-arduinoespressif32/tools/sdk/bin/bootloader_dio_80m.bin 
        0x8000 c:/Users/errouaguy/Documents/PlatformIO/Projects/ESP32_Blink/.pioenvs/esp32doit-devkit-v1/partitions.bin 
        0xe000 C:/Users/errouaguy/.platformio/packages/framework-arduinoespressif32/tools/partitions/boot_app0.bin 
        0x10000 .pioenvs/esp32doit-devkit-v1/firmware.bin
```

Screenshot of results

[image]

ivankravets, 2018-10-17:
Could you try in PIO IDE Terminal `pio run -t upload -v`?

ERROPiX, 2018-10-17:
Hi @ivankravets 
If you can just point where to add this command to run after compilation

ivankravets, 2018-10-17:
Just click on Terminal icon in the bottom toolbar and type
```
pio run -t upload -v
```

ERROPiX, 2018-10-17:
Here is it the output of `pio run -t upload -v` command:
```
[ESP32_Blink] pio run -t upload -v
Processing esp32doit-devkit-v1 (platform: espressif32; board: esp32doit-devkit-v1; framework: arduino)
----------------------------------------------------------------------------------------------------------------------------------
PLATFORM: Espressif 32 > DOIT ESP32 DEVKIT V1
SYSTEM: ESP32 240MHz 320KB RAM (4MB Flash)
DEBUG: CURRENT(esp-prog) EXTERNAL(esp-prog, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny)
Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF MODES: FINDER(chain) COMPATIBILITY(soft)
Collected 24 compatible libraries
Scanning dependencies...
No dependencies
<lambda>(["checkprogsize"], [".pioenvs\esp32doit-devkit-v1\firmware.elf"])
MethodWrapper(["checkprogsize"], [".pioenvs\esp32doit-devkit-v1\firmware.elf"])
Memory Usage -> http://bit.ly/pio-memory-usage
DATA:    [          ]   4.2% (used 13884 bytes from 327680 bytes)
PROGRAM: [=         ]  13.6% (used 178020 bytes from 1310720 bytes)
.pioenvs\esp32doit-devkit-v1\firmware.elf  :
section                                                         size         addr
.rtc.text                                                          0   1074528256
.rtc_noinit                                                        0   1342177792
.iram0.vectors                                                  1024   1074266112
.iram0.text                                                    40744   1074267136
.dram0.data                                                     9172   1073479680
.noinit                                                            0   1073488852
.dram0.bss                                                      4712   1073488856
.flash.rodata                                                  34012   1061158944
.flash.text                                                    93068   1074593816
.comment                                                         419            0
.xtensa.info                                                      56            0
.xt.lit._ZN9IPAddressD5Ev                                          0            0
.xt.prop._ZN9IPAddressD5Ev                                         0            0
.xt.prop._ZN9IPAddressD2Ev                                        36            0
.xt.prop._ZN9IPAddressD0Ev                                        36            0
.xt.prop._ZTV9IPAddress                                           12            0
.xt.lit._ZN5Print5writeEPKc                                        8            0
.xt.prop._ZN5Print5writeEPKc                                      60            0
.xt.prop._ZTISt9exception                                         12            0
.xt.prop._ZTISt9bad_alloc                                         12            0
.xt.prop._ZTVN10__cxxabiv117__class_type_infoE                    12            0
.xt.lit._ZNK9__gnu_cxx24__concurrence_lock_error4whatEv            8            0
.xt.lit._ZNK9__gnu_cxx26__concurrence_unlock_error4whatEv          8            0
.xt.lit._ZN9__gnu_cxx24__concurrence_lock_errorD5Ev                8            0
.xt.lit._ZN9__gnu_cxx26__concurrence_unlock_errorD5Ev              8            0
.xt.lit._ZN9__gnu_cxx7__mutex4lockEv                               8            0
.xt.lit._ZN9__gnu_cxx13__scoped_lockD5Ev                           8            0
.xt.prop._ZNK9__gnu_cxx24__concurrence_lock_error4whatEv          48            0
.xt.prop._ZNK9__gnu_cxx26__concurrence_unlock_error4whatEv        48            0
.xt.prop._ZN9__gnu_cxx24__concurrence_lock_errorD5Ev              12            0
.xt.prop._ZN9__gnu_cxx26__concurrence_unlock_errorD5Ev            12            0
.xt.prop._ZN9__gnu_cxx7__mutex4lockEv                             60            0
.xt.prop._ZN9__gnu_cxx13__scoped_lockD5Ev                         12            0
.xt.prop._ZN9__gnu_cxx24__concurrence_lock_errorD2Ev              36            0
.xt.prop._ZN9__gnu_cxx26__concurrence_unlock_errorD2Ev            36            0
.xt.prop._ZN9__gnu_cxx24__concurrence_lock_errorD0Ev              36            0
.xt.prop._ZN9__gnu_cxx26__concurrence_unlock_errorD0Ev            36            0
.xt.prop._ZN9__gnu_cxx13__scoped_lockD2Ev                         72            0
.xt.prop._ZTIN9__gnu_cxx24__concurrence_lock_errorE               12            0
.xt.prop._ZTIN9__gnu_cxx26__concurrence_unlock_errorE             12            0
.xt.prop._ZTVN9__gnu_cxx24__concurrence_lock_errorE               12            0
.xt.prop._ZTVN9__gnu_cxx26__concurrence_unlock_errorE             12            0
.xt.prop._ZTVSt9bad_alloc                                         12            0
.xt.prop._ZTVN10__cxxabiv120__si_class_type_infoE                 12            0
.debug_line                                                   557152            0
.debug_info                                                  1175446            0
.debug_abbrev                                                 148960            0
.debug_aranges                                                 26704            0
.debug_frame                                                   67616            0
.debug_loc                                                    468255            0
[... 10 lines cut ...]
Connecting...
Traceback (most recent call last):
File "C:\Users\errouaguy\.platformio\packages\tool-esptoolpy\esptool.py", line 2849, in <module>
_main()
[... cut ...]
File "C:\Users\errouaguy\.platformio\packages\tool-esptoolpy\esptool.py", line 306, in command
self.write(pkt)
File "C:\Users\errouaguy\.platformio\packages\tool-esptoolpy\esptool.py", line 269, in write
self._port.write(buf)
File "c:\users\errouaguy\.platformio\penv\lib\site-packages\serial\serialwin32.py", line 323, in write
raise writeTimeoutError
serial.serialutil.SerialTimeoutException: Write timeout
*** [upload] Error 1
=================================================== [ERROR] Took 6.36 seconds ===================================================
[ESP32_Blink]
```

ERROPiX, 2018-10-17:
Hi,

I even took that command from the debug output `"c:\users\errouaguy\.platformio\penv\scripts\python.exe" "C:\Users\errouaguy\.platformio\packages\tool-esptoolpy\esptool.py" --chip esp32 --port "COM5" --baud 921600 --before default_reset --after hard_reset write_flash -z --flash_mode dio --flash_freq 80m --flash_size detect 0x1000 C:\Users\errouaguy\.platformio\packages\framework-arduinoespressif32\tools\sdk\bin\bootloader_dio_80m.bin 0x8000 C:\Users\errouaguy\Documents\PlatformIO\Projects\ESP32_Blink\.pioenvs\esp32doit-devkit-v1\partitions.bin 0xe000 C:\Users\errouaguy\.platformio\packages\framework-arduinoespressif32\tools\partitions\boot_app0.bin 0x10000 .pioenvs\esp32doit-devkit-v1\firmware.bin` and it worked fine

ERROPiX, 2018-10-17:
The issue is not in the esptool command, it's something else related to how PlatformIO call that command, which seems to be weird

ERROPiX, 2018-10-17:
Hi,

I have some updates about the bug, when I created a custom script
```
Import("env")

def on_upload(source, target, env):
    cmd  = "python C:/Users/errouaguy/.platformio/packages/tool-esptoolpy/esptool.py --chip esp32 --port COM5 --baud 921600 --before default_reset --after hard_reset write_flash -z --flash_mode dio --flash_freq 80m --flash_size detect 0x1000 C:/Users/errouaguy/.platformio/packages/framework-arduinoespressif32/tools/sdk/bin/bootloader_dio_80m.bin 0x8000 c:/Users/errouaguy/Documents/PlatformIO/Projects/ESP32_Blink/.pioenvs/esp32doit-devkit-v1/partitions.bin 0xe000 C:/Users/errouaguy/.platformio/packages/framework-arduinoespressif32/tools/partitions/boot_app0.bin 0x10000 .pioenvs/esp32doit-devkit-v1/firmware.bin"

    print ""
    print "---------------------- CUSTOM SCRIPT -------------------------"
    env.Execute( cmd )
    print "---------------------- CUSTOM SCRIPT -------------------------"

env.Replace(UPLOADCMD=on_upload)
```

I got the same error as above, I added these "CUSTOM SCRIPT" print to be sure it used my script
The same command run fine when executed from the platformio terminal

any ideas?

ERROPiX, 2018-10-26:
Hi @ivankravets 

I have made a solution to my problem using two vs code extensions and some custom powershell scripts, Basically I only used the `pio run` command to compile the code then execute the above command (from previous comments)

- I used [terminals manager](https://marketplace.visualstudio.com/items?itemName=fabiospampinato.vscode-terminals) to create two terminal wrappers

- Then used the [commands extension](https://marketplace.visualstudio.com/items?itemName=fabiospampinato.vscode-commands) to help me add two icons to the status bar that execute the previously said terminal wrappers

If anyone have this problem and want me to share the scripts and more details, please tag me

ivankravets, 2019-01-10:
Could you reproduce this issue with the latest upstream version https://docs.platformio.org/en/latest/platforms/espressif32.html#upstream ?

ERROPiX, 2019-01-10:
How to use that version in my vscode, should I update or change some settings?

ERROPiX, 2019-01-10:
I got the same error, I am think this related to python script and nodejs on windows, I didn't yet tested on linux

ERROPiX, 2019-01-10:
The issue, it raise timeout error without waiting for connection, as I explained before, I used the commands that I got from pio verbose debug and run them via another tool and got success

If you are interested in fixing this issue, I can give you remote access to test it

ivankravets, 2019-01-10:
I've just updated esptool.py to 2.6.0 version. Please run `pio update` if you are already on an upstream version of dev/platform. Does it work now?

ERROPiX, 2019-01-10:
tool-esptoolpy @ 1.20600.0 has been successfully installed!

is this the right version?

ivankravets, 2019-01-10:
Yes

ERROPiX, 2019-01-10:
No luck :(

ivankravets, 2019-01-10:
Could you comment `upload_port = COM5` line in `platformio.ini`? Please check that other tools don't use com port.

ERROPiX, 2019-01-10:
If I comment out the port, it will do search and find the port com5 then the error again

gerritv, 2019-07-05:
Although my symptoms are slightly different, I can get the Upload to work by pressing BOOT then EN on the Devkit board at the appropriate time (i.e. once esptool.py has found the com port). A bit hit and miss but it gets it most of the time.

bloobird0, 2019-10-28:
@ERROPiX 
I currently have the same problem. Could you please tell me if you found a solution to this problem? 
If not, could you please share your workaround to use another terminal for uploading the project?

Thanks in advance for your reply

ERROPiX, 2019-10-29:
This is not a PlatformIO issue, it is an issue in some third-party boards, is described at the end of this section https://github.com/espressif/esptool/wiki/ESP32-Boot-Mode-Selection#automatic-bootloader

> Adding a 100nF (or higher) value capacitor between EN pin and GND may make automatic reset more reliable.

It worked for me as said

bloobird0, 2019-11-03:
It's no problem with the board in my case:
* I tried the capacitor on the very first M5 model I have and also on 25 new ones. Several values on RST pin fromn 100nF to 10uF. Again, the same boards can be uploaded on the same PC using the same esptool version from Arduino IDE. It does not work on the PC using PIO although it works on 2 other PCs. There must be something wrong on my PC with PIO but I can't figure out what.

ivankravets, 2019-11-03:
@bloobird0 does PIO detect the right upload port? `pio run upload -v`.

ERROPiX, 2019-11-03:
@bloobird0 the capacitor should be between these pins **EN** and **GND**, check the pinout for your board

closed 2022-08-10 by stale[bot]