platformio/platform-espressif32 #133: Upload failure when trying to upload esp-idf to esp32

wegunterjr, 2018-10-23:
Linux OS - compiles fine, and I can connect serial, but when I try to upload, it gets the following error.
I have also deleted .pioenvs and started the build again to make sure and get the same error. I cannot upload any compiled code.

```
Uploading .pioenvs/esp-wrover16/firmware.bin
sh: 1: Syntax error: "(" unexpected
*** [upload] Error 2
```

Maybe it has to do with python on Linux - bash vs sh ?

ivankravets, 2018-10-23:
Did you make some changes in your `.bashrc` script?

ivankravets, 2019-01-10:
Could you reproduce this issue with the latest upstream version https://docs.platformio.org/en/latest/platforms/espressif32.html#upstream?

valeros, 2020-05-25:
Please reopen if you still need any help.

closed 2020-05-25 by valeros

oxihormesis, 2020-09-22:
I'm having this same error

`Uploading .pio/build/custom32_LEDPIN_16/firmware.bin
sh: 1: Syntax error: "(" unexpected
*** [upload] Error 2`

Tried many different environments. Project builds fine then I get this when I upload