platformio/platform-espressif32 #136: ESP32 fails upload due to incorrect image.base_address option value

yoofie, 2018-11-08:
Hello, 
I am using ESP-Prog and trying to upload a simple sketch to `esp32doit-devkit-v1`. I running into similar issues found in [this thread](https://community.platformio.org/t/esp32-pio-unified-debugger/4541) .

I get the following error:

```
DATA:    [          ]   4.2% (used 13884 bytes from 327680 bytes)
PROGRAM: [=         ]  13.6% (used 178016 bytes from 1310720 bytes)
Configuring upload protocol...
AVAILABLE: esp-prog, esptool, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa
CURRENT: upload_protocol = esp-prog
Uploading .pioenvs\esp32doit-devkit-v1\firmware.bin
Open On-Chip Debugger 0.10.0-dev (2018-11-05-04:10)
Licensed under GNU GPL v2
For bug reports, read
http://openocd.org/doc/doxygen/bugs.html
none separate
adapter speed: 20000 kHz
esp32 interrupt mask on
Error: libusb_open() failed with LIBUSB_ERROR_NOT_SUPPORTED
Info : ftdi: if you experience problems at higher adapter clocks, try the command "ftdi_tdo_sample_edge falling"
Info : clock speed 20000 kHz
Info : JTAG tap: esp32.cpu0 tap/device found: 0x120034e5 (mfg: 0x272 (Tensilica), part: 0x2003, ver: 0x1)
Info : JTAG tap: esp32.cpu1 tap/device found: 0x120034e5 (mfg: 0x272 (Tensilica), part: 0x2003, ver: 0x1)
Info : Target halted. PRO_CPU: PC=0x40000400 (active)    APP_CPU: PC=0x40000400
Info : JTAG tap: esp32.cpu0 tap/device found: 0x120034e5 (mfg: 0x272 (Tensilica), part: 0x2003, ver: 0x1)
Info : JTAG tap: esp32.cpu1 tap/device found: 0x120034e5 (mfg: 0x272 (Tensilica), part: 0x2003, ver: 0x1)
Info : esp32: Debug controller 0 was reset (pwrstat=0x5F, after clear 0x0F).
Info : esp32: Core 0 was reset (pwrstat=0x5F, after clear 0x0F).
Info : Target halted. PRO_CPU: PC=0x5000004B (active)    APP_CPU: PC=0x00000000
Info : esp32: Core 0 was reset (pwrstat=0x1F, after clear 0x0F).
Info : esp32: Debug controller 1 was reset (pwrstat=0x5F, after clear 0x0F).
Info : esp32: Core 1 was reset (pwrstat=0x5F, after clear 0x0F).
Info : Target halted. PRO_CPU: PC=0x40000400 (active)    APP_CPU: PC=0x40000400
** Programming Started **
auto erase enabled
Info : Target halted. PRO_CPU: PC=0x4009171A (active)    APP_CPU: PC=0x40000400
Info : Flash mapping 0: 0x10020 -> 0x3f400020, 33 KB
Info : Flash mapping 1: 0x20018 -> 0x400d0018, 90 KB
Info : Target halted. PRO_CPU: PC=0x4009171A (active)    APP_CPU: PC=0x40000400
Info : Auto-detected flash size 4096 KB
Info : Using flash size 4096 KB
Info : Target halted. PRO_CPU: PC=0x4009171A (active)    APP_CPU: PC=0x40000400
Info : Target halted. PRO_CPU: PC=0x4009171A (active)    APP_CPU: PC=0x40000400
wrote 180224 bytes from file .pioenvs\esp32doit-devkit-v1\firmware.bin in 2.804591s (62.754 KiB/s)
** Programming Finished **
** Verify Started **
Info : Target halted. PRO_CPU: PC=0x4009171A (active)    APP_CPU: PC=0x40000400
read 178160 bytes from file .pioenvs\esp32doit-devkit-v1\firmware.bin and flash bank 0 at offset 0x00010000 in 0.893627s (194.695 KiB/s)
contents match
** Verified OK **
Info : JTAG tap: esp32.cpu0 tap/device found: 0x120034e5 (mfg: 0x272 (Tensilica), part: 0x2003, ver: 0x1)
Info : JTAG tap: esp32.cpu1 tap/device found: 0x120034e5 (mfg: 0x272 (Tensilica), part: 0x2003, ver: 0x1)
Info : esp32: Debug controller 0 was reset (pwrstat=0x5F, after clear 0x0F).
Info : esp32: Core 0 was reset (pwrstat=0x5F, after clear 0x0F).
Info : Target halted. PRO_CPU: PC=0x5000004B (active)    APP_CPU: PC=0x00000000
Info : esp32: Core 0 was reset (pwrstat=0x1F, after clear 0x0F).
Info : esp32: Debug controller 1 was reset (pwrstat=0x5F, after clear 0x0F).
Info : esp32: Core 1 was reset (pwrstat=0x5F, after clear 0x0F).
Info : Target halted. PRO_CPU: PC=0x40000400 (active)    APP_CPU: PC=0x40000400
** Programming Started **
auto erase enabled
Error: Invalid command argument
image.base_address option value ('ramework-arduinoespressif32') is not valid
embedded:startup.tcl:454: Error: ** Programming Failed **
in procedure 'program_esp32'
in procedure 'program_error' called at file "C:\Users\mazeb\.platformio\packages\tool-openocd-esp32/share/openocd/scripts/target/esp32.cfg", line 152
at file "embedded:startup.tcl", line 454
*** [upload] Error 1
```

I noticed `image.base_address option value ('ramework-arduinoespressif32') is not valid` has a incorrect name. It looks like people [ran into this issue back on July 20th](https://community.platformio.org/t/esp32-pio-unified-debugger/4541/12) and was fixed. However, it looks like its back and I couldn't find the specific fix/commit that addresses this issue.

I am currently using the following:

```
PS C:\Users\mazeb\Documents\PlatformIO\Projects\Test1> platformio update
Updating tool-scons                      @ 2.20501.4      [Up-to-date]
Updating tool-unity                      @ 1.20403.0      [Up-to-date]
Updating contrib-pysite                  @ 0.3.2          [Up-to-date]
Updating contrib-piohome                 @ 1.0.6          [Up-to-date]
Updating tool-pioplus                    @ 1.5.2          [Up-to-date]

Platform Espressif 32
--------
Updating espressif32                     @ 1.5.0          [Up-to-date]
Updating toolchain-xtensa32              @ 2.50200.80     [Up-to-date]
Updating tool-openocd-esp32              @ 1.1000.20181105[Up-to-date]
Updating tool-mkspiffs                   @ 2.230.0        [Up-to-date]
Updating framework-arduinoespressif32    @ 2.10000.0      [Up-to-date]
Updating tool-espotapy                   @ 1.1.2          [Up-to-date]
Updating tool-esptoolpy                  @ 1.20500.0      [Up-to-date]

Library Manager
===============
Updating FreeRTOS_ARM                    @ 5d410b29e6     [Up-to-date]
PS C:\Users\mazeb\Documents\PlatformIO\Projects\Test1>
```

If this is not a bug, please close this issue and advise how to overcome this roadblock.

Thanks

closed 2018-11-23 by ivankravets

ciphermaster, 2018-12-06:
Hi, 
This seems to happen again this time for board:

nodemcu-32s

with the olimex-arm-usb-ocd-h.

Do you think you can check all the paths for all boards? Seems to be an issue for all of them....

Thanks

ivankravets, 2018-12-06:
@ciphermaster please file a new issue https://github.com/platformio/platform-espressif32/issues and add some small project how to reproduce it. Thanks!

PR #139 (2018-11-23): Force forward slash for paths for openocd-esp32, new board D-duino-32
fixes #136
commit 4f2ca7db (2018-11-23): Force forward slash for paths for openocd-esp32, new board D-duino-32 (#139)

* Force forward slash for paths for openocd-esp32

fixes #136

* adding D-duino-32 board type.

This board has a built in OLED and non-standard SCL/SDA pins.