platformio/platform-espressif32 #137: board_build.partitions not recognized on one PC but works on another one

beegee-tokyo, 2018-11-12:
Strange problem.

I am trying to compiling the same source code on my PC (Windows 7) and on my laptop (Windows 7).
The source code is hosted on bitbucket and I pull the sources from there on both my PC and my laptop, so it should be identical.

Project folder is identical.
platformIO.ini is identical.
```
; PlatformIO Project Configuration File
;
;   Build options: build flags, source filter
;   Upload options: custom upload port, speed and extra flags
;   Library options: dependencies, extra library storages
;   Advanced options: extra scripting
;
; Please visit documentation for the other options and examples
; https://docs.platformio.org/page/projectconf.html

[env:remote-smart]
; platform = espressif32
platform = https://github.com/platformio/platform-espressif32.git#feature/stage
; board = esp32dev
board = esp-wrover-kit

framework = arduino

upload_port = 192.168.0.113
; upload_port = COM12
; upload_speed = 921600

build_flags =
	-DBOARD_HAS_PSRAM
	-mfix-esp32-psram-cache-issue
	-Os
	-fmerge-all-constants
	-ffast-math
	-fira-loop-pressure
	-fdata-sections
	-fexceptions
	-DCORE_DEBUG_LEVEL=ARDUHAL_LOG_LEVEL_NONE
	-DSW_VERSION=0
	-DSW_SUBVERSION=0
	-DSW_MINVERSION=1
	-DFS_VERSION=0
	-DFS_SUBVERSION=0
	-DFS_MINVERSION=1

board_build.partitions = custompart.csv
; lib_ldf_mode = deep
lib_ignore=Time

[platformio]
data_dir = ./src/data
```
On my PC the project compiles without any error and is flashed to the ESP32 Wrover board.
On my laptop the compilation stops after linking with the error that the app size exceeds the available memory space.

Any idea what I am missing?
Any msvc or platformio files I am missing when I push/pull to bitbucket?

ivankravets, 2019-01-10:
See https://github.com/espressif/arduino-esp32/blob/master/tools/platformio-build.py#L215

Where did you put your `custompart.csv`?

beegee-tokyo, 2019-01-10:
On both PC the custom part.csv is at the same location, same level as the src, libs and platformio.ini.

What drives me crazy is that on both PCs I clone the same repo.

beegee-tokyo, 2019-01-26:
Ok, I was not using my laptop for several months.
Started it today, updated VSC, PlatformIO and all boards and libs and the problem is gone.
Not sure why or what changed beside of updating everything.

closed 2019-01-26 by beegee-tokyo