platformio/platform-espressif32 #143: Issues with OTA - cannot flash new firmware

wegunterjr, 2018-12-04:
I have used the example OTA from the esp-idf github.  I was able to successfully download and install OTA, but I am unable to flash new OTA (using PlatformIO in VSCode).

I was able to flash using the esp-idf and program new firmware, but cannot do it in the IDE.

Thoughts?

[https://github.com/espressif/esp-idf/blob/b4b1b361da4a9dd0f7f9be0187c3cdaecb8da43a/examples/system/ota/native_ota_example/main/native_ota_example.c]

ivankravets, 2019-01-10:
> Do not know how to make File target erase

You need the latest upstream version https://docs.platformio.org/en/latest/platforms/espressif32.html#upstream

Does it work now?

ivankravets, 2019-01-18:
Fixed in https://www.reddit.com/r/esp8266/comments/ah5f0g/weird_ota_issue_with_platformio/

closed 2019-01-18 by ivankravets