platformio/platform-espressif32 #145: ESP32 fails upload due to incorrect image.base_address option value - AGAIN

ciphermaster, 2018-12-06:
Hi,

the issue with the invalid image.base_adress is still pending:

> Executing task: C:\Users\Michael\.platformio\penv\Scripts\platformio.exe run --target upload <

Processing esp32doit-devkit-v1 (framework: arduino; platform: espressif32; board: esp32doit-devkit-v1)
---------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/espressif32/esp32doit-devkit-v1.html
PLATFORM: Espressif 32 > DOIT ESP32 DEVKIT V1
HARDWARE: ESP32 240MHz 320KB RAM (4MB Flash)
DEBUG: CURRENT(olimex-arm-usb-ocd-h) EXTERNAL(esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)
Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF MODES: FINDER(chain) COMPATIBILITY(soft)
Collected 24 compatible libraries
Scanning dependencies...
Dependency Graph
|-- <ESP32> 1.0
|-- <Wire> 1.0
Retrieving maximum program size .pioenvs\esp32doit-devkit-v1\firmware.elf
Checking size .pioenvs\esp32doit-devkit-v1\firmware.elf
Memory Usage -> http://bit.ly/pio-memory-usage
DATA:    [          ]   4.7% (used 15276 bytes from 327680 bytes)
PROGRAM: [=         ]  14.6% (used 191888 bytes from 1310720 bytes)
Configuring upload protocol...
AVAILABLE: esp-prog, esptool, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa
CURRENT: upload_protocol = olimex-arm-usb-ocd-h
Uploading .pioenvs\esp32doit-devkit-v1\firmware.bin
Open On-Chip Debugger 0.10.0-dev (2018-11-05-04:10)
Licensed under GNU GPL v2
For bug reports, read
http://openocd.org/doc/doxygen/bugs.html
adapter speed: 20000 kHz
esp32 interrupt mask on
Info : ftdi: if you experience problems at higher adapter clocks, try the command "ftdi_tdo_sample_edge falling"
Info : clock speed 20000 kHz
Info : JTAG tap: esp32.cpu0 tap/device found: 0x120034e5 (mfg: 0x272 (Tensilica), part: 0x2003, ver: 0x1)
Info : JTAG tap: esp32.cpu1 tap/device found: 0x120034e5 (mfg: 0x272 (Tensilica), part: 0x2003, ver: 0x1)
Info : Target halted. PRO_CPU: PC=0x40000400 (active)    APP_CPU: PC=0x40000400
Info : JTAG tap: esp32.cpu0 tap/device found: 0x120034e5 (mfg: 0x272 (Tensilica), part: 0x2003, ver: 0x1)
Info : JTAG tap: esp32.cpu1 tap/device found: 0x120034e5 (mfg: 0x272 (Tensilica), part: 0x2003, ver: 0x1)
Info : esp32: Debug controller 0 was reset (pwrstat=0x5F, after clear 0x0F).
Info : esp32: Core 0 was reset (pwrstat=0x5F, after clear 0x0F).
Info : Target halted. PRO_CPU: PC=0x5000004B (active)    APP_CPU: PC=0x00000000
Info : esp32: Core 0 was reset (pwrstat=0x1F, after clear 0x0F).
Info : esp32: Debug controller 1 was reset (pwrstat=0x5F, after clear 0x0F).
Info : esp32: Core 1 was reset (pwrstat=0x5F, after clear 0x0F).
Info : Target halted. PRO_CPU: PC=0x40000400 (active)    APP_CPU: PC=0x40000400
** Programming Started **
auto erase enabled
Info : Target halted. PRO_CPU: PC=0x4009171A (active)    APP_CPU: PC=0x40000400
Info : Flash mapping 0: 0x10020 -> 0x3f400020, 36 KB
Info : Flash mapping 1: 0x20018 -> 0x400d0018, 96 KB
Info : Target halted. PRO_CPU: PC=0x4009171A (active)    APP_CPU: PC=0x40000400
Info : Auto-detected flash size 4096 KB
Info : Using flash size 4096 KB
Info : Target halted. PRO_CPU: PC=0x4009171A (active)    APP_CPU: PC=0x40000400
Info : Target halted. PRO_CPU: PC=0x4009171A (active)    APP_CPU: PC=0x40000400
wrote 192512 bytes from file .pioenvs\esp32doit-devkit-v1\firmware.bin in 2.894262s (64.956 KiB/s)
** Programming Finished **
** Verify Started **
Info : Target halted. PRO_CPU: PC=0x4009171A (active)    APP_CPU: PC=0x40000400
read 192032 bytes from file .pioenvs\esp32doit-devkit-v1\firmware.bin and flash bank 0 at offset 0x00010000 in 0.996777s (188.138 KiB/s)
contents match
** Verified OK **
Info : JTAG tap: esp32.cpu0 tap/device found: 0x120034e5 (mfg: 0x272 (Tensilica), part: 0x2003, ver: 0x1)
Info : JTAG tap: esp32.cpu1 tap/device found: 0x120034e5 (mfg: 0x272 (Tensilica), part: 0x2003, ver: 0x1)
Info : esp32: Debug controller 0 was reset (pwrstat=0x5F, after clear 0x0F).
Info : esp32: Core 0 was reset (pwrstat=0x5F, after clear 0x0F).
Info : Target halted. PRO_CPU: PC=0x5000004B (active)    APP_CPU: PC=0x00000000
Info : esp32: Core 0 was reset (pwrstat=0x1F, after clear 0x0F).
Info : esp32: Debug controller 1 was reset (pwrstat=0x5F, after clear 0x0F).
Info : esp32: Core 1 was reset (pwrstat=0x5F, after clear 0x0F).
Info : Target halted. PRO_CPU: PC=0x40000400 (active)    APP_CPU: PC=0x40000400
** Programming Started **
auto erase enabled
Error: Invalid command argument
image.base_address option value ('ramework-arduinoespressif32') is not valid
embedded:startup.tcl:454: Error: ** Programming Failed **
in procedure 'program_esp32'
in procedure 'program_error' called at file "C:\Users\Michael\.platformio\packages\tool-openocd-esp32/share/openocd/scripts/target/esp32.cfg", line 152
at file "embedded:startup.tcl", line 454
*** [upload] Error 1

I followed all the tasks mentioned in 
[https://github.com/platformio/platform-espressif32/issues/136](url)

But still no luck. My plattform.ini with "dev-upgrade":

[env:esp32doit-devkit-v1]
platform = espressif32
board = esp32doit-devkit-v1
framework = arduino

monitor_speed = 115200

debug_tool = olimex-arm-usb-ocd-h
upload_protocol = olimex-arm-usb-ocd-h

Please help here. What can I do to move forwared.

atanisoft, 2018-12-11:
same as #136 and the fix is in https://github.com/platformio/platform-espressif32/pull/139 which was merged but I don't think a release has been made yet with it. you can manually apply it though.

ciphermaster, 2018-12-11:
Do you have some hands on for that? Would be really nice. I have already tried to do a "dev-update".
Thanks

ivankravets, 2019-01-10:
This issue was fixed in the latest upstream version https://docs.platformio.org/en/latest/platforms/espressif32.html#upstream

closed 2019-01-10 by ivankravets