platformio/platform-espressif32 #148: Unable to disambiguate libraries with the same name

ciband, 2018-12-15:
I am having a similar issue as #72 with the HTTPClient package library on Arduino ESP32.

It seems that when I build my library from source, the system picks the HTTPClient from the Arduino ESP32 package library folder, which is what I want.

When I attempt to use my library through the platformIO library manager in a seperate app, the system picks a HTTPClient library from the library manager.  Telling the system to ignore that specific library does not work.

I think there are two issues:

1) There is inconsistent behavior of LDF between compiling the code as a library v. being included in an app via the library manager.

2) There is no way to tell the system to explicitly link package libraries.

I am new to PIO so maybe I missed something.

I appreciate any help anyone can give.

ivankravets, 2018-12-18:
Have you seen this order https://docs.platformio.org/en/latest/librarymanager/ldf.html#storage?

Maybe, this feature request will help? https://github.com/platformio/platformio-core/issues/1824

ciband, 2018-12-18:
I have seen this order.  My issue is that there are two HTTPClient libraries.  The first is at level 3, the second is at level 5.

I would like to explicitly use the SDK library at level 5 instead of level 3.

I tried ignoring the level 3 library explicitly using lib_ignore but that did not seem to work.

I don't think the feature helps for my use case.

ivankravets, 2019-01-10:
Could you try to add "use the SDK library at level 5" path directly to https://docs.platformio.org/en/latest/projectconf/section_env_library.html#lib-extra-dirs ? It will have a higher priority.

ciband, 2019-01-10:
That ended up being my workaround. 

The issue is that I don’t want the user of my library to have to do that. I would prefer it to be hidden from them. 

I opened this ticket as a way to fix that. 

https://github.com/platformio/platformio-core/issues/1977

pfeerick, 2019-06-30:
@ciband  If I understand this issue correctly, you're saying that when you use your library in a program, PIO is grabbing the HTTPClient library from the registry, rather than using the version included with the esp32 core? Perhaps [this one](https://platformio.org/lib/show/66/HttpClient)? My confusion there is then, why is it getting it...unless told to... do you by any chance specify it in your dependencies? If so, I'd suspect that you shouldn't - if it's in the platform/framework - it's not an 'extra' library you're dependent on. 

Maybe this would all become a bit clearer if you link to the library, and a minimal code example and `platformio.ini` which demonstrates the issue.

ivankravets, 2019-06-30:
It's fixed in PlatformIO Core 4.0. Now, new LDF looks into built-in library storages first. If you need to force library with the same name from the registry, just add SemVer requirement. For example, `HTTPClient@*`

closed 2019-06-30 by ivankravets

sylque, 2024-08-01:
@ivankravets , maybe this issue is back. I've just tried to use the native ESP32 `Preferences` library, but PlatformIO picks an external one instead:
```ini
[env:lolin32]
platform = espressif32
platform_packages=
  framework-arduinoespressif32 @ https://github.com/espressif/arduino-esp32.git#3.0.3
  framework-arduinoespressif32-libs @ https://github.com/espressif/arduino-esp32/releases/download/3.0.3/esp32-arduino-libs-3.0.3.zip
board = lolin_d32
framework = arduino
```
By default, PlatformIO picks this library, leading to compilation errors:
```
name=Preferences
version=2.1.0
author=Volodymyr Shymanskyy
url=https://github.com/vshymanskyy/Preferences
```
A workaround is to specify the version number of the native library, which is fortunately higher:
```ini
[env:lolin32]
platform = espressif32
platform_packages=
  framework-arduinoespressif32 @ https://github.com/espressif/arduino-esp32.git#3.0.3
  framework-arduinoespressif32-libs @ https://github.com/espressif/arduino-esp32/releases/download/3.0.3/esp32-arduino-libs-3.0.3.zip
board = lolin_d32
framework = arduino
lib_deps = Preferences@3.0.3
```
With this modified ini file, PlatformIO loads the right library and the project compiles fine.

ivankravets, 2024-08-01:
Could you remove the built in library from `lib_deps` and add its “include” to the main.cpp?

sylque, 2024-08-01:
When the library is removed from `lib_deps` and `#include <Preferences.h>` is added to main.cpp, PlatformIO still picks the wrong library.

ivankravets, 2024-08-01:
Please provide a simple project to reproduce this issue and attach as ZIP to this ticket

sylque, 2024-08-02:
@ivankravets , please see [PIO Issue - ESP32 Preferences Lib.zip](https://github.com/user-attachments/files/16471530/PIO.Issue.-.ESP32.Preferences.Lib.zip). As you'll see, the issue is not as simple as I initially though, as it involves a library.json file. 

When you asked me earlier to "remove the built in library from lib_deps", I did so, but forgot to also remove it from the library.json file. Removing it from library.json is indeed a working workaround.

ivankravets, 2024-08-03:
```
Scanning dependencies...
Dependency Graph
|-- Lib (License: Unknown, Path: /Users/ikravets/Downloads/issue/Lib)
|   |-- Preferences @ 2.0.0 (License: Unknown, Path: /Users/ikravets/.platformio/packages/framework-arduinoespressif32/libraries/Preferences)
```
[issue.zip](https://github.com/user-attachments/files/16478671/issue.zip)

sylque, 2024-08-04:
@ivankravets , not sure about your latest comment. Is there anything I can do?

ivankravets, 2024-08-04:
I attached the project where everything works as expected.

sylque, 2024-08-04:
Your projects doesn't target arduino-esp32 v3.0.3 as mine does. 
Anyway, I'm happy with the 2 workarounds described in this thread. I just wanted to help pinpoint a potential issue in PIO.

ivankravets, 2024-08-04:
Sorry, we currently do not support ESP32 Core for Arduino v3.0. Please follow https://github.com/platformio/platform-espressif32/issues/1225

vortigont, 2024-08-22:
@sylque you may want to check [platformio/platformio-core/#4941](https://github.com/platformio/platformio-core/pull/4941)
similar issue, suggested patch does work fine for me

sylque, 2024-08-22:
Thanks @vortigont. I've just tried your patch. Unfortunately, it doesn't solve the issue in the provided project [PIO Issue - ESP32 Preferences Lib.zip](https://github.com/user-attachments/files/16471530/PIO.Issue.-.ESP32.Preferences.Lib.zip).

vortigont, 2024-08-22:
@sylque
it works fine for me.
You do not need any hacks with deps in jsons with that patch.
so you may remove that from ini file
```
lib_deps = 
    symlink://../Lib
```

If you need internal lib that is provided by Arduino core, it should pick it

```
Dependency Graph before correct
|-- Preferences @ 3.0.3
D _correct_found_libs: run recursive search
D _correct_found_libs: run recursive search for <class 'piolib.ArduinoLibBuilder'>('/home/emil/.platformio/packages/framework-arduinoespressif32/libraries/Preferences')
D _correct_found_libs: run remover
Dependency Graph
|-- Preferences @ 3.0.3
```

Jason2866, 2024-08-22:
Probably **not** to pin the platform version is causing the problem. So change in your setup from
`platform = espressif32` to `platform = espressif32@6.8.1`

vortigont, 2024-08-22:
@Jason2866 first I've faced this issue was with Tasmota's arduino core for 2.x, BUT only when used in GitHub CI.
Then I've start digging and reproduced it locally when removed dependencies placed into `lib_deps` under libs json file. Still awaiting more feedback on suck cases. Have three of those already - this issue, AsyncWebserver and my flashz lib.