platformio/platform-espressif32 #150: Ai-Thinker ESP32-CAM board support

dontsovcmc, 2018-12-20:
I cant' understand what board I should write for this item:

http://www.cnx-software.ru/wp-content/uploads/2018/10/ESP32-CAM.jpg

[ESP-IDF repo ](https://github.com/donny681/ESP32_CAMERA_QR) example

I try to compile as:
- esp-wrover-kit
- esp32dev 

output:

```
--- Miniterm on /dev/tty.SLAB_USBtoUART  115200,8,N,1 ---
--- Quit: Ctrl+C | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H ---
␀ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:952
load:0x40078000,len:6084
load:0x40080000,len:7944
entry 0x40080310
user code done
```

-DBOARD_HAS_PSRAM doesn't help.

Thanks a lot!

michgz, 2019-04-02:
I have that board, and esp32dev works without issue. For example project at ~/.platformio/platforms/espressif32/examples/espidf-blink/

In .ini file: change
board = esp32dev
build_flags = -D CONFIG_BLINK_GPIO=4

I'm running in Linux, Atom, Home=2.0.0 Core=3.6.5

closed 2019-04-17 by ivankravets

dontsovcmc, 2019-04-17:
Good job! Thanks!

ivankravets, 2019-04-17:
Please re-test with the latest upstream version http://docs.platformio.org/en/latest/platforms/espressif32.html#upstream

The board profile is `esp32cam`.

commit 07204768 (2019-04-17): Add new boards: AI Thinker ESP32-CAM, Frog Board ESP32, Heltec WiFi LoRa 32 (V2), Heltec Wireless Stick, M5Stick-C, SparkFun LoRa Gateway 1-Channel // Resolve #150