platformio/platform-espressif32 #153: cannot enable bluetooth

Herschdorfer, 2019-01-04:
Hi,

i'm trying to get the [A2DP  sink example](https://github.com/espressif/esp-idf/tree/master/examples/bluetooth/a2dp_sink) to run on a PIO installation.

The code (see below) is failing at:
```cpp
    ESP_ERROR_CHECK(esp_bt_controller_mem_release(ESP_BT_MODE_BLE));
```

if I remove this line (i have seen other issues with this) this line

```cpp
    if ((err = esp_bt_controller_init(&bt_cfg)) != ESP_OK)
```

is failing with an invalid state.

> [E][bluetooth.cpp:95] Bluetooth_init(): Bluetooth_init initialize controller failed: ESP_ERR_INVALID_STATE

### Hardware:
Board:							WEMOS Bluetooth and Battery
Core Installation/update date:		PIO Core v3.6.3, platform-espressif32 v.1.5.0
IDE name:						Platform.io IDE 
Flash Frequency:					40Mhz
PSRAM enabled:                                      no
Upload Speed:					115200
Computer OS:                                          Windows 10

### Code:
```cpp

void Bluetooth_init()
{
    esp_err_t err = nvs_flash_init();
    if (err == ESP_ERR_NVS_NO_FREE_PAGES)
    {
        ESP_ERROR_CHECK(nvs_flash_erase());
        err = nvs_flash_init();
    }
    ESP_ERROR_CHECK(err);

    ESP_ERROR_CHECK(esp_bt_controller_mem_release(ESP_BT_MODE_BLE));

    esp_bt_controller_config_t bt_cfg = BT_CONTROLLER_INIT_CONFIG_DEFAULT();
    if ((err = esp_bt_controller_init(&bt_cfg)) != ESP_OK)
    {
        ESP_LOGE(BT_AV_TAG, "%s initialize controller failed: %s\n", __func__, esp_err_to_name(err));
        return;
    }

    if ((err = esp_bt_controller_enable(ESP_BT_MODE_CLASSIC_BT)) != ESP_OK)
    {
        ESP_LOGE(BT_AV_TAG, "%s enable controller failed: %s\n", __func__, esp_err_to_name(err));
        return;
    }

    if ((err = esp_bluedroid_init()) != ESP_OK)
    {
        ESP_LOGE(BT_AV_TAG, "%s initialize bluedroid failed: %s\n", __func__, esp_err_to_name(err));
        return;
    }

    if ((err = esp_bluedroid_enable()) != ESP_OK)
    {
        ESP_LOGE(BT_AV_TAG, "%s enable bluedroid failed: %s\n", __func__, esp_err_to_name(err));
        return;
    }

    /* create application task */
    bt_app_task_start_up();

    /* Bluetooth device name, connection mode and profile set up */
    bt_app_work_dispatch(bt_av_hdl_stack_evt, 0, NULL, 0, NULL);
}

```

### platformio.ini
```
[env:wemosbat]
platform = espressif32
framework = arduino
board = wemosbat

monitor_speed = 115200
monitor_port = COM4
upload_port = COM4

board_build.partitions = partitions.csv
```

Herschdorfer, 2019-01-04:
I also tried the plain esp-idf example on the same device and it works flawless.

ivankravets, 2019-01-10:
Could you try the latest development version of ESP32 core for Arduino (stage version)? See https://docs.platformio.org/en/latest/platforms/espressif32.html#using-arduino-framework-with-staging-version

ivankravets, 2019-01-18:
New version of Arduino ESP32 core is out with the latest version of dev/platform. Please update all packages via `pio update`.

closed 2019-01-18 by ivankravets

tobiasebsen, 2019-10-04:
I have an issue that is similar to this. Trying to get the `bt_discovery` example from esp-idf compiled and uploaded with PlatformIO, but with esp-idf as framework.

The `esp_bt_controller_init` also fails but with error `ESP_ERR_INVALID_ARG`

I suspected the sdkconfig to have wrong settings, but since I copied it from the example it should work, right? Unless the toolchain somehow ignores the sdkconfig