platformio/platform-espressif32 #154: Problems with Espressif32 toolchain on Raspbian Jessie (Debian 8)

CVBruce, 2019-01-21:


CVBruce, 2019-01-21:
While installing Platformio on my Raspberry Pi, I ran into problems with the ESP32 toolchain.

The error message referred a missing CXXABI 1.3.9 entry in libstdc++.

It appears that Raspbian 8 (Jessie) is incompatible with the ESP32 environment because the highest level of CXXABI supported is version 1.3.8.

Raspbian 9 (Stretch) appears to be working.

This may be due to Raspbian Jessie supports up to gcc 4.9 while Raspbian Stretch supports gcc 6.3

CVBruce, 2019-01-21:
pi@RPi3B7:~ $ pio init --board lolin32;pio run -e lolin32
The current working directory /home/pi will be used for the project.

The next files/directories have been created in /home/pi
include - Put project header files here
lib - Put here project specific (private) libraries
src - Put project source files here
platformio.ini - Project Configuration File

Project has been successfully updated! Useful commands:
`pio run` - process/build project from the current directory
`pio run --target upload` or `pio run -t upload` - upload firmware to a target
`pio run --target clean` - clean project (remove compiled files)
`pio run --help` - additional information

Processing lolin32 (platform: espressif32; board: lolin32; framework: arduino)
------------------------------------------------------------------------------------------------------------------------------------------------
PlatformManager: Installing espressif32
Downloading  [####################################]  100%
Unpacking  [####################################]  100%
espressif32 @ 1.6.0 has been successfully installed!
The platform 'espressif32' has been successfully installed!
The rest of packages will be installed automatically depending on your build environment.
PackageManager: Installing toolchain-xtensa32 @ ~2.50200.0
Downloading  [####################################]  100%          
Unpacking  [####################################]  100%          
PackageManager: Installing framework-arduinoespressif32 @ ~2.10001.190111
Downloading  [####################################]  100%          
Unpacking  [####################################]  100%          
PackageManager: Installing tool-esptoolpy @ ~1.20600.0
Downloading  [####################################]  100%
Unpacking  [####################################]  100%
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/espressif32/lolin32.html
PLATFORM: Espressif 32 > WEMOS LOLIN32
HARDWARE: ESP32 240MHz 320KB RAM (4MB Flash)
DEBUG: CURRENT(esp-prog) EXTERNAL(esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)
Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF MODES: FINDER(chain) COMPATIBILITY(soft)
Collected 26 compatible libraries
Scanning dependencies...
No dependencies
Compiling .pioenvs/lolin32/src/main.cpp.o
xtensa-esp32-elf-g++: /usr/lib/arm-linux-gnueabihf/libstdc++.so.6: version `CXXABI_1.3.9' not found (required by xtensa-esp32-elf-g++)
Generating partitions .pioenvs/lolin32/partitions.bin
Archiving .pioenvs/lolin32/libFrameworkArduinoVariant.a
Indexing .pioenvs/lolin32/libFrameworkArduinoVariant.a
*** [.pioenvs/lolin32/src/main.cpp.o] Error 1
Compiling .pioenvs/lolin32/FrameworkArduino/Esp.cpp.o
xtensa-esp32-elf-g++: /usr/lib/arm-linux-gnueabihf/libstdc++.so.6: version `CXXABI_1.3.9' not found (required by xtensa-esp32-elf-g++)
*** [.pioenvs/lolin32/FrameworkArduino/Esp.cpp.o] Error 1
========================================================= [ERROR] Took 228.66 seconds =========================================================

================================================================== [SUMMARY] ==================================================================
Environment uno      	[SKIP]
Environment nodemcuv2	[SKIP]
Environment teensy31 	[SKIP]
Environment lolin32  	[ERROR]
========================================================= [ERROR] Took 228.66 seconds =========================================================
pi@RPi3B7:~ $ cat /etc/os-release
PRETTY_NAME="Raspbian GNU/Linux 8 (jessie)"
NAME="Raspbian GNU/Linux"
VERSION_ID="8"
VERSION="8 (jessie)"
ID=raspbian
ID_LIKE=debian
----- Links removed to allow posting ----
pi@RPi3B7:~ $ dpkg -l | grep libstdc++
ii  libstdc++-4.9-dev:armhf               4.9.2-10+deb8u2                           armhf        GNU Standard C++ Library v3 (development files)
ii  libstdc++6:armhf                      4.9.2-10+deb8u2                           armhf        GNU Standard C++ Library v3
pi@RPi3B7:~ $ strings /usr/lib/arm-linux-gnueabihf/libstdc++.so.6 | grep CXXABI
CXXABI_1.3
CXXABI_1.3.1
CXXABI_1.3.2
CXXABI_1.3.3
CXXABI_1.3.4
CXXABI_1.3.5
CXXABI_1.3.6
CXXABI_1.3.7
CXXABI_1.3.8
CXXABI_TM_1
CXXABI_ARM_1.3.3
pi@RPi3B7:~ $

hoatienii, 2019-04-23:
i have a same problem
xtensa-esp32-elf-g++: /usr/lib/arm-linux-gnueabihf/libstdc++.so.6: version `CXXABI_1.3.9' not found (required by xtensa-esp32-elf-g++)

IMR84, 2019-12-15:
**How to solve this problem i am very frustrated.**

Arduino: 1.8.10 (Linux), Board: "ESP32 Wrover Module, Huge APP (3MB No OTA), QIO, 40MHz, 115200, None"

/home/pi/Downloads/arduino-1.8.10/arduino-builder -dump-prefs -logger=machine -hardware /home/pi/Downloads/arduino-1.8.10/hardware -hardware /home/pi/.arduino15/packages -hardware /home/pi/Arduino/hardware -tools /home/pi/Downloads/arduino-1.8.10/tools-builder -tools /home/pi/Downloads/arduino-1.8.10/hardware/tools/avr -tools /home/pi/.arduino15/packages -built-in-libraries /home/pi/Downloads/arduino-1.8.10/libraries -libraries /home/pi/Arduino/libraries -fqbn=espressif:esp32:esp32wrover:PartitionScheme=huge_app,FlashMode=qio,FlashFreq=40,UploadSpeed=115200,DebugLevel=none -ide-version=10810 -build-path /tmp/arduino_build_112919 -warnings=default -build-cache /tmp/arduino_cache_492425 -prefs=build.warn_data_percentage=75 -verbose /home/pi/Arduino/hardware/espressif/esp32/libraries/ESP32/examples/Camera/CameraWebServer/CameraWebServer.ino
/home/pi/Downloads/arduino-1.8.10/arduino-builder -compile -logger=machine -hardware /home/pi/Downloads/arduino-1.8.10/hardware -hardware /home/pi/.arduino15/packages -hardware /home/pi/Arduino/hardware -tools /home/pi/Downloads/arduino-1.8.10/tools-builder -tools /home/pi/Downloads/arduino-1.8.10/hardware/tools/avr -tools /home/pi/.arduino15/packages -built-in-libraries /home/pi/Downloads/arduino-1.8.10/libraries -libraries /home/pi/Arduino/libraries -fqbn=espressif:esp32:esp32wrover:PartitionScheme=huge_app,FlashMode=qio,FlashFreq=40,UploadSpeed=115200,DebugLevel=none -ide-version=10810 -build-path /tmp/arduino_build_112919 -warnings=default -build-cache /tmp/arduino_cache_492425 -prefs=build.warn_data_percentage=75 -verbose /home/pi/Arduino/hardware/espressif/esp32/libraries/ESP32/examples/Camera/CameraWebServer/CameraWebServer.ino
Using board 'esp32wrover' from platform in folder: /home/pi/Arduino/hardware/espressif/esp32
Using core 'esp32' from platform in folder: /home/pi/Arduino/hardware/espressif/esp32
Detecting libraries used...
/home/pi/Arduino/hardware/espressif/esp32/tools/xtensa-esp32-elf/bin/xtensa-esp32-elf-g++ -DESP_PLATFORM "-DMBEDTLS_CONFIG_FILE=\"mbedtls/esp_config.h\"" -DHAVE_CONFIG_H -I/home/pi/Arduino/hardware/espressif/esp32/tools/sdk/include/config -I/home/pi/Arduino/hardware/espressif/esp32/tools/sdk/include/app_trace -I/home/pi/Arduino/hardware/espressif/esp32/tools/sdk/include/app_update -I/home/pi/Arduino/hardware/espressif/esp32/tools/sdk/include/asio -I/home/pi/Arduino/hardware/espressif/esp32/tools/sdk/include/bootloader_support -I/home/pi/Arduino/hardware/espressif/esp32/tools/sdk/include/bt -I/home/pi/Arduino/hardware/espressif/esp32/tools/sdk/include/coap -I/home/pi/Arduino/hardware/espressif/esp32/tools/sdk/include/console -I/home/pi/Arduino/hardware/espressif/esp32/tools/sdk/include/driver -I/home/pi/Arduino/hardware/espressif/esp32/tools/sdk/include/esp-tls -I/home/pi/Arduino/hardware/espressif/esp32/tools/sdk/include/esp32 -I/home/pi/Arduino/hardware/espressif/esp32/tools/sdk/include/esp_adc_cal -I/home/pi/Arduino/hardware/espressif/esp32/tools/sdk/include/esp_event -I/home/pi/Arduino/hardware/espressif/esp32/tools/sdk/include/esp_http_client -I/home/pi/Arduino/hardware/espressif/esp32/tools/sdk/include/esp_http_server -I/home/pi/Arduino/hardware/espressif/esp32/tools/sdk/include/esp_https_ota -I/home/pi/Arduino/hardware/espressif/esp32/tools/sdk/include/esp_ringbuf -I/home/pi/Arduino/hardware/espressif/esp32/tools/sdk/include/ethernet -I/home/pi/Arduino/hardware/espressif/esp32/tools/sdk/include/expat -I/home/pi/Arduino/hardware/espressif/esp32/tools/sdk/include/fatfs -I/home/pi/Arduino/hardware/espressif/esp32/tools/sdk/include/freemodbus -I/home/pi/Arduino/hardware/espressif/esp32/tools/sdk/include/freertos -I/home/pi/Arduino/hardware/espressif/esp32/tools/sdk/include/heap -I/home/pi/Arduino/hardware/espressif/esp32/tools/sdk/include/idf_test -I/home/pi/Arduino/hardware/espressif/esp32/tools/sdk/include/jsmn -I/home/pi/Arduino/hardware/espressif/esp32/tools/sdk/include/json -I/home/pi/Arduino/hardware/espressif/esp32/tools/sdk/include/libsodium -I/home/pi/Arduino/hardware/espressif/esp32/tools/sdk/include/log -I/home/pi/Arduino/hardware/espressif/esp32/tools/sdk/include/lwip -I/home/pi/Arduino/hardware/espressif/esp32/tools/sdk/include/mbedtls -I/home/pi/Arduino/hardware/espressif/esp32/tools/sdk/include/mdns -I/home/pi/Arduino/hardware/espressif/esp32/tools/sdk/include/micro-ecc -I/home/pi/Arduino/hardware/espressif/esp32/tools/sdk/include/mqtt -I/home/pi/Arduino/hardware/espressif/esp32/tools/sdk/include/newlib -I/home/pi/Arduino/hardware/espressif/esp32/tools/sdk/include/nghttp -I/home/pi/Arduino/hardware/espressif/esp32/tools/sdk/include/nvs_flash -I/home/pi/Arduino/hardware/espressif/esp32/tools/sdk/include/openssl -I/home/pi/Arduino/hardware/espressif/esp32/tools/sdk/include/protobuf-c -I/home/pi/Arduino/hardware/espressif/esp32/tools/sdk/include/protocomm -I/home/pi/Arduino/hardware/espressif/esp32/tools/sdk/include/pthread -I/home/pi/Arduino/hardware/espressif/esp32/tools/sdk/include/sdmmc -I/home/pi/Arduino/hardware/espressif/esp32/tools/sdk/include/smartconfig_ack -I/home/pi/Arduino/hardware/espressif/esp32/tools/sdk/include/soc -I/home/pi/Arduino/hardware/espressif/esp32/tools/sdk/include/spi_flash -I/home/pi/Arduino/hardware/espressif/esp32/tools/sdk/include/spiffs -I/home/pi/Arduino/hardware/espressif/esp32/tools/sdk/include/tcp_transport -I/home/pi/Arduino [... cut ...] hardware/espressif/esp32/variants/esp32 /tmp/arduino_build_112919/sketch/CameraWebServer.ino.cpp -o /tmp/arduino_build_112919/preproc/ctags_target_for_gcc_minus_e.cpp
/home/pi/Arduino/hardware/espressif/esp32/tools/xtensa-esp32-elf/bin/xtensa-esp32-elf-g++: /usr/lib/arm-linux-gnueabihf/libstdc++.so.6: version `CXXABI_1.3.9' not found (required by /home/pi/Arduino/hardware/espressif/esp32/tools/xtensa-esp32-elf/bin/xtensa-esp32-elf-g++)
exit status 1
Error compiling for board ESP32 Wrover Module.

CVBruce, 2019-12-15:
There is no fix.
You can get paste this problem by upgrading to Raspbian 9 (Stretch) or newer.

closed 2022-08-10 by stale[bot]