platformio/platform-espressif32 #160: ESP32 OTA very flakey

luishasmail, 2019-02-16:
Hello! I have been trying to make the OTA example found here work: 

http://docs.platformio.org/en/latest/platforms/espressif32.html#over-the-air-ota-update
https://github.com/espressif/arduino-esp32/tree/master/libraries/ArduinoOTA/examples/BasicOTA

I can get the example to upload every 10 tries or so, but not consistently

I am using PlatformIDE on VSCode on macOS Mojave

Here are the versions that I am running:

VSCode: 1.31.1
PlatformIO: 3.6.4

I have tried disabling my firewall but that did not help. Any help or further guidance would be appreciated, thank you!

Here is the code that I am using, it is identical to the OTA example:

```cpp
#include <WiFi.h>
#include <ESPmDNS.h>
#include <WiFiUdp.h>
#include <ArduinoOTA.h>

const char* ssid = "HOME-C44A-2.4";
const char* password = "3PD94CA9TFDV34AT";

void setup() {
  Serial.begin(115200);
  Serial.println("Booting");
  WiFi.mode(WIFI_STA);
  WiFi.begin(ssid, password);
  while (WiFi.waitForConnectResult() != WL_CONNECTED) {
    Serial.println("Connection Failed! Rebooting...");
    delay(5000);
    ESP.restart();
  }

  // Port defaults to 3232
  // ArduinoOTA.setPort(3232);

  // Hostname defaults to esp3232-[MAC]
  // ArduinoOTA.setHostname("myesp32");

  // No authentication by default
  // ArduinoOTA.setPassword("admin");

  // Password can be set with it's md5 value as well
  // MD5(admin) = 21232f297a57a5a743894a0e4a801fc3
  // ArduinoOTA.setPasswordHash("21232f297a57a5a743894a0e4a801fc3");

  ArduinoOTA
    .onStart([]() {
      String type;
      if (ArduinoOTA.getCommand() == U_FLASH)
        type = "sketch";
      else // U_SPIFFS
        type = "filesystem";

      // NOTE: if updating SPIFFS this would be the place to unmount SPIFFS using SPIFFS.end()
      Serial.println("Start updating " + type);
    })
    .onEnd([]() {
      Serial.println("\nEnd");
    })
    .onProgress([](unsigned int progress, unsigned int total) {
      Serial.printf("Progress: %u%%\r", (progress / (total / 100)));
    })
    .onError([](ota_error_t error) {
      Serial.printf("Error[%u]: ", error);
      if (error == OTA_AUTH_ERROR) Serial.println("Auth Failed");
      else if (error == OTA_BEGIN_ERROR) Serial.println("Begin Failed");
      else if (error == OTA_CONNECT_ERROR) Serial.println("Connect Failed");
      else if (error == OTA_RECEIVE_ERROR) Serial.println("Receive Failed");
      else if (error == OTA_END_ERROR) Serial.println("End Failed");
    });

  ArduinoOTA.begin();

  Serial.println("Ready");
  Serial.print("IP address: ");
  Serial.println(WiFi.localIP());
}

void loop() {
  ArduinoOTA.handle();
}
```

Here is my `platform.ini`

```ini
[env:esp32thing]
platform = espressif32
board = esp32thing
framework = arduino
monitor_speed = 115200
```

Here is the command I am using to upload

```zsh
platformio run -t upload --upload-port 10.0.0.132
```

davidsiemers, 2019-08-06:
Did you resolve this issue?
Curious if you found a solution as OTA has been hit or miss for me, both with ESP32s and ESP8266s.

Generally, I've found you can get it fairly stable with a high quality PSU and a non-Aliexpress sourced board, but still feels like a crapshoot. Would be worthwhile to try a few different PSU/boards out and see if you can get it to work.
I've been looking at the HTTPUpdate library and have heard great things about it, Haven't had a chance to try it yet, but I'd give it a look. If it resolves your issues, I may have to give it a go on my next project!

https://github.com/espressif/arduino-esp32/tree/master/libraries/HTTPUpdate
https://github.com/espressif/arduino-esp32/tree/master/libraries/Update
https://github.com/espressif/arduino-esp32/blob/master/libraries/Update/examples/AWS_S3_OTA_Update/AWS_S3_OTA_Update.ino

davidsiemers, 2019-08-06:
@luishasmail Did you resolve this issue?
Curious if you found a solution as OTA has been hit or miss for me, both with ESP32s and ESP8266s.

Generally, I've found you can get it fairly stable with a high quality PSU and a non-Aliexpress sourced board, but still feels like a crapshoot. Would be worthwhile to try a few different PSU/boards out and see if you can get it to work.
I've been looking at the HTTPUpdate library and have heard great things about it, Haven't had a chance to try it yet, but I'd give it a look. If it resolves your issues, I may have to give it a go on my next project!

https://github.com/espressif/arduino-esp32/tree/master/libraries/HTTPUpdate
https://github.com/espressif/arduino-esp32/tree/master/libraries/Update
https://github.com/espressif/arduino-esp32/blob/master/libraries/Update/examples/AWS_S3_OTA_Update/AWS_S3_OTA_Update.ino

luishasmail, 2019-08-06:
I haven’t had any luck getting this to work. Right now I’m using Sparkfun ESP32 Things and an Apple iPad power supply. If that’s not good enough I don’t know what will be.

I’ll have to look into those libraries.

joba-1, 2019-08-06:
it would probably help if you post output of failed otas (both sides, verbose). 

I don’t have board related (or origin related) problems with it, once power supply is good - mostly using esp8266 though. But I can remember people had issues with which port is default for esp32s (3232 vs 8266) lately. You could try setting the port explicitly on both sides.

davidsiemers, 2019-08-06:
I'd definitely give other power supplies a shot. The 5v voltage regulators on these boards can be finicky and while a PSU may be in spec, it's starting voltage or even running voltage may be below what the regulator needs to run reliably. Also, some companies like to implement things slightly off spec, too; the Nintendo Switch PSU is a great example of that. A generic, 1A/5W or similar PSU may be your best bet for something like this. May need to try every wall wort you've got hoarded away, though ;)
I'd also try running in it on just a LiPo battery, which should bypass that regulator. Worth a shot :).

closed 2022-08-10 by stale[bot]