platformio/platform-espressif32 #161: COMPONENT_EMBED_TXTFILES  modifies included files

thomasonw, 2019-02-18:
There seems to be a bug where PlatformIO corrupts embedded files.

Following the directions here:  [http://docs.platformio.org/en/latest/platforms/espressif32.html#embedding-binary-data](http://docs.platformio.org/en/latest/platforms/espressif32.html#embedding-binary-data)

I created a test case.    Using PlatformIO to create a new project for the ESP32dev board, selecting the IDF (Not Arduino) env.

Create these three files:
**main.c** 
```
#include <stdint.h>

extern const uint8_t code_js_start[] asm("_binary_src_main_js_start");				
extern const uint8_t code_js_end[] asm("_binary_src_main_js_end");

void app_main() {
    
}
```

**main.js**
```
var mh = document.querySelector('h1');
mh.textContent = 'Hello World';
```

**platformio.ini**
```
; PlatformIO Project Configuration File
;
;   Build options: build flags, source filter
;   Upload options: custom upload port, speed and extra flags
;   Library options: dependencies, extra library storage
;   Advanced options: extra scripting
;
; Please visit documentation for the other options and examples
; https://docs.platformio.org/page/projectconf.html

[env:esp32dev]
platform = espressif32
board = esp32dev
framework = espidf
monitor_speed = 115200

build_flags =
    -DCOMPONENT_EMBED_TXTFILES=src/main.js

```

Before compiling the file size  of main.js is 70 bytes.   After compiling the file has been modified with an extra byte added to the end increasing the filesize of 71 bytes.  This extra byte is only added once, in that future 'compiles' will not continue to append an extra byte.  Looking at sample files which have been modified - it appears that a NULL byte(0x00)  is being appending to the end of text files before they are included.  And this NULL is carried forward into the binary image, as well as written back to the file on the disk.

This is very repeatable and seems to impact only text files (??).  Uploading a .gz (binary) file I do not notice this behavior.  (Though it could be the .gz already has a 0x00 ending)

Host:  Ubuntu 14.04lts
Visual Studio Code ver: 1.31.1
PlatformIO:  Home 2.0.0·Core 3.6.4

teliot, 2019-06-23:
This is be design.

Per this document
https://docs.espressif.com/projects/esp-idf/en/latest/api-guides/build-system.html#embedding-binary-data

Embedding Binary Data

Sometimes you have a file with some binary or text data that you’d like to make available to your component - but you don’t want to reformat the file as C source.

You can set a variable COMPONENT_EMBED_FILES in component.mk, giving the names of the files to embed in this way:

COMPONENT_EMBED_FILES := server_root_cert.der

Or if the file is a string, you can use the variable COMPONENT_EMBED_TXTFILES. This will embed the contents of the text file as a null-terminated string:

COMPONENT_EMBED_TXTFILES := server_root_cert.pem

The file’s contents will be added to the .rodata section in flash, and are available via symbol names as follows:"

joba-1, 2019-06-24:
I agree with op: src files should not be modified. End result on flash is another story.

thomasonw, 2019-06-24:
@teliot   Thank you for the additional reference, that clears things up nicely how to handle text/string files vs. binary files.   Though I have moved on for now and am not able to verify, the link is rather clear what path to take and I will close this issue.

Though I might offer, the original document link [http://docs.platformio.org/en/latest/platforms/espressif32.html#embedding-binary-data](http://docs.platformio.org/en/latest/platforms/espressif32.html#embedding-binary-data) is a bit misleading, talking about embedding binary files, then using the text include macro in the example,  and not making the distinction clear..

Thank you again,

closed 2019-06-24 by thomasonw