platformio/platform-espressif32 #162: heltec_wifi_kit_32 - Vext not defined

affordabletechnology, 2019-02-23:
What kind of issue is this?

- [ x] **Arduino IDE**.

------------------------------------------------------------------

### Configuration
Ide ver 1.8.8

**Operating system**:
Win 10
**PlatformIO Version** (`platformio --version`):
Confirming issue #2107 also applies when compiling with Arduino

### Description of problem
C:\Users\paul\Documents\Arduino\libraries\Heltec_ESP32\src\heltec.cpp:97:10: error: 'Vext' was not declared in this scope pinMode(Vext,OUTPUT);

#### Steps to Reproduce
1. Heltec_ESP32 at version 1.0.2
2. Open example app Vextcontrol and compile
3. Shed a tear and look for a work-around :-(

### Actual Results
C:\Users\paul\Documents\Arduino\libraries\Heltec_ESP32\src\heltec.cpp:97:10: error: 'Vext' was not declared in this scope pinMode(Vext,OUTPUT);

### Expected Results
Compiled Ok!

### If problems with PlatformIO Build System:
no - problem is with Heltec library

### Additional info
Suspect issue relates to Heltec ESP32 Wifi Kit does not support 'Vext' control of OLED display

Looking into putting an ifdef conditional build to eliminate offending code but need to check board schematic first.

ivankravets, 2019-02-23:
Does it work with Arduino IDE and ESP32 core?  See https://github.com/espressif/arduino-esp32/blob/master/docs/arduino-ide/boards_manager.md

if it does not work with Arduino IDE, please forward issue to https://github.com/espressif/arduino-esp32/issues

cyberman54, 2019-04-11:
This must be patched in [pins_arduino.h](https://github.com/espressif/arduino-esp32/blob/master/variants/heltec_wifi_lora_32_V2/pins_arduino.h) in arduino-esp32 repository, it's not a platformio issue.

ivankravets, 2019-04-17:
Please re-test with upstream version http://docs.platformio.org/en/latest/platforms/espressif32.html#upstream

closed 2019-04-17 by ivankravets