platformio/platform-espressif32 #163: Unable to include WiFi.h

pvint, 2019-03-03:
New to PIO, so forgive me if this is simply a RTFM thing, but I've been struggling with this for many hours: When I try to use Arduino.h and WiFi.h I cannot compile:

```
platformio run
Processing esp32doit-devkit-v1 (platform: https://github.com/platformio/platform-espressif32.git#feature/stage; board:
esp32doit-devkit-v1; framework: arduino)
--------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/espressif32/esp32doit-devkit-v1.html
PLATFORM: Espressif 32 (Stage) > DOIT ESP32 DEVKIT V1
HARDWARE: ESP32 240MHz 320KB RAM (4MB Flash)

DEBUG: CURRENT(esp-prog) EXTERNAL(esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h,
olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)
Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF MODES: FINDER(chain) COMPATIBILITY(soft)
Collected 28 compatible libraries
Scanning dependencies...
Dependency Graph
|-- <WiFi> 1.0
|-- <AsyncTCP> 1.0.3
Compiling .pioenvs/esp32doit-devkit-v1/src/main.cpp.o
Archiving .pioenvs/esp32doit-devkit-v1/lib2f5/libWiFi.a
Indexing .pioenvs/esp32doit-devkit-v1/lib2f5/libWiFi.a
In file included from /home/pvint/.platformio/packages/framework-arduinoespressif32@src-537c58760dafe7fcc8a1d9bbcf00b6f
6/tools/sdk/include/esp32/esp_event_legacy.h:23:0,
from /home/pvint/.platformio/packages/framework-arduinoespressif32@src-537c58760dafe7fcc8a1d9bbcf00b6f6/tools/sdk/inclu
de/esp_event/esp_event.h:26,
from /home/pvint/.platformio/packages/framework-arduinoespressif32@src-537c58760dafe7fcc8a1d9bbcf00b6f6/tools/sdk/inclu
de/esp32/esp_event_loop.h:22,
from /home/pvint/.platformio/packages/framework-arduinoespressif32@src-537c58760dafe7fcc8a1d9bbcf00b6f6/libraries/WiFi/
src/WiFiGeneric.h:27,
from /home/pvint/.platformio/packages/framework-arduinoespressif32@src-537c58760dafe7fcc8a1d9bbcf00b6f6/libraries/WiFi/
src/WiFiSTA.h:28,
from /home/pvint/.platformio/packages/framework-arduinoespressif32@src-537c58760dafe7fcc8a1d9bbcf00b6f6/libraries/WiFi/
src/WiFi.h:32,
from src/main.cpp:5:
/home/pvint/.platformio/packages/framework-arduinoespressif32@src-537c58760dafe7fcc8a1d9bbcf00b6f6/tools/sdk/include/tc
pip_adapter/tcpip_adapter.h:112:5: error: 'ip_addr_t' does not name a type
ip_addr_t ip; /**< IPV4 address of DNS server */
^
In file included from /home/pvint/.platformio/packages/framework-arduinoespressif32@src-537c58760dafe7fcc8a1d9bbcf00b6f
6/tools/sdk/include/esp32/esp_event_legacy.h:23:0,
```

platformio.ini:
```
[env:esp32doit-devkit-v1]
platform = https://github.com/platformio/platform-espressif32.git#feature/stage
board = esp32doit-devkit-v1
framework = arduino
```
I also tried with develop branch and with just "espressif32" all with the same result. If I change to `framework = espidf` I simply get `fatal error: Arduino.h: No such file or directory`

My main.cpp:
```
#include <Arduino.h>
#include <WiFi.h>

void setup() {

  // put your setup code here, to run once:
}

void loop() {
  // put your main code here, to run repeatedly:
}
void app_main()
{

}
```
(I actually want app_main(), but until I get past this issue it's not relevant)

J-Rios, 2019-03-19:
Hi,

Like it said in [this thread](https://community.platformio.org/t/platformio-esp32-tcp-compile-errors/6605/4), adding:

#define CONFIG_TCPIP_LWIP 1

to sdkconfig.h file should solve this problem.

Maybe missing lwip config with menuconfig for this platformio ESP32 Arduino framework?

pvint, 2019-03-20:
That's it!

I was quite confident that it was something simple - thank you very much! :+1:

closed 2019-03-20 by pvint