platformio/platform-espressif32 #164: Linking error: multiple definitions of `app_main`

jakubtopic, 2019-03-03:
After updating espressif32 platform from version 1.5.0 to either 1.6.0 or 1.7.0 I am not able to override definition of the app_main function the way I was before. The following code in my `main.ccp` no longer works due to an error during linking.

main.cpp:
```
extern "C" void app_main() {
    initArduino();
    ...
```

Linking error:
```
Linking .pioenvs/esp32doit-devkit-v1/firmware.elf
.pioenvs/esp32doit-devkit-v1/libFrameworkArduino.a(main.cpp.o): In function `app_main':
main.cpp:(.text.app_main+0x0): multiple definition of `app_main'
.pioenvs/esp32doit-devkit-v1/src/main.cpp.o:main.cpp:(.text.app_main+0x0): first defined here
collect2: error: ld returned 1 exit status
*** [.pioenvs/esp32doit-devkit-v1/firmware.elf] Error 1
```

Is this a bug or the way how to achieve this changed? I'm sorry, if the problem is in fact with ESP-IDF and totally unrelated to PlatformIO.

alvarolb, 2019-03-12:
This is related with my previous post:

https://github.com/platformio/platform-espressif32/pull/142#issuecomment-458913172

Something changed in the framework somehow...

atanisoft, 2019-03-14:
What do you have in your sdkconfig.h, did you set CONFIG_AUTOSTART_ARDUINO to 0 (or comment it), did you do a clean build?

jakubtopic, 2019-03-22:
Sorry for the delay. @atanisoft, thanks for the tip; setting CONFIG_AUTOSTART_ARDUINO to 0 indeed fixes the issue, but how am I supposed to make it persistent? `sdkconfig.h` is outside the project and belongs to the Arduino framework managed by PlatformIO. Next time the framework is updated, I will have to edit the config file again. I definitely need that change to be part of the project (CI, other contributors, etc). At least I know how to fix it temporarily :)

atanisoft, 2019-03-22:
What are you running in app_main? You may consider moving it to init() which is called as part of the Arduino startup code.

jakubtopic, 2019-03-22:
In my app_main I just call initArduino and start several FreeRTOS tasks. Direct use of app_main actually is not even necessary, I just did not find any use for the `setup` and `loop` functions. Actually I don't like it, because this design inherently forces users to use global variables.

atanisoft, 2019-03-22:
Makes sense, you don't need to do much in setup or loop in fact. You can simply call vTaskDelete(NULL); from setup after firing your tasks etc and leave loop as an empty method. You wouldn't need to call initArduino or anything else really.

You may also find it better to use Arduino as an IDF component for your use case as well.

jakubtopic, 2019-03-22:
Thanks for quick answers, I'll give it a try. Regarding the 'Arduino as an IDF component' - is there any material on how to do that while using the PlatformIO?

atanisoft, 2019-03-22:
No problem, check the open PR which adds support for building arduino-esp32 as a component for espidf framework based projects. It should include the necessary pieces of the arduino-esp32 code to allow intermingled IDF and Arduino library code etc.

jakubtopic, 2019-03-22:
Oh, I noticed it, but the open status did not give me much hope. Thanks a lot, I'll take look. Closing the issue :)

closed 2019-03-22 by jakubtopic