platformio/platform-espressif32 #165: BT SPP Acceptor Example: Linker problem

te-bachi, 2019-03-22:
Can't link to ESP SPP API. Maybe a sdkconfig problem.

See also [ESP32-IDF, Linker problem](https://community.platformio.org/t/esp32-idf-linker-problem/6159)

```.pioenvs\lolin32\src\example_spp_acceptor_demo.o:(.literal.esp_bt_gap_cb+0x20): undefined reference to `esp_bt_gap_pin_reply'
.pioenvs\lolin32\src\example_spp_acceptor_demo.o:(.literal.esp_spp_cb+0x48): undefined reference to `esp_bt_gap_set_scan_mode'
.pioenvs\lolin32\src\example_spp_acceptor_demo.o:(.literal.esp_spp_cb+0x4c): undefined reference to `esp_spp_start_srv'
.pioenvs\lolin32\src\example_spp_acceptor_demo.o:(.literal.app_main+0x48): undefined reference to `esp_bt_gap_register_callback'
.pioenvs\lolin32\src\example_spp_acceptor_demo.o:(.literal.app_main+0x4c): undefined reference to `esp_spp_register_callback'
.pioenvs\lolin32\src\example_spp_acceptor_demo.o:(.literal.app_main+0x50): undefined reference to `esp_spp_init'
.pioenvs\lolin32\src\example_spp_acceptor_demo.o:(.literal.app_main+0x54): undefined reference to `esp_bt_gap_set_pin'
.pioenvs\lolin32\src\example_spp_acceptor_demo.o: In function `esp_bt_gap_cb':
example_spp_acceptor_demo.c:(.text.esp_bt_gap_cb+0xbf): undefined reference to `esp_bt_gap_pin_reply'
.pioenvs\lolin32\src\example_spp_acceptor_demo.o: In function `esp_spp_cb':
example_spp_acceptor_demo.c:(.text.esp_spp_cb+0x7b): undefined reference to `esp_bt_gap_set_scan_mode'
example_spp_acceptor_demo.c:(.text.esp_spp_cb+0x8a): undefined reference to `esp_spp_start_srv'
.pioenvs\lolin32\src\example_spp_acceptor_demo.o: In function `app_main':
example_spp_acceptor_demo.c:(.text.app_main+0x124): undefined reference to `esp_bt_gap_register_callback'
example_spp_acceptor_demo.c:(.text.app_main+0x154): undefined reference to `esp_spp_register_callback'
example_spp_acceptor_demo.c:(.text.app_main+0x181): undefined reference to `esp_spp_init'
example_spp_acceptor_demo.c:(.text.app_main+0x1b3): undefined reference to `esp_bt_gap_set_pin'
collect2.exe: error: ld returned 1 exit status```

te-bachi, 2019-03-22:
When I set these line to ```sdkconfig.h```, it links.
But without ```make menuconfig``` I couldn't find it out. Are there plans to add a feature to generate ```sdkconfig.h```?

```
#define CONFIG_CLASSIC_BT_ENABLED 1
#define CONFIG_BT_SPP_ENABLED 1
```

valeros, 2019-09-18:
At the moment, the only way to configure framework is to manually add these config lines either in `sdkconfig.h` or in `build_flags`.

closed 2019-09-18 by valeros

SomenKumarDas, 2024-06-13:
For the people still visiting this thread in'24 (Like me). Enable Classic Bluetooth under component config -> Bluetooth. When you enable classic bluetooth, SPP will also comes, check the both of them and you are good to go.