platformio/platform-espressif32 #166: E (28866) event: mismatch or invalid event, id=63

treii28, 2019-03-27:
This is related to [/espressif/arduino-esp32#2285](https://github.com/espressif/arduino-esp32/issues/2285)
updated in v3.2 of idf (I'm showing 1.7.0 is still on 3.1.*)
[/espressif/esp-idf#2924](https://github.com/espressif/esp-idf/issues/2924)

When client connect to AP:

```
E (28866) event: mismatch or invalid event, id=63
E (28867) event: default event handler failed!
dhcps: send_offer>>udp_sendto result 0
```

treii28, 2019-03-27:
In the meantime, I don't suppose there is anyway to tell platform io to use the newer idf?

velbn, 2019-03-27:
I have the same issue

treii28, 2019-03-28:
as an update, they claim to have a fix for v3.3 which is still alpha. They are trying to port it back to v3.2 which is still beta. Presumably, platformio will see it if they release 3.2 with the patch applied any time soon.

treii28, 2019-03-29:
Wow, so someone points out to me this shouldn't be crashing my board. I look deeper and my crash is yet another bug in a dhcp server exception that is misgenerated and thus is not handled properly. I insert a fix for that.... and find yet another bug that crashes the chip. Is this platform still Beta? WTF?  A wifi device you can't connect to without crashing it?

valeros, 2019-09-18:
We've updated `arduino` and `esp-idf` packages to the latest available. Please retest your project with the [upstream version of the platform](http://docs.platformio.org/en/latest/platforms/espressif32.html#upstream)

closed 2019-09-18 by valeros