platformio/platform-espressif32 #171: dl_matrix3du_alloc failed ESP32-CAM with platformio but not with arduino IDE

meditant, 2019-04-14:
Hello,

The program (CameraWebServer from espressif example) work good but when i activate the face detection i have this error if the programme has been compiled and uploaded with platformIO (the last and updated version) :
matrix3du item alloc failed.
dl_matrix3du_alloc failed

But the same program compiled and uploaded with ArduinoIDE work good ???

I try differents board esp-wrover-kit and esp32dev, but same issue...

[env:esp-wrover-kit]
platform = espressif32
;board = esp-wrover-kit
board = esp32dev
framework = arduino
upload_speed = 115200
;board_build.partitions = default.csv
board_build.partitions = huge_app.csv
board_build.flash_mode = qio
; set frequency to 80MHz
board_build.f_flash = 80000000L

If you have idea ?

Best,

ivankravets, 2019-04-17:
Could you re-test with the latest upstream version http://docs.platformio.org/en/latest/platforms/espressif32.html#upstream and use `esp32cam` as board profile without extra options in `platformio.ini`?

See https://github.com/platformio/platform-espressif32/blob/develop/boards/esp32cam.json

meditant, 2019-04-18:
Hello Ivan,

I try and it's look like good, just change the "upload_speed = 460800", but it's due to my USB to TTL unit.
When i activate de face detection, now i dont have a error !
I do now somes others tests but it's look good !

Thanks you very much
Best,

ivankravets, 2019-04-18:
Thanks! Fixed. Please re-run `pio update`. Now everything should work with the default configuration.

closed 2019-04-18 by ivankravets

Jen0055, 2021-11-09:
Can someone help me i have the same error while running the esp32 cam module. Face detection and face recognition is not working 

Item psram alloc failed. Size: 126720 x 1
matrix3du item alloc failed.
dl_matrix3du_alloc failed

Nicole-Moon, 2022-12-17:
> Could you re-test with the latest upstream version http://docs.platformio.org/en/latest/platforms/espressif32.html#upstream and use `esp32cam` as board profile without extra options in `platformio.ini`?
> 
> See https://github.com/platformio/platform-espressif32/blob/develop/boards/esp32cam.json

can you explain how to do that? I am working in Arduino

Jason2866, 2022-12-18:
@Nicole-Moon This is platformio github. Only for issues with Platformio.
Github issues is NOT a forum for general questions.

commit 69ae2116 (2019-04-18): Set default upload speed to 460800 for ESP32 Cam board // Resolve #171