platformio/platform-espressif32 #172: Support for ESP-WROOM-32

SantiMoskito, 2019-04-18:
- [ ] **Please add support to ESP-WROOM-32 board**.

ivankravets, 2019-04-18:
Could you try this board configuration http://docs.platformio.org/en/latest/boards/espressif32/esp32dev.html ?

CornyjK, 2019-04-24:
Well, I am using this board, so I guess there is the support already.

pfeerick, 2019-09-15:
Using MH ET LIVE ESP32MiniKit boards which use a ESP-WROOM-32, and they work just fun. I suggest this issue be closed. The `esp32dev` configuration should be sufficient for a board not specifically targeted.

closed 2019-09-16 by ivankravets

CiriousJoker, 2020-01-08:
For clarification, does this mean that esp-wroom-32 supports debugging or not?

ivankravets, 2020-01-08:
@CiriousJoker See https://docs.platformio.org/en/latest/platforms/espressif32.html#tools-debug-probes

CiriousJoker, 2020-01-08:
@ivankravets I'm new to this, but based on this article, I would assume that I can't debug the esp-wroom-32 at all, since it's not listed under "onboard" or "external"

ivankravets, 2020-01-08:
You will need external JTAG debugger.

CiriousJoker, 2020-01-08:
@ivankravets 
Thanks for the guidance. Sorry for bugging you again, but I'd really appreciate it, if you could confirm that this combination works in theory:

**ESP32:** https://www.ebay.de/itm/401824398255?var=671645331627 (already bought)
\+ one of these:
**JTAG 1:** https://www.ebay.de/itm/173640005773 (would buy that one since it costs half as much)
**JTAG 2:** https://www.aliexpress.com/item/33048891555.html (other one I found)

I'm kind of confused since both JTAG adapters have "RS232" in their name but look completely different. Also, I fear that the first one won't work because in the platform.ini file I have to set 
`debug_tool = esp-prog`
and esp-prog only exists in the second JTAG adapter and only the second JTAG adapter looks like the one shown on [this ](https://docs.platformio.org/en/latest/plus/debug-tools/esp-prog.html) page.

**So basically:**
Will my esp32 work with the first one, with the second one or not at all?

Again, I'd really appreciate your help!

pfeerick, 2020-01-08:
According to this article, they are both be fine : https://medium.com/@manuel.bl/low-cost-esp32-in-circuit-debugging-dbbee39e508b

The second adapter is an ESP-Prog board, whereas the first has the same FT2232HL chip, but is instead laid out as a breakout board. Because it's the same chip at heart, it's just a matter of connecting the right GPIO as Manuel explains in the article linked above.

joba-1, 2020-01-09:
I have jtag2 and it works.
let me know if you try jtag1...

CiriousJoker, 2020-02-14:
I tried with the first one and it worked, but it was a little bit buggy sometimes.

1. I first had a wire connection issue but that issue has gone away, no idea why
2. My esp32 has 16mb flash but I cannot use all of it with the partition layout. As soon as I fill up the whole flash storage, the esp32 resets in an infinite loop if I try to connect the debugger
3. I cannot upload the program via cable if I have a debugger connected. I have to disconnect the debugger from my pc first.
4. Debugging is relatively slow (at least compared to javascript debugging), but that's probably expected on a microcontroller
5. I had to turn off autoflashing & instead I have to flash manually first. This might actually be the partition layout issue from earlier but I haven't bothered trying to find that out (never change a running system)

Also, I haven't changed with the esp32 from above but instead with a Lolin D32 Pro (pins are the same though so it should work exactly the same on both).

pfeerick, 2020-02-15:
>     3\. I cannot upload the program via cable if I have a debugger connected. I have to disconnect the debugger from my pc first.

If by 'tried with the first one` you mean the CJMCU-2232 'jtag' board, you might be able to use the debugger to upload if you set `upload_protocol = esp-prog` in your `platformio.ini`