platformio/platform-espressif32 #174: Garbage output esp wrover after upgrading to v1.8.0

Bmooij, 2019-04-24:
After upgrading to platform-espressif32 to v1.8.0, the esp continuous outputs garbage:

```
!�t����R��fgRӒ��fg␒Ӓ��fgR�R��fgR�R��f�RӔ��␄gR����␆g.....
```

I tried the following settings:

**platformio.ini**
```
[env:esp-esp32cam]
platform = espressif32
board = esp32cam
framework = arduino
```

and

**platformio.ini**
```
[env:esp-wrover-kit]
platform = espressif32
board = esp-wrover-kit
framework = arduino
```

The current work around is using v1.7.0:
**platformio.ini**
```
[env:esp-wrover-kit]
platform = https://github.com/platformio/platform-espressif32.git#v1.7.0
board = esp-wrover-kit
framework = arduino
```

ivankravets, 2019-04-24:
What is your board? ESP32Cam has PSRAM enabled by default => https://github.com/platformio/platform-espressif32/blob/develop/boards/esp32cam.json#L4

Bmooij, 2019-04-24:
I have ESP32Cam (ESP32-S AI-Thinker)

**platformio.ini**
```
[env:esp-esp32cam]
platform = https://github.com/platformio/platform-espressif32.git#v1.8.0
board = esp32cam
framework = arduino
```

main.cpp
```
#include <Arduino.h>

void setup() {
    Serial.begin(9600);
    Serial.println();
    delay(500);
}

void loop() {
    Serial.printf("%d\n", ESP.getFreeHeap());
    delay(1000);
}
```

ivankravets, 2019-04-24:
Does it work with Arduino IDE?

Bmooij, 2019-04-24:
Not  (yet) checked.

v1.7.0 with ```esp32cam.json``` does work.

ivankravets, 2019-04-24:
Please report this issue to https://github.com/espressif/arduino-esp32/issues

closed 2019-04-24 by ivankravets