platformio/platform-espressif32 #176: updated to latest 1.8.0 in platformio - now having AWS issues

wegunterjr, 2019-04-30:
Is anyone seeing any issues connecting to AWS after the update?

valeros, 2019-05-02:
Hi @wegunterjr! Could you please add more information about the issue? What framework and board do you use?

wegunterjr, 2019-05-02:
ah yeah..sorry about that.
I am using esp32 wrover with esp-idf.  We are using it currently fairly successfully, but noticed some things not quite working once we updated to 1.8.
1. sdkconfig.h was replaced even though it already existed?
2. aws library was moved to a different repo? removed from esp32?
We rolled back for production and are testing 1.8, but wonder if there is any general issues.

I use VSCode PlatformIO IDE.

valeros, 2019-05-02:
Hmm, 1.8.0 was released with the new 3.2 version of the esp-idf framework, so the `sdkconfig.h` was accordingly. As for the `aws_iot` library, I'm not sure what you mean, it's still located in the same place as before so probably it's not the issue. Can you describe what exactly the issue is? At what stage does your application stop working?

bzeeman, 2019-05-02:
There are some breaking changes in 1.8 I discovered. One is that they disable PSRAM by default. So if you are using that include the boot flags to enable PSRAM in your .ini file for the board.

There is also an issue where previously if you had the wrong size for your PSRAm it still worked, but now causes a error on boot that aborts the process because that is the wrong size. Changing the value to reflect the PSRAM size it discovers in the log messages fixes the issue.

valeros, 2019-05-02:
Currently, we supply only the default version of the `sdkconfig.h` file. As for PSRAM, there is [a separate issue](https://github.com/platformio/platform-espressif32/issues/175) that describes how to configure PSRAM.

Returning to the AWS issues, can you provide any steps to reproduce them?

bzeeman, 2019-06-04:
This had/has to do with 1.8 completely writing over the SDKConfig.h file. It's boggling to think that it assumes a 1 size fits all configuration. Is there some better way in PIO to customize the config other than directly editing the sdkconfig.h?

joba-1, 2019-06-04:
there are other ways, but better? No.

Recommended seems to put equivalent defines in platformio.ini build flags
-- 
iphone so i typo

pfeerick, 2019-06-30:
Hm... somethings not quite right there... Going by this line... your SDKConfig.h shouldn't get touched if it exists(*)... if the file is missing, a default one will be used. The asterisk is because if it does exist, but doesn't contain `CONFIG_PTHREAD_STACK_MIN`, again, the default one will be used, but that seems to be it. 

https://github.com/platformio/platform-espressif32/blob/develop/builder/frameworks/espidf.py#L546

At what point is it replacing the `src/sdkconfig.h` file?

valeros, 2020-05-25:
Please reopen if you still need any help.

closed 2020-05-25 by valeros