platformio/platform-espressif32 #178: BLEExceptions error C++ exception handling must be enabled

elettro1, 2019-05-07:
hi, I'm trying to use a simple BLEDevice :: getScan () but I get this error:

In file included from /.platformio/packages/framework-arduinoespressif32/libraries/BLE/src/BLEClient.h:18:0,
from /.platformio/packages/framework-arduinoespressif32/libraries/BLE/src/BLEDevice.h:19,
from /Documents/PlatformIO/Projects/controller/flowerTask.ino:1:
/.platformio/packages/framework-arduinoespressif32/libraries/BLE/src/BLEExceptions.h:13:2: error: #error "C++ exception handling must be enabled within make menuconfig. See Compiler Options > Enable C++ Exceptions."
#error "C++ exception handling must be enabled within make menuconfig. See Compiler Options > Enable C++ Exceptions."

[env:lolin_d32_pro]
platform = espressif32
board = lolin_d32_pro
framework = arduino
monitor_speed = 115200
board_build.partitions = no_ota.csv

How can I solve it?

elettro1, 2019-05-07:
for now I have solved by adding in sdkconfig.h 
#define CONFIG_CXX_EXCEPTIONS 1

valeros, 2019-05-07:
The proper way to enable exceptions in your project is to use the next special flag:
`build_flags = -D PIO_FRAMEWORK_ESP_IDF_ENABLE_EXCEPTIONS`

elettro1, 2019-05-17:
@valeros that works only with framework =espidf

I realized that when changing the framework from arduino to espidf, platformio generates the file sdkconfig.h
while when returning from espidf to arduino it does not delete the file sdkconfig.h and can give that error.

closed 2019-05-17 by elettro1

wegunterjr, 2020-06-12:
is this still true?  I am finding that this doesn't work:
- sdkconfig is overwritten when building
- build_flags doesn't seem to work with the new cmake

valeros, 2020-06-12:
@wegunterjr Exceptions should be enabled using `pio run -t menuconfig` command https://docs.platformio.org/en/latest/frameworks/espidf.html#configuration-for-4-0

wegunterjr, 2020-06-12:
hmmm....I hope it is that easy.  I have the latest platformio and espressif32 platform, but when I try to run : 
`pio run -t menuconfig`
I get the following error:
`FAILED: mconf-idf-prefix/src/mconf-idf-stamp/mconf-idf-build kconfig_bin/mconf-idf`

wegunterjr, 2020-06-13:
hmmmm....so, i uninstalled platformio from vscode, deleted the .platformio folder, and then reinstalled all over again.   I used espressif32 platform, and it seems to be allowing me to run the menuconfig.
WIth the break from v3.3 to v4 for esp-idf, maybe it can't co-exist?