platformio/platform-espressif32 #180: Update to latest ESP-IDF 3.2 release

sle118, 2019-05-14:
Would it be possible to update the platform with the newest version of the esp-idf 3.2?

https://github.com/espressif/esp-idf/commit/1d4eeeb464016101d4d32d902e54a730951dcaa3

One of the things I am interested in is using the poll() function implementation and some other POSIX fixes.

thank you!

valeros, 2019-05-15:
Hi @sle118 ! Please make sure you are using the latest version of the platform. The `esp-idf` framework is already updated to 3.2 in [v1.8.0](https://github.com/platformio/platform-espressif32/releases/tag/v1.8.0)

sle118, 2019-05-17:
I did update the platform to the latest. Note that the function was back ported to 3.2 after 1.8.0 was released. Would 1.8.0 have to be re-packaged from the espressif repository? Not sure how PlatformIO gets the ESP-IDF content.

exocode, 2019-05-17:
is it possible somehow to use Version v4.0-dev (https://github.com/espressif/esp-idf/tree/v4.0-dev)  within PlatformIO?

valeros, 2019-05-20:
@sle118 PlatformIO uses this release https://github.com/espressif/esp-idf/releases/tag/v3.2
Could you please point out the commit with the changes you are looking for?

@exocode We only support the stable version of the framework at the moment.

sle118, 2019-05-23:
Actually, you can choose this. I decided to go bare bone with Eclipse and ESP-IDF. It definitely isn't as convenient as PlatformIO. However, the project I'm working on really doesn't benefit enough from the platform's overhead that it wasn't worth the hassle. I'm sure I'll be back soon, though, with some LoraWan goals in mind. Knowing the ESP-IDF better means I'll be in a better position to understand any problems that typically gets shielded by you strong offering.

Thank you!

ivankravets, 2019-06-05:
Resolved in https://github.com/platformio/platform-espressif32/releases/tag/v1.8.0

closed 2019-06-05 by ivankravets