platformio/platform-espressif32 #185: Banked himem (8 MB psram) support

jcw, 2019-05-28:
I'm trying to get 8 MB PSRAM working in PIO. This requires "banking" to get the area above 4 MB mapped into memory. The demo code used as starting point is from Espressif: [himem_test_main.c](https://github.com/espressif/esp-idf/blob/master/examples/system/himem/main/himem_test_main.c).

First attempts were with `framework = arduino` (see discussion [here](https://community.platformio.org/t/esp32-8-mb-psram-himem/7966)). Then I tried with the Arduino IDE (see https://github.com/espressif/arduino-esp32/issues/2830). No joy with either one so far.

Even with `framework = espidf` in PIO, issues remain. I added the following to `src/sdkconfig.h`:

```
#define CONFIG_ESP32_SPIRAM_SUPPORT 1
#define CONFIG_SPIRAM_BOOT_INIT 1
#define CONFIG_SPIRAM_IGNORE_NOTFOUND 0
/*
#define CONFIG_SPIRAM_USE_MALLOC 1
*/
#define CONFIG_SPIRAM_TYPE_AUTO 1
#define CONFIG_SPIRAM_SIZE -1
#define CONFIG_SPIRAM_SPEED_40M 1
#define CONFIG_SPIRAM_MEMTEST 1
#define CONFIG_SPIRAM_CACHE_WORKAROUND 1
#define CONFIG_SPIRAM_BANKSWITCH_ENABLE 1
#define CONFIG_SPIRAM_BANKSWITCH_RESERVE 4
```

Note: setting CONFIG_SPIRAM_USE_MALLOC breaks the re-build in several places.

The number of linker errors increases. For some reason, `spiram.c` is not being compiled in.

brgaulin, 2019-06-04:
I was having the same issue, see #175 for more information, but the following was what helped me:
```
#define CONFIG_SPIRAM_SUPPORT 1
```

My final additions to sdkconfig.h
```
#define CONFIG_SPIRAM_SUPPORT 1
#define CONFIG_PICO_PSRAM_CS_IO 10
#define CONFIG_ESP32_SPIRAM_SUPPORT 1
#define CONFIG_SPIRAM_BOOT_INIT 1
#define CONFIG_SPIRAM_IGNORE_NOTFOUND 0
#define CONFIG_SPIRAM_TYPE_AUTO 1
#define CONFIG_SPIRAM_SIZE -1
#define CONFIG_SPIRAM_USE_MEMMAP 1
#define CONFIG_SPIRAM_SPEED_40M 1
#define SPIRAM_USE_MALLOC 1
#define CONFIG_SPIRAM_CACHE_WORKAROUND 1
#define CONFIG_SPIRAM_BANKSWITCH_ENABLE 1
#define CONFIG_SPIRAM_BANKSWITCH_RESERVE 8
```

However, it only detects 4MB, not sure how to get the bankswitch working, or if that is expected.
Output:
```
␛[0;32mI (203) heap_init: Initializing. RAM available for dynamic allocation:␛[0m
␛[0;32mI (210) heap_init: At 3FFAE6E0 len 0000F480 (61 KiB): DRAM␛[0m
␛[0;32mI (216) heap_init: At 3FFC0A40 len 0001F5C0 (125 KiB): DRAM␛[0m
␛[0;32mI (223) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM␛[0m
␛[0;32mI (229) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM␛[0m
␛[0;32mI (235) heap_init: At 40089E14 len 000161EC (88 KiB): IRAM␛[0m
␛[0;32mI (242) cpu_start: Pro cpu start user code␛[0m
␛[0;32mI (259) esp_himem: Initialized. Using last 8 32KB address blocks for bank switching on 4352 KB of physical memory.␛[0m
␛[0;32mI (260) cpu_start: Starting scheduler on PRO CPU.␛[0m
␛[0;32mI (0) cpu_start: Starting scheduler on APP CPU.␛[0m
Himem has 4352KiB of memory, 4352KiB of which is free. Testing the free memory...
```

Link is here for the 3.2 ESP-IDF flags: https://docs.espressif.com/projects/esp-idf/en/v3.2/api-reference/kconfig.html

I'm off to bed, but wondering how to make it see the other 4MB, so it can see all 8MB at once, or is that not possible?

jcw, 2019-06-04:
Looks like you've enabled PSRAM support. Which is fine, but not quite the issue here, I think.

As I understand it, the ESP32 can only map up to 4 MB PSRAM into its address space. To access all 8 MB, the ESP32 implements a "bank" mechanism: i.e. we set aside N KB of those 4 MB, and make them switchable (via explicit s/w calls) to select a small section of the remaining RAM. Where N is a multiple of 32. In your example, 8x32 = 256 KB are banked.

Soo, 4096-256 = 3840 KB is "normal" PSRAM, mapped to a region in the ESP32's memory space. The other 8192-3840 = 4352 KB can only be accessed by switching that 256 KB banked "window" around (right now, I don't know whether we have one 256K window, or eight independent 32K windows).

And that's exactly where those special "himem" calls come in. And that's also exactly where I haven't figured out yet how to make that banked area available to an Arduino sketch, built with PlatformIO ...

brgaulin, 2019-06-04:
That makes sense. I have no idea how the banking works, I wonder if you just can't access it all at the same time. I don't have anything to load in just yet, but I'll have 1 large file I need to load into memory, then the rest can be dynamic. So I might just try and set it up that way.

TridentTD, 2019-06-25:
I have just created new repository for  Arduino-ESP32's patch  
that support  ESP32 PSRAM HIMEM.

Try at
https://github.com/TridentTD/arduino_esp32_himem

closed 2022-08-10 by stale[bot]