platformio/platform-espressif32 #188: Compiling errors with OTA rollback feature in 1.8

PaulEverblix, 2019-06-11:
Hello,

For a project, I want to use ESP-IDF's OTA update with rollback functionality via GSM, but I get compiling errors.
To better understand the problem, I first created a new project, with the latest core version (1.8).
It does compile flawlessly.
Trouble happens when I add `#define CONFIG_APP_ROLLBACK_ENABLE` in `.platformio\packages\framework-arduinoespressif32\tools\sdk\include\config\sdkconfig.h` to enable the rollback.
I then try to compile and get those errors : 

C:\.platformio\packages\framework-arduinoespressif32\cores\esp32\esp32-hal-misc.c: In function 'initArduino':
C:\.platformio\packages\framework-arduinoespressif32\cores\esp32\esp32-hal-misc.c:182:5: error: unknown type name 'esp_ota_img_states_t'
     esp_ota_img_states_t ota_state;
     ^

C:\.platformio\packages\framework-arduinoespressif32\cores\esp32\esp32-hal-misc.c:183:9: warning: implicit declaration of function 'esp_ota_get_st
ate_partition' [-Wimplicit-function-declaration]
     if (esp_ota_get_state_partition(running, &ota_state) == ESP_OK) {
         ^
C:\.platformio\packages\framework-arduinoespressif32\cores\esp32\esp32-hal-misc.c:184:26: error: 'ESP_OTA_IMG_PENDING_VERIFY' undeclared (first us
e in this function)
         if (ota_state == ESP_OTA_IMG_PENDING_VERIFY) {
                          ^
C:\.platformio\packages\framework-arduinoespressif32\cores\esp32\esp32-hal-misc.c:184:26: note: each undeclared identifier is reported only once f
or each function it appears in
C:\.platformio\packages\framework-arduinoespressif32\cores\esp32\esp32-hal-misc.c:186:17: warning: implicit declaration of function 'esp_ota_mark_
app_valid_cancel_rollback' [-Wimplicit-function-declaration]
                 esp_ota_mark_app_valid_cancel_rollback();

C:\.platformio\packages\framework-arduinoespressif32\cores\esp32\esp32-hal-misc.c:189:17: warning: implicit declaration of function 'esp_ota_mark_
app_invalid_rollback_and_reboot' [-Wimplicit-function-declaration]
                 esp_ota_mark_app_invalid_rollback_and_reboot();

It appears that the files in `.platformio\packages\framework-arduinoespressif32\tools\sdk\include\bootloader_support` are not the latest and do not include the mandatory classes and enums.

OTA works fine in 1.7, but do lack the rollback feature, that I need.
I have tried to delete and re-download the package, with no success.
Can you check if all the files from ESP-IDF 3.2 have been correctly added in the package for PlatformIO ?
Thanks a lot for you help.

valeros, 2019-06-11:
Does it work with Arduino IDE? Have you tried the latest [upstream version](http://docs.platformio.org/en/latest/platforms/espressif32.html#stable-and-upstream-versions) of the platform?

ikstream, 2019-07-24:
I tried to compile rollback functions for a project and 
using the upstream github link as platform and espidf as framework and encounter basically the same problem. 

I had a look at .platformio/framework-espidf/components/app_update/esp_ota_ops.c and
.platrformio/framework-espidf/components/app_update/include/esp_ota_ops.h to look for the missing type and function.

$ platformio platform show espressif32

espressif32 ~ Espressif 32
==========================
[...]
Version: 1.9.0
Home: http://platformio.org/platforms/espressif32
Repository: https://github.com/platformio/platform-espressif32.git
Vendor: https://espressif.com/
License: Apache-2.0
Frameworks: arduino, espidf, pumbaa, simba

Packages
--------

Package toolchain-xtensa32
--------------------------
Type: toolchain
Requirements: ~2.50200.0
Installed: Yes
Version: 2.50200.80
Original version: 5.2.0
Description: GCC for SoC Xtensa L108

Package toolchain-esp32ulp
--------------------------
Type: toolchain
Requirements: ~1.22851.190618
Installed: No (optional)

Package framework-arduinoespressif32
------------------------------------
Type: framework
Requirements: ~2.10002.190416
Installed: Yes
Version: 2.10002.190628
Original version: 1.0.2
Description: Arduino Wiring-based Framework (ESP32 Core)

Package framework-espidf
------------------------
Type: framework
Requirements: ~3.30202.0
Installed: Yes
Version: 3.30202.190627
Original version: 3.2.2
Description: Espressif IoT Development Framework with the ESP32

Package framework-simba
-----------------------
Type: framework
Requirements: >=12.2.0
Installed: No (optional)

Package framework-pumbaa
------------------------
Type: framework
Requirements: >=2.3.0
Installed: No (optional)

Package tool-esptoolpy
----------------------
Type: uploader
Requirements: ~1.20600.0
Installed: Yes
Version: 1.20600.0
Original version: 2.6.0
Description: ESP8266 and ESP32 serial bootloader utility

Package tool-openocd-esp32
--------------------------
Type: debugger
Requirements: ~1.1000.20181026
Installed: No (optional)

Package tool-mkspiffs
---------------------
Type: uploader
Requirements: ~2.230.0
Installed: Yes
Version: 2.230.0
Original version: 2.30
Description: Tool to build and unpack SPIFFS images

As far as I understand this output the version 3.3 should be installed/in use, am I right?
I checked the git of espressifs esp-idf and compared it to the files mentioned above. The ota files in use by platformio seem to be on the 3.2 release, or earlier.

If I can provide any further information or help fixing this let me know.

valeros, 2019-09-18:
We've updated `arduino` and `esp-idf` packages to the latest available. Please retest your project with the [upstream version of the platform](http://docs.platformio.org/en/latest/platforms/espressif32.html#upstream).
Feel free to reopen the issue if you still have this problem.

closed 2019-09-18 by valeros