platformio/platform-espressif32 #191: Can't connect to ESP32 AP using Windows 10 PC/Laptop

GeorgeFlorian, 2019-06-26:
Board: ESP32 DevKit V4 with Wrover-B
OS: Windows 10
platform = https://github.com/platformio/platform-espressif32.git

I'm having a peculiar issue here. I can't access the WebServer when I am connected to the ESP32 AP Mode using Windows 10. But I can access the WebServer when I am connected to the AP using my Galaxy Note 9.

#and

After the PC fails to access the WebServer , when trying to connect with the phone to the AP it doesn't work. It seems like I am limited to one connection to the AP. It just doesn't work. No debug message, no nothing. Not even the phone returns an error. Which is rather strange I'd say...

This is how I set up AP Mode.
```
//------------------------- logOutput(String)
void logOutput(String string1) {
	circle.push(string1);	
	Serial.println(string1);	
}

logOutput((String)"Starting AP ... ");
  logOutput(WiFi.softAP(ssidAP, passwordAP) ? "Ready" : "WiFi.softAP failed ! (Password must be at least 8 characters long )");
  delay(500);
  logOutput((String)"Setting AP configuration ... ");
  logOutput(WiFi.softAPConfig(local_IP_AP, local_IP_AP, subnetAP) ? "Ready" : "Failed!");
  delay(500);
  logOutput((String)"Soft-AP IP address: ");
  logOutput(WiFi.softAPIP().toString());
```

Nothing strange until now.
This is the `-DCORE_DEBUG_LEVEL=5` aka `Verbose` :
```
[D][WiFiGeneric.cpp:336] _eventCallback(): Event: 15 - AP_STACONNECTED
dhcps: send_offer>>udp_sendto result 0
[D][WiFiGeneric.cpp:336] _eventCallback(): Event: 17 - AP_STAIPASSIGNED
```

See the `dhcps: send_offer>>udp_sendto result 0` ?
I don't get that when I use the phone first and try to access the AP.

And that is all. No other verbose message.

## Update 1: I have tried with another Windows 10 Laptop PC and it still does not work.
## Update 2:

Without changing the code I've started to get `dhcps: send_offer>>udp_sendto result 0` even when I am connecting with my phone. It can connect though.

Here are the `serial debug verbose` when I hit Connect (to ESP's AP) on my phone:
```
Soft-AP IP address: 
109.108.112.114
[D][WiFiGeneric.cpp:336] _eventCallback(): Event: 15 - AP_STACONNECTED
[D][WiFiGeneric.cpp:336] _eventCallback(): Event: 17 - AP_STAIPASSIGNED
[D][WiFiGeneric.cpp:336] _eventCallback(): Event: 17 - AP_STAIPASSIGNED
[D][WiFiGeneric.cpp:336] _eventCallback(): Event: 17 - AP_STAIPASSIGNED
dhcps: send_offer>>udp_sendto result 0
dhcps: send_offer>>udp_sendto result 0
dhcps: send_offer>>udp_sendto result 0
[D][WiFiGeneric.cpp:336] _eventCallback(): Event: 17 - AP_STAIPASSIGNED
```

As you can see:
`[D][WiFiGeneric.cpp:336] _eventCallback(): Event: 17 - AP_STAIPASSIGNED` and `dhcps: send_offer>>udp_sendto result 0` show up **3 times** just for one connection. 

What is happening ?

MkLHX, 2019-07-24:
Hi got the same here!!

GeorgeFlorian, 2019-07-29:
https://github.com/espressif/arduino-esp32/issues/3037

joba-1, 2019-07-29:
have you tried adding this before the send()?
  web_server.sendHeader("Connection", "close");

ESPs can only handle a very limited number of simultaneous connects and Windows browsers tend to do „clever“ things like preloading other pages.

GeorgeFlorian, 2019-07-29:
> have you tried adding this before the send()?
> web_server.sendHeader("Connection", "close");
> 
> ESPs can only handle a very limited number of simultaneous connects and Windows browsers tend to do „clever“ things like preloading other pages.

Before what `send()` ?

joba-1, 2019-07-29:
don't you have in your code a webserver object that uses send() to respond to client requests?

valeros, 2020-05-25:
Please reopen if you still need any help.

closed 2020-05-25 by valeros