platformio/platform-espressif32 #196: Board request: TTGO-T8 v1.7

dpsm64, 2019-07-04:
Board request: TTGO-T8 v1.7 - https://github.com/LilyGO/TTGO-T8-ESP32#t8-v17

What kind of issue is this?

- [ ] **Question**.
      This issue tracker is not the place for questions. If you want to ask how to do something,
      or to understand why something isn't working the way you expect it to, 
      use [Community Forums](https://community.platformio.org) or [Premium Support](https://platformio.org/support)

- [ ] **PlatformIO IDE**.
      All issues related to PlatformIO IDE should be reported to appropriate repository:
      [PlatformIO IDE for Atom](https://github.com/platformio/platformio-atom-ide/issues) or
      [PlatformIO IDE for VSCode](https://github.com/platformio/platformio-vscode-ide/issues)

- [ X] **Development Platform or Board**.
      All issues (building, uploading, adding new boards, etc.) related to PlatformIO development platforms
      should be reported to appropriate repository related to your hardware
      https://github.com/topics/platformio-platform

- [ ] **Feature Request**.
      Start by telling us what problem you’re trying to solve. Often a solution
      already exists! Don’t send pull requests to implement new features without first getting our
      support. Sometimes we leave features out on purpose to keep the project small.

- [ ] **PlatformIO Core**.
      If you’ve found a bug, please provide an information below.

*You can erase any parts of this template not applicable to your Issue.*

------------------------------------------------------------------

### Configuration

**Operating system**:

**PlatformIO Version** (`platformio --version`):

### Description of problem
Board request: TTGO-T8 v1.7 - https://github.com/LilyGO/TTGO-T8-ESP32#t8-v17

#### Steps to Reproduce

1.
2.
3.

### Actual Results

### Expected Results

### If problems with PlatformIO Build System:

**The content of `platformio.ini`:**
```ini
Insert here...
```

**Source file to reproduce issue:**
```cpp
Insert here...
```

### Additional info

atanisoft, 2019-07-05:
@dpsm64 please file an issue on https://github.com/espressif/arduino-esp32 as this will need to be added there before PIO can expose it properly.

dpsm64, 2019-07-09:
Not really clear where to make such a report. The link you sent has an
Issues tab, but it seems geared to bug reports, not feature requests.

DPM

atanisoft, 2019-07-09:
> The link you sent has an
> Issues tab, but it seems geared to bug reports, not feature requests.

It covers both. Same as this area does for PIO.

valeros, 2019-09-18:
Please open an issue here https://github.com/espressif/arduino-esp32/issues in the first place, so the board will be added to the Arduino framework officially.

dpsm64, 2019-09-18:
I did that months ago and a bot recently marked it as stale: https://github.com/espressif/arduino-esp32/issues/2972

atanisoft, 2019-09-18:
@dpsm64 send a PR with your issue to add the board definition, you can use the wrover board definition as a base and name it as ttgo-t8.