platformio/platform-espressif32 #199: Need some help figuring out which version corresponds to which esp-idf version

ssilverman, 2019-07-10:
I need some help with this issue: https://github.com/espressif/esp-idf/issues/3698

They need to know which esp-idf version corresponds to the PlatformIO version.

valeros, 2019-07-10:
You can find packages and their versions in your build log, e.g. 
```
Processing esp32dev (platform: espressif32; board: esp32dev; framework: espidf)
--------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------- Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/espressif32/esp32dev.html
PLATFORM: Espressif 32 1.9.0 #e61786c > Espressif ESP32 Dev Module
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
DEBUG: Current (esp-prog) External (esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)
PACKAGES: toolchain-xtensa32 2.50200.80 (5.2.0), framework-espidf 3.30202.0 (3.2.2), tool-esptoolpy 1.20600.0 (2.6.0)
```

pfeerick, 2019-09-16:
The build log gives you the ESP-IDF version currently in use, and the [release notes](https://github.com/platformio/platform-espressif32/releases) tell you what platform version supports what ESP-IDF version. Pretty sure this issue can be closed now? ;)

valeros, 2019-09-18:
Feel free to reopen the issue if you still need help.

closed 2019-09-18 by valeros