platformio/platform-espressif32 #200: Board request: Geekworm ESP32-WROVER board

badvision, 2019-07-14:
It has similar board specs to the WROVER-KIT but some boards now have 8mb flash on them, otherwise all else is the same including the XTAL frequency.  On the bottom of the board it has a model number: KE32-WROVER-C20

alkhimey, 2019-07-18:
It has 8MByte **PSRAM**. It still has 4MByte of FLASH.

Here is the product page:
https://geekworm.com/products/esp32-worver-development-board
Here is the datasheet:
https://www.espressif.com/sites/default/files/documentation/esp32-wrover_datasheet_en.pdf

valeros, 2019-09-16:
Please open an issue here https://github.com/espressif/arduino-esp32/issues in the first place, so the board will be added to the Arduino framework officially.