platformio/platform-espressif32 #203: Partition table select no longer working

cyberman54, 2019-07-17:
Not sure this is a pio issue, perhaps it's some weird side effect.
Partition table selection seems no longer working.

I have in platformio.ini (complete file see [here)](https://github.com/cyberman54/ESP32-Paxcounter/blob/master/platformio.ini)

```
[env]
framework = arduino
board = esp32dev
board_build.partitions = min_spiffs.csv
```

But the default.csv table from arduino-esp32 framework is used.
pio run -t envdump shows

`  'PARTITIONS_TABLE_CSV': 'C:\\Users\\FooBar\\.platformio\\packages\\framework-arduinoespressif32\\tools\\partitions\\default.csv',`

cyberman54, 2019-07-20:
After some testing i found out that this issue comes up if the board's manifest file does not contain a JSON objekt 

```
{
  "build": {
    "partitions": "<xxx.csv>",
   },
```

It seems then

> You can override default Espressif ESP32 Dev Module settings per build environment using board_*** option, where *** is a JSON object path from board manifest esp32dev.json. 

does not work.

/cc @ivankravets

ivankravets, 2019-07-22:
1. Please remove this line [build_cache_dir = /tmp/platformio-shared-cache](https://github.com/cyberman54/ESP32-Paxcounter/blob/master/platformio.ini#L40). What is it? It will work only on Unix systems. Also, it is recommended for advanced using (cloud computing, etc).
2. You can move `upload_*` and `monitor_*` options from `common` to `env`.

cyberman54, 2019-07-22:
I followed your suggestion No. 1, but unfortunately this seems to have no impact on the partition select issue.

Also followed No. 2, thanks for hint.

ivankravets, 2019-07-23:
You killed original PlatformIO Board Object in https://github.com/cyberman54/ESP32-Paxcounter/blob/master/build.py#L67

Please do not override PlatformIO reserved variables.

closed 2019-07-23 by ivankravets