platformio/platform-espressif32 #204: Core 1 abort call

rolf0303, 2019-07-20:
Hi everyone!
I'm new here so first of all sorry if this is something really obvious.
So, here is my problem:
I'm using VSCode and platformio to program a ESP32 dev board, the idea of this code is just a try of define and execute two different tasks, just to get used to FreeRTOS.
Here is the code:
```
/* This project is a test that runs two tasks */

#include <stdio.h>
#include <stdlib.h>
#include "freertos/FreeRTOS.h"
#include "freertos/task.h"
#include "freertos/queue.h"

/*Variables */
int led = 0;
int count = 0;

/*Function that turns on/off the LED */

void Task_1 (void *p) {
    while (1) {
        
        if (led==0) {
            led=1;
            printf("LED is on\n");

        }
        else {
            led=0;
            printf("LED is off\n");
        }
        vTaskDelay(pdMS_TO_TICKS(2000));
    }
}

/* Function that counts from 0 to 10 */
void Task_2 (void *p) {
    while (1) {
       
        if (count>=10){
            count = 0;
        }
        else {
            count++;
        }
        printf("Count is %d\n",count);
        vTaskDelay(pdMS_TO_TICKS(1000));
    }
}

void app_main() {

/* Create the tasks */
xTaskCreate(Task_1,"Tarea 1",configMINIMAL_STACK_SIZE,NULL,1,NULL);
xTaskCreate(Task_2,"Tarea 2",configMINIMAL_STACK_SIZE,NULL,2,NULL);

/* Start the Scheduler */
vTaskStartScheduler();

}
```

And this is the terminal message:
```
Rebooting...
ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:6128
ho 0 tail 12 room 4
load:0x40078000,len:8816
load:0x40080400,len:6412
entry 0x40080748
␛[0;32mI (30) boot: ESP-IDF 3.30202.190627 2nd stage bootloader␛[0m
␛[0;32mI (31) boot: compile time 18:18:53␛[0m
␛[0;32mI (31) boot: Enabling RNG early entropy source...␛[0m
␛[0;32mI (36) boot: SPI Speed      : 40MHz␛[0m
␛[0;32mI (40) boot: SPI Mode       : DIO␛[0m
␛[0;32mI (44) boot: SPI Flash Size : 4MB␛[0m
␛[0;32mI (48) boot: Partition Table:␛[0m
␛[0;32mI (52) boot: ## Label            Usage          Type ST Offset   Length␛[0m
␛[0;32mI (59) boot:  0 nvs              WiFi data        01 02 00009000 00006000␛[0m
␛[0;32mI (67) boot:  1 phy_init         RF data          01 01 0000f000 00001000␛[0m
␛[0;32mI (74) boot:  2 factory          factory app      00 00 00010000 00100000␛[0m
␛[0;32mI (82) boot: End of partition table␛[0m
␛[0;32mI (86) esp_image: segment 0: paddr=0x00010020 vaddr=0x3f400020 size=0x06df4 ( 28148) map␛[0m
␛[0;32mI (104) esp_image: segment 1: paddr=0x00016e1c vaddr=0x3ffbdb60 size=0x021e0 (  8672) load␛[0m
␛[0;32mI (108) esp_image: segment 2: paddr=0x00019004 vaddr=0x40080000 size=0x00400 (  1024) load␛[0m
␛[0;32mI (113) esp_image: segment 3: paddr=0x0001940c vaddr=0x40080400 size=0x06c04 ( 27652) load␛[0m
␛[0;32mI (132) esp_image: segment 4: paddr=0x00020018 vaddr=0x400d0018 size=0x13e60 ( 81504) map␛[0m
␛[0;32mI (161) esp_image: segment 5: paddr=0x00033e80 vaddr=0x40087004 size=0x0472c ( 18220) load␛[0m
␛[0;32mI (175) boot: Loaded app from partition at offset 0x10000␛[0m
␛[0;32mI (175) boot: Disabling RNG early entropy source...␛[0m
␛[0;32mI (175) cpu_start: Pro cpu up.␛[0m
␛[0;32mI (179) cpu_start: Starting app cpu, entry point is 0x40080f1c␛[0m
␛[0;32mI (170) cpu_start: App cpu up.␛[0m
␛[0;32mI (190) heap_init: Initializing. RAM available for dynamic allocation:␛[0m
␛[0;32mI (197) heap_init: At 3FFAE6E0 len 0000F480 (61 KiB): DRAM␛[0m
␛[0;32mI (203) heap_init: At 3FFC1138 len 0001EEC8 (123 KiB): DRAM␛[0m
␛[0;32mI (209) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM␛[0m
␛[0;32mI (215) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM␛[0m
␛[0;32mI (222) heap_init: At 4008B730 len 000148D0 (82 KiB): IRAM␛[0m
␛[0;32mI (228) cpu_start: Pro cpu start user code␛[0m
␛[0;32mI (246) cpu_start: Starting scheduler on PRO CPU.␛[0m
␛[0;32mI (0) cpu_start: Starting scheduler on APP CPU.␛[0m
assertion "next > (intptr_t)block" failed: file "C:\Users\rolfe\.platformio\packages\framework-espidf\components\heap\multi_heap.c", line 124, function: get_next_block
abort() was called at PC 0x400d2243 on core 1

Backtrace: 0x40089403:0x3ffb3c30 0x400896fd:0x3ffb3c50 0x400d2243:0x3ffb3c70 0x40088c8e:0x3ffb3ca0 0x40088d6a:0x3ffb3cc0 0x40089063:0x3ffb3cf0 0x400821f5:0x3ffb3d20 0x40082234:0x3ffb3d50 0x40085399:0x3ffb3d70 0x4000beaf:0x3ffb3d90 0x40086d67:0x3ffb3db0 0x40086f3c:0x3ffb3dd0 0x400851b6:0x3ffb3df0 0x400851db:0x3ffb3e10 0x40085319:0x3ffb3e50 0x400d8ada:0x3ffb3e70 0x400d2739:0x3ffb4180 0x400d0975:0x3ffb41d0 0x400867ed:0x3ffb41f0
```
I'm not sure where the mistake is or what am I doing wrong.
Thank you for your help!

cyberman54, 2019-07-21:
Look like an RTOS / c++ related issuse, not platformio.

rolf0303, 2019-07-21:
Ok, thanks. I will open a issue there.

cyberman54, 2019-07-21:
and close issue here ;-)

closed 2019-07-21 by rolf0303