platformio/platform-espressif32 #206: using Arduino esp32 core in ESP-IDF Framework

sajeel-ahmed, 2019-07-21:
I have made a project on Arduino IDE by using arduino-esp32 libraries. Now, I have switched on PlatformIO and I want to make same project in PlatformIO by using ESP-IDF Framework. Actually, I was adding all Arduino libraries in the "lib" folder. but this is a hectic process. 

Is it possible I can use Arduino esp32 core in ESP-IDF framework so I dont need to write whole libraries again. 

I am new to PlatformIO so it is getting difficult for me to understand what to do. Your help will be really a solution for my problem.

Thank you and Best regards
Sajeel

valeros, 2019-07-30:
Hi @sajeel-ahmed! Can you explain what the issue is? We have [an example](https://github.com/platformio/platform-espressif32/tree/develop/examples/espidf-arduino-wifiscan) that demonstrates how to use `esp-idf` and `Arduino` frameworks together.

sajeel-ahmed, 2019-08-06:
Hi, I am very sorry for the late reply. 

I want to use arduino libraries such as Adafruit Neopixel and other arduino libraries in platformIO along with esp-idf framework. I could not do this and for this reason I was looking for help. 

Thank you for your time and consideration.

valeros, 2019-09-18:
Please take a look [at this example](https://github.com/platformio/platform-espressif32/tree/develop/examples/espidf-arduino-wifiscan)

closed 2019-09-18 by valeros