platformio/platform-espressif32 #209: New Board Request: TTGO-T-Display

ivmaks, 2019-07-24:
Please add support for TTGO-T-Display board:
https://github.com/Xinyuan-LilyGO/TTGO-T-Display

DaQue, 2019-08-22:
Have 1 and two more stuck in customs right now.

ivmaks, 2019-08-24:
https://github.com/JakubAndrysek/TTGO_T_Display
may be useful

valeros, 2019-09-18:
Please open an issue here https://github.com/espressif/arduino-esp32/issues in the first place, so the board will be added to the Arduino framework officially.

g19i85g, 2020-06-01:
The board was added to the Arduino framework.
https://github.com/espressif/arduino-esp32/pull/3693

valeros, 2020-07-13:
@g19i85g It looks like the PR wasn't merged, so the board is still not available https://github.com/espressif/arduino-esp32/tree/master/variants

rwmingis, 2020-07-18:
I got it to compile and run using the TTGO-T1 board instead, but on that board they moved SCL to GPIO23 instead of default 22 because that shares a pin with the built in LED.  I've changed it back to 22, but I2C still doesn't work.  REally hope they get this sorted as i've based my latest project on this only to realise today that PIO doesn't support T-display (even though hardware built, ooops!)

rwmingis, 2020-07-20:
If it helps anyone the i2c issue was solved by changing the default address. Can't explain how changing a board in platformio would change the default i2c address but that fixed it.

JustSomeRandomDIYGuy, 2020-10-20:
Actually, for a quick solve I found ivmak's comment really helpfull. Simply copy the build_flags entries from the platformio.ini from Kuba Andrýsek's github example and it works instantly: 
[`example platformio.ini`](https://github.com/JakubAndrysek/TTGO_T_Display/blob/master/TTGO_example/platformio.ini)

Basically, the problem seams to be the missed defined pins for talking to the display.

avdz, 2021-12-08:
The provided workaround works. I also noticed that a similar file is provided in the TFT_eSPI library by LilyGO.
However it would still be nice if the board is added to the Board Explorer.

rwmingis, 2021-12-26:
Yes, I am surprised this board is still not in the system after 18 months.  It's becoming quite popular now looking at my youtube feed!

kgrozdanovski, 2022-01-05:
Doesn't look like anybody is bothering with this. Makes one think if this project is a good choice, given that simple requests like this one are ignored for over two years. Also why do we require a board is first supported by Arduino IDE?

projectitis, 2022-06-07:
Just here to add my "me too"

BlueAndi, 2022-06-07:
According to @valeros first the board must be available in https://github.com/espressif/arduino-esp32.
In the past there was this pull request https://github.com/espressif/arduino-esp32/pull/3693 which was closed by the pull request author and so never gets merged. If someone is able to start a new PR there and reference it here, this would be perfect.

fdxrate, 2022-07-05:
@JustSomeRandomDIYGuy  - his suggestion works...
copy this into your project - [https://github.com/JakubAndrysek/TTGO_T_Display/blob/master/TTGO_example/platformio.ini](url)

commit 13a4062e (2023-12-27): Add new boards

Resolves #1231, resolves #209
closed 2023-12-05 by valeros