platformio/platform-espressif32 #212: Question regarding a ESP32 Core commit

GeorgeFlorian, 2019-07-29:
Hello !

I need to include the following commit into PIO: https://github.com/espressif/arduino-esp32/commit/b0d8d4dd442b2953e57b16d05d6bf2a893334f1d

How can I do that ?

valeros, 2019-07-29:
You can try to compile your project using the [staging version of the platform](http://docs.platformio.org/en/latest/platforms/espressif32.html#using-arduino-framework-with-staging-version)

GeorgeFlorian, 2019-07-29:
> You can try to compile your project using the [staging version of the platform](http://docs.platformio.org/en/latest/platforms/espressif32.html#using-arduino-framework-with-staging-version)

Thank you for your reply.
I am afraid that the Stage version will break my code.
Also, can you please check this thread:https://community.platformio.org/t/how-to-properly-modify-sdkconfig-change-some-values-save-it-and-thats-it/8951/2 ?

joba-1, 2019-07-29:
isn't that obvious?

Changing stable might break the code of everyone else. Not an option (with very few exceptions that I think do not apply here).

What I would do is
1) fork stable, add your change and use that fork for your project.
2) fork staging, add your change and create a pull request with arguments why this change is necessary and safe. This way you might be able to stop maintaining your own stable fork with the next stable release.

atanisoft, 2019-08-02:
@GeorgeFlorian You can copy sdkconfig.h to your project src directory and it will be included instead of the copy in the framework directories. You can safely modify some of the parameters in the sdkconfig.h as listed in https://github.com/espressif/arduino-esp32/issues/2899.

GeorgeFlorian, 2019-08-02:
> @GeorgeFlorian You can copy sdkconfig.h to your project src directory and it will be included instead of the copy in the framework directories. You can safely modify some of the parameters in the sdkconfig.h as listed in [espressif/arduino-esp32#2899](https://github.com/espressif/arduino-esp32/issues/2899).

Does that apply to PlatformIO as well ?

GeorgeFlorian, 2019-08-02:
> @GeorgeFlorian You can copy sdkconfig.h to your project src directory and it will be included instead of the copy in the framework directories. You can safely modify some of the parameters in the [...]

Great to know ! Will definitely try ! Thanks !

closed 2019-08-02 by GeorgeFlorian

GeorgeFlorian, 2019-08-05:
> @GeorgeFlorian You can copy sdkconfig.h to your project src directory and it will be included instead of the copy in the framework directories. You can safely modify some of the parameters in the [...]

What about the other `sdkconfig` file, the one that is a text document ?
If I only copy `sdkconfig.h` to the `src` directory and modify it accordingly, what happens to the other `sdkconfig`. Will it change ? Do I have to change it ?

atanisoft, 2019-08-05:
> What about the other `sdkconfig` file, the one that is a text document ?

This is not used at compilation time, it is used by the "make menuconfig" utility to generate the sdkconfig.h file which configures the compilation side of configuration.

> Also, one noob question: do I have to `#include <sdkconfig.h>` in the `main.cpp` ?

Nope, it will be included automatically by the components that require it.