platformio/platform-espressif32 #213: Including <mutex> causes ctime "not been declared" errors

ssilverman, 2019-07-29:
I've filed these two bugs already:
* https://github.com/espressif/esp-idf/issues/3698
  They make some points about how includes are specified.
* https://github.com/espressif/arduino-esp32/issues/3044
  Simple sketches seem to compile correctly in both the Arduino IDE and in PlatformIO/VSCode, but more complex ones do not.

When I create a non-simple sketch and then include `<mutex>`, I get these errors:

```
In file included from /Users/shawn/.platformio/packages/toolchain-xtensa32/xtensa-esp32-elf/include/c++/5.2.0/chrono:41:0,
                 from /Users/shawn/.platformio/packages/toolchain-xtensa32/xtensa-esp32-elf/include/c++/5.2.0/mutex:39,
                 from src/ui/OSCHandler.h:11,
                 from src/ui/OSCHandler.cpp:4:
/Users/shawn/.platformio/packages/toolchain-xtensa32/xtensa-esp32-elf/include/c++/5.2.0/ctime:64:11: error: '::clock' has not been declared
   using ::clock;
           ^
/Users/shawn/.platformio/packages/toolchain-xtensa32/xtensa-esp32-elf/include/c++/5.2.0/ctime:65:11: error: '::difftime' has not been declared
   using ::difftime;
           ^
/Users/shawn/.platformio/packages/toolchain-xtensa32/xtensa-esp32-elf/include/c++/5.2.0/ctime:66:11: error: '::mktime' has not been declared
   using ::mktime;
           ^
/Users/shawn/.platformio/packages/toolchain-xtensa32/xtensa-esp32-elf/include/c++/5.2.0/ctime:67:11: error: '::time' has not been declared
   using ::time;
           ^
/Users/shawn/.platformio/packages/toolchain-xtensa32/xtensa-esp32-elf/include/c++/5.2.0/ctime:68:11: error: '::asctime' has not been declared
   using ::asctime;
           ^
/Users/shawn/.platformio/packages/toolchain-xtensa32/xtensa-esp32-elf/include/c++/5.2.0/ctime:69:11: error: '::ctime' has not been declared
   using ::ctime;
           ^
/Users/shawn/.platformio/packages/toolchain-xtensa32/xtensa-esp32-elf/include/c++/5.2.0/ctime:70:11: error: '::gmtime' has not been declared
   using ::gmtime;
           ^
/Users/shawn/.platformio/packages/toolchain-xtensa32/xtensa-esp32-elf/include/c++/5.2.0/ctime:71:11: error: '::localtime' has not been declared
   using ::localtime;
           ^
/Users/shawn/.platformio/packages/toolchain-xtensa32/xtensa-esp32-elf/include/c++/5.2.0/ctime:72:11: error: '::strftime' has not been declared
   using ::strftime;
```

valeros, 2019-07-30:
Hi @ssilverman ! Can you please provide a minimal example to reproduce the issue?

ssilverman, 2019-07-31:
When I have some time I’ll try to put one together. All my simple tests compile fine. :(
I was hoping those error messages and the note about `newlib` might help in the meantime.

valeros, 2019-07-31:
Include paths might be added in an inaccurate order so some of the standard headers include a wrong header file. Please make sure you don't have files with widespread names like `time.h` etc in your project.

ssilverman, 2019-08-01:
Turns out that I had `Time.h` and `Time.cpp` files in my root source directory (note the capitalization). I tried both renaming these and moving them to a subdirectory without renaming, and both approaches solved the problem.

closed 2019-08-01 by ssilverman

muriloribeiro34, 2019-10-09:
> Acontece que eu tinha `Time.h`e `Time.cpp`arquivos no meu diretório de origem raiz (observe a capitalização). Tentei renomeá-los e movê-los para um subdiretório sem renomear, e as duas abordagens resolveram o problema.

Está correto! basta alterar o nome dos arquivos e o problema é resolvido