platformio/platform-espressif32 #214: Wrong default port for ESP32 OTA

maxgerhardt, 2019-08-01:
The [current documentation](http://docs.platformio.org/en/latest/platforms/espressif32.html#authentication-and-upload-options) states: 

>--port=ESP_PORT ESP32 OTA Port. Default 8266

And then only if there are problems to use 

```
upload_flags =
    --port=3232
```

However, the current default upload port for Arduino OTA is 3232.

https://github.com/espressif/arduino-esp32/blob/a070884441308c5bfa6805d6f515801769ea0b1b/libraries/ArduinoOTA/src/ArduinoOTA.h#L34-L35

https://github.com/espressif/arduino-esp32/blob/a070884441308c5bfa6805d6f515801769ea0b1b/libraries/ArduinoOTA/src/ArduinoOTA.cpp#L101-L109

In fact the default port has always been 3232 in the implementation, though the first commit wrongly says 8266 in a comment ([here](https://github.com/espressif/arduino-esp32/commit/fa1716e73eb746088921ca2d3695a04f68beb1b6#diff-2a79d0176d2bc6341eb1f75eef591998)) because it was pretty much copied from the Esp8266 version. [This commit](https://github.com/espressif/arduino-esp32/commit/cabc4c5bb578fe47af56a575f156129bf177c22f) corrects that.

Therefore I think that for the ESP32 platform, the default upload port should be set to 3232, and the documentation should extended to double check the upload port (which can be customly set by calling `setPort()`) if upload errors occur.

Also see discussion here: https://community.platformio.org/t/esp32-ota-via-ethernet-doesnt-want-to-work/8308/20?u=maxgerhardt

pfeerick, 2019-08-02:
I brought this up about a month ago [here](https://github.com/platformio/platform-espressif32/commit/d6981a0e98875828e93240b7987440c41fbd8b91#commitcomment-34131668) and did PR ([which has been merged](https://github.com/espressif/arduino-esp32/pull/2948/commits/36bc34015c34deae1a266f85e2f8521b81dabd08)) to `esptool.py` which is where the confusion is coming from... as the documentation is correct, since that is still the output and the default port used by 'release' versions of `esptool.py`. :man_facepalming: 

While there hasn't been release to the espressif32 core since, since we know that change is coming, and in fact is how it should have been all along... I'd love to see this done. Maybe just bumping to the current version of the file, since that looks to be the only change made to it since about two released back? :)

maxgerhardt, 2019-08-02:
@pfeerick Alright thanks if the code is already merged then there's no need for this issue and we'll wait for a new release.

closed 2019-08-02 by maxgerhardt

tablatronix, 2019-09-14:
When will this be fixed, not sure I understood the above posts.

pfeerick, 2019-09-14:
@tablatronix  It's fixed in the git version, just hasn't been included in a release version from the espressif/arduino-esp32 core yet. 

In a nutshell, the `espopal.py` tool used to do the uploads had **correct** documentation in it saying default OTA upload port was 8266 - because the same tool is used in the esp8266 project. However the espressif/arduino-esp32 OTA uses port 3232 by default, which was set by passing the `-p / --port` parameter to override the port used by espota. PlatformIO didn't pass this parameter, so is trying to upload via port 8266 on the esp32 (last time I checked, anyway). The PR I reference sets the default port to 3232, and also the documentation output from `espota.py`. In the meantime, the following is needed on the ESP32 until this flows through for OTA uploads to work. 

```
upload_flags =
    --port=3232
```

tablatronix, 2019-09-14:
Ok I am just not seeing the pr or commit, the ones referenced here are for comments and docs not code. Did I miss something.

pfeerick, 2019-09-15:
The PR is against the espressif/arduino-esp32 core, not platformio, as the fix is needed there: https://github.com/espressif/arduino-esp32/pull/2948/commits/36bc34015c34deae1a266f85e2f8521b81dabd08

tablatronix, 2019-09-15:
Ok I was confused because this is already released.

This is already pushed in esp32
https://github.com/espressif/arduino-esp32/commit/2e32022611b417d752aa9aebee323cd815522370

esp32 1.03 was just released 2 days ago.

pfeerick, 2019-09-15:
Oh sweet... thanks! I knew it had been merged... didn't realise the release was out (has only been two mounths since I did the PR :-O).

tablatronix, 2019-09-15:
Yeah it always takes awhile for pio to sync, not sure the reason, maybe there is testing that needs to be done

pfeerick, 2019-09-16:
Download [espota.py](https://raw.githubusercontent.com/espressif/arduino-esp32/master/tools/espota.py) and shove it into `~\.platformio\packages\framework-arduinoespressif32\tools` ? Not pretty but should work! 😆 Or give `upload_flags` command above to specify the port override. Or possibly play with the custom upload command parameter that recently got introduced in `platformio.ini`? I know which one I'd be doing... `curl -LO ... ` 😉 

```
upload_protocol = custom
upload_command = /path/to/upload/tool arg1 --option1 value1 $SOURCE
```