platformio/platform-espressif32 #216: Clarification needed: Stable / Development / Staging

weinrank, 2019-08-05:
Hi,

I recently ran into in issue regarding the different versions (Production / Development / Staging).

Background: I've committed a [PR to the arduino-esp32 Project](https://github.com/espressif/arduino-esp32/pull/2784) project and wanted this PR to be included in my platform.io project.

I've changed the `platformio.ini` file and changed the `platform` entry from `espressif32` to the URL of this github repo. I've expected that the most recent version of espressif's `arduino-esp32` repo will be fetched. But this wasn't the case.

Later on I've added the [staging](http://docs.platformio.org/en/latest/platforms/espressif32.html#using-arduino-framework-with-staging-version) tag and got a more recent version which includes my PR. 

Can you clarify the naming scheme for (Production / Development / Staging)?

Thanks!

valeros, 2019-08-05:
Hi @weinrank ! 
`Stable (Production)` version uses only stable releases of packages (e.g. `arduino-esp32` v1.0.2)
`Development` branch is where new features, packages, bug fixes, etc are added but not yet released.
`Staging` special branch where the latest Arduino framework is cloned directly from GitHub with the most recent changes.

weinrank, 2019-08-05:
Hi @valeros ,

thanks for the clear and precise clarification. It might be helpful to add this to the README.md.

closed 2019-08-06 by weinrank